realsenseai/librealsense #10685: Depth Frame Data unavailable when Advanced Mode enabled - Python

AOtter0, 2022-07-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> 2.50.0.3785) | 
| Operating System & Version |   {Win 11  |  
| Platform                                 | {PC}  |
| SDK Version                            |   {2.0}      | 
| Language                            |  {python}                          | 
| Segment			|  { architecture/robotics }                          | 

### Issue Description

I'm working on scanning a model river topography into a 3d CAD program via python, needing RGBD data. I can successfully pull the data into the program, and I want to augment the High Accuracy present with some advanced controls to reduce the noise of the depth image. This 

This function is meant to run inside of another program, hence the -return- function, and the tying together of the xyzrgb data. The code functions properly until I enable the advanced controls section of the code, where it breaks; thoughts on why?

For personal learning purposes, I'm also happy to hear code recommendations. Thanks.

```
## Based on the -depthscanner- code originally published by Ilmar Hurkxkens @ilmihur
# First import the libraries
import compas
import pyrealsense2 as rs
import numpy as np
import PIL
import cv2 as cv
from PIL import Image
from PIL import ImagePalette as ip

from timeit import default_timer as timer

# Start the scanner here
# Declare RealSense pipeline, encapsulating the actual device and sensors

_pipe = None
_pipe = rs.pipeline()
cfg = rs.config()

def get_pipe():
    global _pipe
    return _pipe

## Find the Device
selected_devices = []                     # Store connected device(s)
h = 720
w = 1280

for d in rs.context().devices:
    selected_devices.append(d)
    print(d.get_info(rs.camera_info.name)+ " detected.")
if not selected_devices:
    print("No RealSense device is connected!")

## Define PointCloud Scan - Returns depth at all points returned by scan
# Plus Temporal filters
def pcScan(): 

    # Declare pointcloud object, for calculating pointclouds and texture mappings
    pc = rs.pointcloud()
    
    # We want the points object to be persistent so we can display the last cloud when a frame drops
    points = rs.points()
    
    # Declare RealSense pipeline, encapsulating the actual device and sensors
    _pipe = rs.pipeline()
    cfg = rs.config()
  
    # Create pipeline and config stream_align the streams
[... 167 lines cut ...]
        ## converts millimeters to meters - Rhino won't read otherwise
        pt_m = []
        for pt in ptx:
            if pt[0] != 0:
                r = [i * 1000 for i in pt] #scale by 1000
                pt_m.append(r)
         

        ##Zip the two lists together
        ptss = list(zip(pt_m, rgbx))

        print(pts[0])
        #print(ptss[0])
        return ptss

        
       
    finally:
        _pipe.stop()
        print("pcScan Complete.")

if __name__ == "__main__":
    s = pcScan()

```

MartyG-RealSense, 2022-07-18:
Hi @AOtter0  I would recommend incorporating checks in your script for whether Advanced Mode is supported before using an Advanced Mode instruction.  This should reduce the risk of an error occurring.  This principle is demonstrated in the SDK's Python example script for Advanced Mode.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/python-rs400-advanced-mode-example.py#L15

AOtter0, 2022-07-18:
Hi @MartyG-RealSense, 
Incorporating the check, the camera supports advanced mode, and it can be enabled. 

```
Found device that supports advanced mode: Intel RealSense D455
Advanced mode is enabled
Depth Control:
 minusDecrement: 10, deepSeaMedianThreshold: 500, scoreThreshA: 511, scoreThreshB: 2047, textureDifferenceThreshold: 0, textureCountThreshold: 0, deepSeaSecondPeakThreshold: 325, deepSeaNeighborThreshold: 7, lrAgreeThreshold: 24
RSM:
 rsmBypass: 0, diffThresh: 4, sloRauDiffThresh: 1, removeThresh: 63
RAU Support Vector Control:
 minWest: 1, minEast: 1, minWEsum: 3, minNorth: 1, minSouth: 1, minNSsum: 3, uShrink: 3, vShrink: 1
Color Control:
 disableSADColor: 0, disableRAUColor: 0, disableSLORightColor: 0, disableSLOLeftColor: 0, disableSADNormalize: 0
RAU Thresholds Control:
 rauDiffThresholdRed: 51, rauDiffThresholdGreen: 51, rauDiffThresholdBlue: 51
SLO Color Thresholds Control:
 diffThresholdRed: 72, diffThresholdGreen: 72, diffThresholdBlue: 72
SLO Penalty Control:
 sloK1Penalty: 60, sloK2Penalty: 342, sloK1PenaltyMod1: 105, sloK2PenaltyMod1: 190, sloK1PenaltyMod2: 70, sloK2PenaltyMod2: 130
HDAD:
 lambdaCensus: 26, lambdaAD: 800, ignoreSAD: 0
Color Correction:
 colorCorrection1: -0.493164, colorCorrection2: 0.831055, colorCorrection3: 0.831055, colorCorrection4: -0.368164, colorCorrection5: -0.133789, colorCorrection6: -0.323242, colorCorrection7: -0.323242, colorCorrection8: 1.19141, colorCorrection9: 0.90918, colorCorrection10: -0.272461, colorCorrection11: -0.272461, colorCorrection12: -0.355469
Depth Table:
 depthUnits: 1000, depthClampMin: 0, depthClampMax: 65536, disparityMode: 0, disparityShift: 0
Auto Exposure Control:
 Mean Intensity Set Point: 1000
Census:
 uDiameter: 9, vDiameter: 9
-A-factor:
 a_factor: 0
Depth Control Min Values:
  minusDecrement: 0, deepSeaMedianThreshold: 0, scoreThreshA: 0, scoreThreshB: 0, textureDifferenceThreshold: 0, textureCountThreshold: 0, deepSeaSecondPeakThreshold: 0, deepSeaNeighborThreshold: 0, lrAgreeThreshold: 0
Depth Control Max Values:
  minusDecrement: 255, deepSeaMedianThreshold: 1023, scoreThreshA: 1023, scoreThreshB: 4095, textureDifferenceThreshold: 4095, textureCountThreshold: 1023, deepSeaSecondPeakThreshold: 1023, deepSeaNeighborThreshold: 1023, lrAgreeThreshold: 2047
After Setting new value, Depth Control:
 minusDecrement: 10, deepSeaMedianThreshold: 500, scoreThreshA: 511, scoreThreshB: 2047, textureDifferenceThreshold: 0, textureCountThreshold: 0, deepSeaSecondPeakThreshold: 325, deepSeaNeighborThreshold: 7, lrAgreeThreshold: 24
```

If I remove the except Exception, I begin to run into issues, but I can't seem to see that anything is printed other than the expected print response.

AOtter0, 2022-07-18:
Hi @MartyG-RealSense, 
After marrying the confirmation script into my own, the code seems to be working, though the script does take a healthy amount of time to run. If you have any ideas on speeding it up, I'd be glad to hear it!

closed 2022-07-20 by AOtter0