realsenseai/librealsense #10689: "unhandled exception escaping from a worker thread" i need your help tos0

HeloGiThUb, 2022-07-19:
| | |
|---|---|
|**librealsense**|2.50.0 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Unhandled exception escaping from a worker thread!
Error type: Unknown error (-529697949)!
```

MartyG-RealSense, 2022-07-19:
Hi @HeloGiThUb  Please follow the advice in the case at https://github.com/IntelRealSense/librealsense/issues/10681 and let me know whether it resolves your problem.  Thanks!

HeloGiThUb, 2022-07-20:
when using depth camera D455, i always see it and i can't  solve it. Can you help me a hand, please! Thanks you very much!!
[image]

MartyG-RealSense, 2022-07-20:
Does the Motion Module start working normally if you enable it after this message has appeared, please?

MartyG-RealSense, 2022-07-20:
The Motion Module will work with USB 2.1 too.  The image below shows my own D455 displaying the Gyro and Accel streams in USB 2.1 mode.

[image]

Are you using the offical short 1 meter long cable supplied with the camera, please?  If you are using your own choice of cable then it is important to choose a high quality one that is suited to transferring data and is not just a device charging cable.  Cable quality is especially important when using cable length longer than 2 meters, otherwise the camera connection can degrade or disconnect.

HeloGiThUb, 2022-07-22:
Actually, i changed among cables. Long or short cables just can't connect the Depth camera. So, do you have other options? thanks you

MartyG-RealSense, 2022-07-22:
Next, please try **completely** uninstalling and reinstalling the two RealSense camera drivers (Depth and RGB) in the Windows **Device Manager** interface using the instructions at the link below.

https://support.intelrealsense.com/hc/en-us/community/posts/4419989666323/comments/4431239847443

MartyG-RealSense, 2022-07-28:
Hi @HeloGiThUb  Do you require further assistance with this case, please?  Thanks!

HeloGiThUb, 2022-07-28:
No, i don't. thanks you so much

MartyG-RealSense, 2022-07-28:
Thanks very much for the update, @HeloGiThUb  - as you do not require further assistance, I will close this case.  Thanks again!

closed 2022-07-28 by MartyG-RealSense