realsenseai/librealsense #10691: 3D point cloud is not available

AMITKUMARMALLICK18, 2022-07-19:
| | |
|---|---|
|**librealsense**|2.50.0 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D455|
|**Serial Number**|037322251658|
|**Firmware Version**|05.13.00.50|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|2.1|
|**Product Line**|D400|
|**Asic Serial Number**|035723061890|
|**Firmware Update Id**|035723061890|
|**Name**|Intel RealSense L515|
|**Serial Number**|f0244822|
|**Firmware Version**|01.05.08.01|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|L500|
|**Asic Serial Number**|0003aa98e87a|
|**Firmware Update Id**|0003aa98e87a|

Please provide a description of the problem
[image]
m

MartyG-RealSense, 2022-07-19:
Hi @AMITKUMARMALLICK18  Does the point cloud appear if you disable the RealSense Viewer's **GLSL** options in the Viewer's **Settings** interface using the instructions at https://github.com/IntelRealSense/librealsense/issues/8813#issuecomment-818883285 please?

AMITKUMARMALLICK18, 2022-07-20:
@MartyG-RealSense , thank you for your response. After following the instructions at #8813, it worked well. However, the distance measure option is not available, which required GLSL to be enabled.

MartyG-RealSense, 2022-07-20:
You can keep the measure option active and still retain access to the pointcloud by only disabling the **Processing** option and leaving Rendering enabled.

AMITKUMARMALLICK18, 2022-07-20:
@MartyG-RealSense Thank You!

MartyG-RealSense, 2022-07-28:
Hi @AMITKUMARMALLICK18  Do you require further assistance with this case, please?  Thanks!

AMITKUMARMALLICK18, 2022-07-28:
No, Thanks.

MartyG-RealSense, 2022-07-28:
Thanks very much for the update.  As you do not require further assistance, I will close this case.  Thanks again!

closed 2022-07-28 by MartyG-RealSense