realsenseai/librealsense #10692: issue for doesn't working

Norihisa0120, 2022-07-20:
| | |
|---|---|
|**librealsense**|2.50.0 RELEASE|
|**OS**|Windows|

what we should do for below error?
Please provide a description of the problem

win-helpers.cpp:109-malformed vid string:mshw0030&col01

MartyG-RealSense, 2022-07-20:
Hi @Norihisa0120  Which RealSense program are you experiencing this error in, please?  Thanks!

MartyG-RealSense, 2022-07-20:
Thank you very much for the information.  

Could you first try unplugging the micro-sized end of the USB cable from the base of the L515 camera, reversing the connector's orientation and plugging it back into the camera, please (USB-C cables are two-way insertion at the micro-sized end).  Does doing so make any difference?

Norihisa0120, 2022-07-20:
Thanks for your follow-up.
Unfortunately, I tried it already and  it was no difference.

Today, it was worked only a time, after that doesn’t working.

Thanks

Norihisa0120, 2022-07-26:
> If you installed the RealSense SDK on Windows using the Intel.RealSense.SDK-WIN10 installer program, could you try running the installer again. It will completely overwrite the previous SDK installation and may therefore eliminate any damage in the previous installation that might be causing the error.

It was no difference after re-installation.
So, I changed PC.
However, there was below issue.
***@***.***

What dose it mean?

Thanks

Norihisa Oda

MartyG-RealSense, 2022-07-26:
Hi @Norihisa0120   If you tried to post an image by email, it does not work.  

**However, there was below issue.
***@***.*****

Please paste it into the comment writing box on this forum page so that it can be displayed.  Thanks!

Norihisa0120, 2022-07-27:
Hi
The error message was below,
14:14:43 context.cpp382 – Found 1 RealSense devices (mask 0xff)
What dose it mean?

Thanks
Norihisa

MartyG-RealSense, 2022-07-27:
**Found 1 RealSense devices** is not an error message.  It is stating how many RealSense cameras are currently attached to your computer.

MartyG-RealSense, 2022-07-27:
Are you able to launch the example program **rs-capture** on your Windows computers using the instructions at https://github.com/IntelRealSense/librealsense/issues/8024#issuecomment-747630656 please?

closed 2022-08-10 by MartyG-RealSense