realsenseai/librealsense #10693: ROS2 Humble - docker container falild to open Realsense D435i.

Aki1608, 2022-07-20:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D435i} | 
| Firmware Version                   | -- | 
| Operating System & Version |   Ubuntu 20.04 | 
| Kernel Version (Linux Only)    |  5.10.65-tegra                                          | 
| Platform                                 | Jetson AGX ORIN  |
| SDK Version                            |  {2.50}                          | 
| Segment			|  {Robot }                          | 

### Issue Description
I am trying to run realsense camera in docker container from NVIDIA. (ros2-humble Isaac_ros container.) When I open realsense-viewer,  I get these error output: 

`20/07 10:18:51,355 ERROR [281473491908864] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_NO_DEVICE
 20/07 10:18:51,355 ERROR [281473590802688] (sensor.cpp:572) acquire_power failed: failed to set power state
 20/07 10:18:51,356 WARNING [281473590802688] (rs.cpp:310) null pointer passed for argument "device"
 20/07 10:18:51,356 WARNING [281473590802688] (rs.cpp:2691) Couldn't refresh devices - failed to set power state
`

I also tried copying the `99-realsense-libusb.rules`, but when i run `sudo udevadm control --reload-rules`, it shows `unning in chroot, ignoring request.` 

What can be the problem? @MartyG-RealSense

MartyG-RealSense, 2022-07-20:
Hi @Aki1608  Intel's guide at the link below for using Docker with librealsense 2.50.0 states that support for graphical programs is not currently included.

https://github.com/IntelRealSense/librealsense/tree/master/scripts/Docker

Aki1608, 2022-07-20:
Hi @MartyG-RealSense,  

I am not sure what you meant by graphical programs. I just want to open it in realsense-viewer SDK from the docker container. I have tried before with other docker container (ros2-foxy isaac_ros container) and I was able to open it.

naitiknakrani, 2022-07-20:
Thanks for the clarity. Earlier on Jetson devices, we were running  Ubuntu 18.04 with Jetpack 4.6.1 and Isaac ROS2 foxy docker container for Ubuntu Bionic version. During that both methods you mentioned for camera access worked inside and outside docker. 

On, Jetpack 5.0.1 with Ubuntu 20.04, librealsense SDK is not connecting with camera, however SDK is installed in Isaac ROS2 Humble docker.

MartyG-RealSense, 2022-07-20:
Generally, the most recent configuration that RealSense users have had success with is Ubuntu 18.04 and Jetpack 4.6, and you were successful with that too.

There have been a small number of RealSense users who have reported success with using the SDK with JetPack 5 and Ubuntu 20.04 - such as in https://github.com/IntelRealSense/realsense-ros/issues/2390 and https://github.com/IntelRealSense/librealsense/issues/10416#issuecomment-1107662065 - though the SDK does not currently have official support for JetPack 5.

The link below has a recent Jetson discussion from the Nvidia forum that refers to a RealSense user's using 20.04, JetPack 5, ROS2 and Docker 

https://forums.developer.nvidia.com/t/running-librealsense-2-and-ros-realsense-camera-on-jetpack-5-x-ubuntu-20-04-is-docker-an-option-with-cuda-working/218625

Aki1608, 2022-07-21:
Hi @MartyG-RealSense,

we were able to solve it by running `udevadm control --reload-rules && udevadm trigger` outside the docker. We were trying to run it inside the docker and that is what causing the issue. We were able to open it in realsense-viewer as well as from realsense-ros wrapper.

Thank you for your help. :)

closed 2022-07-21 by Aki1608

MartyG-RealSense, 2022-07-21:
That's great news that you found a solution, @Aki1608 - thanks very much for the update   :)

jwson97, 2022-08-17:
I also use jetson orin and try to use realsense in isaac-ros docker (humble).

But I cannot install realsense sdk 2.5 inside docker.

When I try to install 

_sudo apt-get install librealsense2-dkms_

I got the error as like

_Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package librealsense2-dkms_

So I can install realsense sdk. 

Please let me know your way. How to install sdk in jetson orin.

naitiknakrani, 2022-08-17:
@jwson97 which branch of Isaac ROS common are you using ?

micaltho, 2022-10-24:
> I also use jetson orin and try to use realsense in isaac-ros docker (humble).
>
> But I cannot install realsense sdk 2.5 inside docker.
>
> When I try to install
>
> _sudo apt-get install [...]

Hi jwson97 

> I also use jetson orin and try to use realsense in isaac-ros docker (humble).
>
> But I cannot install realsense sdk 2.5 inside docker.
>
> When I try to install
>
> _sudo apt-get install [...]

Hi @jwson97 - I got the same error message while installing sdk 2.51 inside isaac-ros docker (Using Jetson AGX Orin) ...

Have you solved the problem yet?

micaltho, 2022-10-24:
Hi @jwson97  - I got the same error message while installing sdk 2.51 inside isaac-ros docker (Using Jetson AGX Orin) ...

Have you solved the problem yet?

jwson97, 2022-11-01:
Hi micalhto - 
I'm sorry but I haven't solved the problem yet. I'm  in learning process, so I connect realsense to foxy on ubuntu 20.04 and process the published information on jetson orin (isaac ros) . Sorry for not being of any help.