realsenseai/librealsense #10695: Recommendations on calibrating and enhancing depth

AOtter0, 2022-07-20:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D455  | 
| Firmware Version                   | 2.50.0.3785 | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |  N/A                                        | 
| Platform                                 | PC  |
| SDK Version                            |  2.0                          | 
| Language                            |  python                          | 
| Segment			|  architecture/robotics/landscape                          | 

### Issue Description
Looking for recommendations on calibrating and enhancing the resolution of my camera. I will export the pointcloud to another application for digital modeling and fluid dynamics simulations. 

What you're looking at below is a table with only a few millimeters of variation, excluding the buckets and textured item. I have a few regions that are giving me trouble, the left has high ridges and the right has low ridges. 

[image]
[image]

My on chip calibration gives me a -0.05 - -0.08 (Good) reading, and I'm using the settings below (all of which translate into the python wrapper):
High Accuracy preset
Resolution: 1280x720
Frame rate: 15
Autoexposure: Enabled
Laser power: 300
Depth Units: 0.000500 (set in the advanced controls depth table--500.000)

Advanced Controls:
Depth Table > Depth Units: (as above) 500.000
AE Control > Mean Intensity Set Point: 1500.000
Disparity Modulation: A Factor: 0.180

Temporal Filter Enabled (default setting)

I've followed what I'm able to re: the [tuning white paper](https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance), so far as I know. Any recommendations on further tuning or changes that I can do to enhance performance via settings or changes to the environment?

MartyG-RealSense, 2022-07-20:
Hi @AOtter0  Please try setting the preset to **Medium Density** instead of High Accuracy, as Medium Density provides a good balance between accuracy and the amount of depth detail on the image.  High Accuracy can result in an image with more gaps due to excessive removal of coordinates.

AOtter0, 2022-07-20:
@MartyG-RealSense, Seems worse. 
[image]

I could add a decimate filter to help after I solve the first issue of the general noise.

MartyG-RealSense, 2022-07-20:
Is the grey bottom surface reflective, please?  In the image it resembles tinfoil.

AOtter0, 2022-07-20:
It shouldn't be reflective. It's a composition of different plastic particles.

[image]

MartyG-RealSense, 2022-07-20:
The material has a good grain on it so its texture should be clearly readale by the camera.

If the depth noise at the sides is resulting from depth reading fluctuations then reducing the 'alpha' of the Temporal Filter from its default of 0.4 down to **0.1** can significantly reduce fluctuation.

AOtter0, 2022-07-20:
It does help stablize the fluctuations. I'll need to throw that into my python script. This gets me pretty close, I think. but still has some errant noise. I don't want to totally eliminate the fidelity of the experiments that I'll run by using the spatial filter, but want to reduce the 1.5cm variation that I might see as below to get as perfect as possible. This is very helpful but want to push it until my ability to manipulate the settings meets the hardware limit--aka tell me if I'm meeting that limit.

Here I've clamped the visualization to 1.7m-1.8m.

[image]

MartyG-RealSense, 2022-07-21:
Reducing the Advanced Mode **Second Peak Threshold** option from its default of '325' down to '0' may help to reduce noise further.  You can test this in the RealSense Viewer by going ro **Stereo Module > Advanced Controls > Depth Control > DS Second Peak Threshold** and using the slider to set its value to 0.

[image]

The images below provide a comparison of an image at Second Peak Threshold of 325 (upper image) and 0 (lower image).

**325**
[image]

**0**
[image]

In Python the Second Peak Threshold value can be defined in a custom **json** camera configuration file or with Advanced Mode scripting, as described at https://github.com/IntelRealSense/librealsense/issues/10224#issuecomment-1060412603

AOtter0, 2022-07-22:
This is great, @MartyG-RealSense. It's very helpful for the middle of the table, but unfortunately, I'm still having noise on the left side of the frame. 

Since I'm doing so much custom, I've transferred from python code for each setting to using the json file.  Your resolution is set at 848x480, and, I'd like to increase this to 1080x720 and see if I can push the depth measure to 250 since I'd like to enhance the resolution even further when translating this into my 3d drafting program, Rhino (via the visual coding add-on Grasshopper). 

Adding the JSON file: 
[test-settings4.txt](https://github.com/IntelRealSense/librealsense/files/9169441/test-settings4.txt)

Intel Realsense Viewer, annotated topview:
[image]

Rhino mesh from pointcloud using JSON settings and temporal filter at 1 (valid 8, of 8):
[image]

What do you think about how to keep improving this?

MartyG-RealSense, 2022-07-22:
1080x720 resolution is not supported on 400 Series cameras like the D455.  1280x720 at 30 FPS is supported for depth and RGB.  However, 848x480 is the optimal depth resolution for accuracy on the D455 model.  

Now that you have achieved improvement recently, do you still get a worse image if you use Medium Density instead of High Accuracy?

AOtter0, 2022-07-25:
Apologies, 1280x720 is correct. 

I'm not sure that the Medium Density preset is available via python, as it doesn't appear when I print the preset options. 

```
    #what are the preset options?
    preset_range = depth_sensor.get_option_range(rs.option.visual_preset)
    current_preset = depth_sensor.get_option(rs.option.visual_preset)
    #print('preset range: '+str(preset_range)+str(current_preset))
```
The high accuracy gives me a better view in the bottom center, but has a generally noisier periphery than the Medium Density (or lack of High Accuracy if Medium Density is not available).

New topview:
[image]

1280x720:
High Accuracy:
[image]

Medium Density:
[image]

848x480:
High Accuracy:
[image]

Medium Density:
[image]

MartyG-RealSense, 2022-07-25:
The numbering of the presets is the same as the order that they are listed in the presets drop-down menu in the Viewer, numbered 0-5.  So the Medium Density preset should be '5'.

```
depth_sensor = dev.first_depth_sensor()
depth_sensor.set_option(rs.option.visual_preset, 5)
```

AOtter0, 2022-07-25:
> The numbering of the presets is the same as the order that they are listed in the presets drop-down menu in the Viewer, numbered 0-5. So the Medium Density preset should be '5'.

Thanks; I have made the changes to the code and updated the images above. In my code, the preset is read before the json file; please let me know if it should read after.

MartyG-RealSense, 2022-07-25:
I agree, I would set the Visual Preset first and then load a json file secondly after that.

MartyG-RealSense, 2022-07-31:
Hi @AOtter0  Do you require further assistance with this case, please?  Thanks!

AOtter0, 2022-08-03:
Hi @MartyG-RealSense, I think this is the best that I might be able to get directly with the camera and settings itself. I'm going to look into some further sub-sampling techniques and see where I can get. Thank you.

MartyG-RealSense, 2022-08-03:
You are very welcome, @AOtter0 - good luck!

closed 2022-08-21 by MartyG-RealSense