realsenseai/librealsense #10696: Request rosbag-inspector exe for checking the frame number of L515 bag file

erwangccc, 2022-07-21:
### Issue Description
Not a issue,  I just want the rosbag-inspector exe to checking the frame number of bag file capturing via L515.  So please send a **.exe** to me,  **_email: wangsc_up@163.com_**. Thanks in advance.

MartyG-RealSense, 2022-07-21:
Hi @erwangccc  There is not a pre-built exe version of the rosbag-inspector tool available.  You can though check the frame number of frames in a bag file by drag-and-dropping the file into the center panel of the RealSense Viewer tool, pausing playback with the keyboard space bar key and then browsing through frames manually with the slider in the options side-panel and reading the frame number on the information overlay at the top of the stream panel.

[image]

erwangccc, 2022-07-21:
Thanks for your quick response!

But the viewer i use doesn't show the frame number, the viewer I use is v2.50.0. 
[image]
If it's caused by diffierent version of Viewer, maybe i should use other version Viewer

MartyG-RealSense, 2022-07-21:
If the 'i' option on top of each stream panel is left-clicked on then it should overlay the stream information at the top of the panel.

erwangccc, 2022-07-21:
OK, i saw the frame counter, but it's the frame id from we open the depth sensor or RGB camera, i want to know the depth\IR\RGB frame recorded in the bag file. 
[image]

For example, we record a bag file on 30fps for 10s, it will have 300 frames for depth or IR data respectively. I hope it will show 300.

closed 2022-07-21 by erwangccc

MartyG-RealSense, 2022-07-21:
Yes, that's right.  End - start = total frame count for that stream.  Thanks very much for the update!