realsenseai/librealsense #10697: How can I get stream data of D435 camera with Wifi connection? (using Jetson nano)

jhforstudy, 2022-07-21:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |  Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  4.9.253-tegra                                        | 
| Platform                                 | Jetson nano  |
| SDK Version                            |  { legacy / 2.50.0 }                          | 
| Language                            |  {Python }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
Hello, I am using a Intel D435 with a Jetson Nano.
My goal is, to get a RGB and Depth stream in my PC (which is connected to the personal WiFi network as of Jetson Nano).

Actually, I succeed to get data with USB cable connection with the code below, but my final goal is to get data with "Wifi" connection which substitutes cable connection.
```python
# Configure depth and color streams
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

# Start streaming
pipeline.start(config)

# color
align_to = rs.stream.color
align = rs.align(align_to)
```

For achieving this goal, I did following and failed:

I tried to command ``rs-server`` in Jetson nano, and open ``realsense-viewer`` in remote PC.
Then, Add source - Add Network Device - type Jetson's ip address.
But, it shows only Infrared stream, and I cannot find any color or depth stream.
[image]

Anyway, I tried to execute my project which includes the code above and it shows
```
Traceback (most recent call last):
  File "/home/jh/copy_ws/src/yolact_ROS/src/eval_ROS.py", line 1821, in <module>
    evaluate(net, dataset)
  File "/home/jh/copy_ws/src/yolact_ROS/src/eval_ROS.py", line 1580, in evaluate
    evalimage_rs(net)
  File "/home/jh/copy_ws/src/yolact_ROS/src/eval_ROS.py", line 708, in evalimage_rs
    pipeline.start(config)
RuntimeError: No device connected
```

As a result, my question is:
1. How can I get color or depth stream from Jetson nano? (with Wifi connection)
2. How can I bring this streamed data into Python pyrealsense2.pipeline()?

Please help me... If you want more info, I'll upload more. Thanks

MartyG-RealSense, 2022-07-21:
Hi @jhforstudy  The Jetson Nano does not have built-in wi-fi connectivity but can have it added by attaching a wi-fi module.  The **Jetson Nano 2GB Development Kit** includes the module in its bundle.  Can you confirm whether your Nano has wi-fi connectivity, please?

If your Nano is equipped with wi-fi, can you also confirm please whether you built the librealsense SDK from source code with CMake on Jetson with the CMake build flag **-DBUILD_NETWORK_DEVICE=ON** included in the CMake build instruction in order to build the realsense2-net module and the rs-server tool?

if rs-server is built on the Nano and running, have you defined a **static IP** after logging into the Jetson via ssh before you tried accessing the camera from the Viewer as a Network Device?

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#26-setting-static-ip

**********

In regard to accessing rs-server over a network with Python, the RealSense SDK has an example script called **net_viewer.py**

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/net_viewer.py

jhforstudy, 2022-07-21:
Hi @MartyG-RealSense , thank you for your quick response!

First of all, I'm using a wi-fi dongle, so wi-fi is working well in my Jetson. Don't worry about it :)

Also, I followed this Linux installation guide (build from source):
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md
I followed all the steps, and finally build & install the package by this command:
```
cmake ../ -DBUILD_EXAMPLES=true -DBUILD_NETWORK_DEVICE=ON -DCMAKE_BUILD_TYPE=Release
```
```
sudo make uninstall && make clean && make && sudo make install
```
It is installed without any error, so realsense2-net & rs-server tool was working.
I think it has temporal errors, because when I reboot Jetson & remote PC and connect again, it works! fortunately...
Do you know why this temporal error occurs? It is not working sometimes.

MartyG-RealSense, 2022-07-21:
If you built from source using installation.md and patched the kernel then Jetson has its own dedicated patch script: 

**./scripts/patch-realsense-ubuntu-L4T.sh**  

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md#building-from-source-using-native-backend

I would also recommend including the build flag **-DBUILD_WITH_CUDA=true** in your CMake build instruction for Jetson so that librealsense can take advantage of the Jetson's Nvidia GPU to provide graphics acceleration for color conversion, pointcloud generation and depth-color alignment operations.

If you continue to experience temporal errors then you could build the SDK around the **RSUSB backend** like the networking guide strongly suggests for Raspberry Pi.  An RSUSB build - which is made by including the build flag **-DFORCE_RSUSB_BACKEND=true** - bypasses the kernel and so is not dependent on Linux versions or kernel versions and does not require patching.

jhforstudy, 2022-07-21:
Sorry for late, but I recognized that ``rs-server`` command is working, but I cannot import ``pyrealsense2_net`` module.
If I run ``python3`` and command like this:
```
Python 3.6.13 |Anaconda, Inc.| (default, Jun  4 2021, 14:25:59) 
[GCC 7.5.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyrealsense2
>>> import pyrealsense2_net
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'pyrealsense2_net'
>>> 
```

I think the whole things are installed, but I don't know why ``python3`` cannot find the module. Could you give me additional advice?

MartyG-RealSense, 2022-07-21:
I have not seen the **stdin** form of the 'No module named pyrealsense2_net' error before.  One RealSense user at https://github.com/IntelRealSense/librealsense/issues/9946#issuecomment-1092710219 stated though that in their particular case they solved the problem by replacing `import pyrealsense2_net as rsnet` with the line below: 

**from pyrealsense2 import pyrealsense2_net as rsnet**

jhforstudy, 2022-07-21:
No way... that also doesn't work for me
```
Python 3.6.13 |Anaconda, Inc.| (default, Jun  4 2021, 14:25:59) 
[GCC 7.5.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from pyrealsense2 import pyrealsense2_net as rsnet
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: cannot import name 'pyrealsense2_net'
>>> 
```

Should I have to uninstall all the packages and install again?

MartyG-RealSense, 2022-07-21:
The **net_viewer.py** example program uses these import lines:

```
import pyrealsense2 as rs
import pyrealsense2_net as rsnet
```

What happens if you use those commands in your own script?

jhforstudy, 2022-07-22:
I'm using my script on ROS, so when I run my node and it gives this error message:
```
(yolact) jh@jh-Lenovo-Legion-5-15ARH05H:~/copy_ws$ rosrun yolact_ROS eval_ROS.py 
Traceback (most recent call last):
  File "/home/jh/copy_ws/src/yolact_ROS/src/eval_ROS.py", line 19, in <module>
    from pyrealsense2 import pyrealsense2_net as rsnet
ImportError: cannot import name 'pyrealsense2_net'
```
This also doesn't work:
```
(yolact) jh@jh-Lenovo-Legion-5-15ARH05H:~/copy_ws/src/yolact_ROS/src$ python eval_ROS.py 
Traceback (most recent call last):
  File "eval_ROS.py", line 19, in <module>
    from pyrealsense2 import pyrealsense2_net as rsnet
ImportError: cannot import name 'pyrealsense2_net'
```

Also, I run ``net_viewer.py`` example code, and it gives:
```
(yolact) jh@jh-Lenovo-Legion-5-15ARH05H:~/librealsense/wrappers/python/examples$ python net_viewer.py
Traceback (most recent call last):
  File "net_viewer.py", line 12, in <module>
    import pyrealsense2_net as rsnet
ModuleNotFoundError: No module named 'pyrealsense2_net'
```

I don't know what is the problem. As you said, you've never seen this situation before, so I'll try to remove all the librealsense package and reinstall again....

MartyG-RealSense, 2022-07-22:
I note that your script at the start of this discussion is using **pipeline.start**.  Whilst this is normal for a pyrealsense2 script, when launching a Python script from ROS as a **node** script the pipeline start instruction is typically not used as the pipeline has already been started by the roslaunch when the node script is run.

Examples of RealSense node scripts provided by the RealSense ROS wrapper that do not use a pipe start instruction are at the link below.

https://github.com/IntelRealSense/realsense-ros/tree/development/realsense2_camera/scripts

MartyG-RealSense, 2022-07-22:
Thanks very much for your patience.  I carefully reviewed your case again from the beginning.  

Use of pyrealsense2_net and net_viewer.py has a pattern of problems in past cases.  So if rs-server is working for you over wi-fi then I would recommend focusing on accessing the camera through the RealSense Viewer rather than with Python code such as net_viewer.  Alternatively, access rs-server with C++ script code instead of using Python if that is possible for your project as C++ networking scripts typically have a higher success rate.

jhforstudy, 2022-07-22:
Thanks @MartyG-RealSense ! so sad to hear that pyrealsense_net module has some problems in Python, I have to find out how to change my project from Python to C++. I'll try it and write the comment if I have any good results. Thanks again.

jhforstudy, 2022-07-25:
I have additional question @MartyG-RealSense, Could you explain more about what you said:

> So if rs-server is working for you over wi-fi then I would recommend focusing on accessing the camera through the RealSense Viewer rather than with Python code such as net_viewer.

Can I bring ``rs.pipeline()`` to my Python script using ``realsense-viewer``?, not ``pyrealsense2-net`` module?

MartyG-RealSense, 2022-07-25:
realsense-viewer uses C++ code so it is unlikely that it could be interfaced with a Python script unfortunately.

Another option that could be considered for transferring streams between your Jetson Nano and a PC is a Python Flask web app that controls a computing board with a RealSense camera attached, like the **Remote-RealSense** project created by a RealSense Python user at https://github.com/IntelRealSense/librealsense/issues/6475#issuecomment-635990864

jhforstudy, 2022-07-26:
Thanks @MartyG-RealSense, now I understand what you said.

closed 2022-07-29 by jhforstudy