realsenseai/librealsense #10699: d435i distance error in Python but not with realsense viewer

bricemiramont, 2022-07-21:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   Ubuntu 20  | 
| Kernel Version (Linux Only)    |  5.4.0-122                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.50.0 }                          | 
| Language                            |  python }                          | 
| Segment			|  }                          | 

### Issue Description
Hello, I searched the past issues but none seems to apply to my issue.
I have a python script that display distance and coordinates of the cursor position. When I launch the script with a D455 cam it works ok, the distance are fine but with a D435i the distances are wrong (10cm error).

In the Realsense Viewer, the distance of the object is ok with both cameras.

Any idea why this occurs and more importantly, how to correct it ?

thank you

[issue-2.pdf](https://github.com/IntelRealSense/librealsense/files/9158286/issue-2.pdf)

MartyG-RealSense, 2022-07-21:
Hi @bricemiramont   If the distance is reported correctly for both cameras in the RealSense Viewer then the issue is unlikely to be related to the calibration of the camera hardware.

Are you using the **get_distance()** instruction in combination with depth to color alignment in your Python script to obtain the distance, please?  If you are then there is a known issue where the coordinates will be accurate at the center of the image but have progressively increasing amounts of measurement error in the coordinates on the outer edges of the image.  This phenomenon is discussed at https://github.com/IntelRealSense/librealsense/issues/6749#issuecomment-654185205

On the D435i camera model the depth and RGB sensors have different field of view (FOV) sizes, with the RGB sensor havig a smaller FOV than the depth sensor.  On the D455 model the RGB and depth sensors are the same size.

My understanding is that using **aligned intrinsics** will provide more accurate measurement results in an aligned image than using depth intrinsics, like at https://github.com/IntelRealSense/librealsense/issues/3735#issuecomment-482933434

bricemiramont, 2022-07-22:
Hi Marty, yes I use the get_distance() instruction but I removed the align_to after reading the issue you mention.
I will check the aligned intrinsics in the 3735 issue. 

I still wonder how realsense viwer achieve its correct result as I point the cursor to an area whith both depth and rgb active in the viewer (in the 3D view). I hoped to do the same in Python.

thank you

MartyG-RealSense, 2022-07-22:
I believe that the Viewer's pointcloud generation is based on the method of using **pc.calculate** to generate the point cloud, like in the 'rs-pointcloud' example program.

https://github.com/IntelRealSense/librealsense/blob/master/examples/pointcloud/rs-pointcloud.cpp#L42-L48

If pc.calculate is used to generate the cloud then typically the XYZ coordinates are obtained with **points.get_vertices()**.  this approach is used by the RealSense SDK's Python example **opencv_pointcloud_viewer.py**

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_pointcloud_viewer.py#L305-L309

bricemiramont, 2022-07-22:
I implemented the technique explained in this issue: [#3688 (comment)](https://github.com/IntelRealSense/librealsense/issues/3688#issuecomment-480570191)
I align the depth frame and get the intrinsics for each frame and after that, the get_distance is getting correct values when using this intrinsecs value.

in the init part of my cam class:
```python
align_to_color = rs.stream.color
self.align = rs.align(align_to_color)
```

in the frame acquisition loop:
```python
frames = self.pipeline.wait_for_frames()
aligned_frames = self.align.process(frames)
depth_frame = aligned_frames.get_depth_frame()
color_frame = aligned_frames.get_color_frame()
self.intr = rs.video_stream_profile(depth_frame.profile).get_intrinsics()
```

Issue is resolved. thank you

closed 2022-07-22 by bricemiramont

MartyG-RealSense, 2022-07-22:
That's great to hear, @bricemiramont - thanks very much for the update and for sharing your solution with the RealSense community!