realsenseai/librealsense #10700: How to send Frames by python Socket

HsiaoCH, 2022-07-21:
Hi 
Since the Raspberry Pi is slow in calculating Align, I want to use socket to send it to other places for calculation
```
frames = pipeline.wait_for_frames()
frames = align.process(frames)#long time
frames.keep()
depth_frame = frames.get_depth_frame()
color_frame = frames.get_color_frame()
```
My problem is as stated in my title
How to send Frames by python Socket
```
frames = pipeline.wait_for_frames()
```
I've searched for a long time, but no answer
If anyone could help me I would appreciate it

MartyG-RealSense, 2022-07-21:
Hi @HsiaoCH  The Python-based RealSense ethernet networking project **EtherSense** (the predecessor of the current RealSense networking system) uses socket.  Is this what you had in mind, please?

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples/ethernet_client_server

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/ethernet_client_server/EtherSenseServer.py

HsiaoCH, 2022-07-21:
@MartyG-RealSense 
thanks for reply
I probably read the content and found that it doesn't seem to do align.process
I want to know the reason

MartyG-RealSense, 2022-07-21:
The EtherSense system is described in detail in the Intel white-paper guide at the link below.

https://dev.intelrealsense.com/docs/depth-camera-over-ethernet-whitepaper

The guide does not make reference to alignment and I have not seen a past EtherSense-based setup that uses alignment.  I have seen no evidence that a Python depth-color alignment script **can't** be used with it though.

 A RealSense user at https://github.com/IntelRealSense/librealsense/issues/10132 was able to demonstrate alignment using the current RealSense networking system based around **rs-server** however.  Whilst this networking system is not desined specifically for Python and is not based on socket, it does use Raspberry Pi as the remote board that the camera is connected to and accessed by the main PC.

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras

[image]

MartyG-RealSense, 2022-07-27:
Hi @HsiaoCH  Do you require further assistance with this case, please?  Thanks!

HsiaoCH, 2022-07-27:
I think I'll keep trying and if I have new questions I'll ask again.

thank you for your help.

closed 2022-07-27 by HsiaoCH

MartyG-RealSense, 2022-07-27:
You are very welcome, @HsiaoCH - please do feel free to create a new issue if you have further questions.  Good luck!