realsenseai/librealsense #10705: Is it possible to set a const vertices value ?

Amayess, 2022-07-22:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Operating System & Version |   Win 10 |
| Platform                                 | PC  |
| SDK Version                            | 2.50.0.0         | 
| Language                            |  C++ | 
| Segment			|  Robot    | 

### Issue Description
Hi, 

As part of the development of a personal project, I was wondering if it was possible to create for example a set_vertices() function and thus be able to modify and set the values of vertices using the attributes of the vertex struct.

For example, make the attribute of type rs2::vertex::z equal to a constant value.

Thank you.

`` /* this segment actually prints the pointcloud */
    auto vertices = points.get_vertices();              // get vertices
    auto tex_coords = points.get_texture_coordinates(); // and texture coordinates

    for (int i = 0; i < points.size(); i++)
    {
        float x = vertices[i].x;
        float y = vertices[i].y;
        float z = vertices[i].z; // I want to assign a constant value here for the vertices[i].z 

        if (vertices[i].z)
        {
            // upload the point and texture coordinates only for points we have depth data for
            glVertex3fv(vertices[i]);

            glTexCoord2fv(tex_coords[i]);
           
        }
    }

MartyG-RealSense, 2022-07-22:
Hi @Amayess  There is not a **set_vertices** SDK instruction that is equivalent to get_vertices.  However, the SDK's 'software device' synthetic camera interface demonstrates in its C++ **rs-software-device** point cloud example program the principle of setting Z to a custom-defined const called BPP and using that as the depth value.  

https://github.com/IntelRealSense/librealsense/blob/master/examples/software-device/rs-software-device.cpp#L17-L36

As rs-software-device makes use of pc.calculate to generate the pointcloud then it should be possible to then use get_vertices to retrieve the XYZ vertices, with the Z vertice determined by the custom BPP value for Z that was defined earlier.

https://github.com/IntelRealSense/librealsense/blob/master/examples/software-device/rs-software-device.cpp#L196-L198

Amayess, 2022-07-22:
Hi @MartyG-RealSense,

The example rs-software-device is uploading an image ( the logo of intel )  then generate a point cloud from it, no ?

MartyG-RealSense, 2022-07-22:
It loads a PNG image with the instruction **create_synthetic_texture** and then generates a point cloud.  You can use live cameras with software device though, like in the discussion at https://github.com/IntelRealSense/librealsense/issues/10067 where it is used to align a RealSense camera's depth stream with RGB from a non-RealSense external RGB camera.

Amayess, 2022-07-22:
I just want to fixe the Z parameter with the realsense no external camera is needed in my case.

MartyG-RealSense, 2022-07-22:
What I meant was that if you were concerned about using software device because it appeared in the rs-software-device example to only support use with a PNG image, it can be used with a live camera too.

Amayess, 2022-07-22:
All right, I will try to follow the example.

Thank you.

Amayess, 2022-07-25:
Hi @MartyG-RealSense 
I followed your suggestions and made this code which I think makes sense since I based it on the example but still nothing works.

Did I forget something? 

Thanks for your help.

`#include <librealsense2/rs.hpp>
#include <librealsense2/hpp/rs_internal.hpp>
#include <iostream>
#include <vector>
#include <csignal>
#include "example.hpp"

// Synthetic frame declaration

struct syntheticframe {
	int W, H, Z_moy;

	std::vector<uint8_t> frame;
};

//Frame parameters  

const int Color_Width = 1280;
const int Color_Height = 720;
const int color_distance = 2;

const int Depth_Width = 848; 
const int Depth_Height = 480;
const int Depth_distance = 2;

 // Intrinsics parameters declaration

// Colore Intrinsics are : Fx is : 642.871 | Fy is : 642.358 | Heigh is : 720 | Width is : 1280 |  model is :  Inverse Brown Conrady |  PPx is : 629.003 |  PPy is :  363.022

const rs2_intrinsics Depth_Intr = {
	Depth_Width, Depth_Height, Depth_Width/2, Depth_Height/2, 434.939 , 434.939 , RS2_DISTORTION_BROWN_CONRADY ,{ 0,0,0,0,0 }
};

const rs2_intrinsics Color_Intr = { Color_Width, Color_Height, Color_Width / 2, Color_Height/2,629.003,363.022,RS2_DISTORTION_INVERSE_BROWN_CONRADY,{0.0668989,-0.000145873,-0.00102118,-0.0203599, -1.07374e+08 } 

};
// - Depth & Color Frame declaration

syntheticframe Depth_Frame; 
syntheticframe Color_Frame;

// Declaration des getters 

syntheticframe& getColor() {
	return Color_Frame;
}

syntheticframe& getDepth() {
	return Depth_Frame;
}

bool isalive = true;

void sighandler(int signal)
{
	isalive = false;
}

int main(int argc, char* argv[]) try {

	window app(1280, 720, "RealSense Software device example");
	glfw_state app_state;
	register_glfw_callbacks(app, app_state);

	rs2::pointcloud pc;
	rs2::points points;

	signal(SIGINT, &sighandler);

	// software-device declaration

	rs2::software_device dev;

	dev.create_matcher(RS2_MATCHER_DEFAULT);

	//declaration des deux capteurs 

	auto depth_sensor = dev.add_sensor("Depth");
	auto color_sensor = dev.add_sensor("Color");

	// Assigner à chaque capteur le stream equivalent 

	auto depth_stream = depth_sensor.add_video_stream({ RS2_STREAM_DEPTH,0,0,Depth_Width,Depth_Height,60,Depth_distance,RS2_FORMAT_Z16,Depth_Intr });

	depth_sensor.add_read_only_option(RS2_OPTION_DEPTH_UNITS, 0.001f);

	auto color_stream = color_sensor.add_video_stream({ RS2_STREAM_COLOR,0,1,Color_Width,Color_Height,60,color_distance,RS2_FORMAT_RGBA8, Color_Intr });

	dev.create_matcher(RS2_MATCHER_COUNT);

	depth_sensor.open(depth_stream);
	color_sensor.open(color_stream);

	//Prepare frame-set getter (the frame queue!)
	rs2::frame_queue q(2);
	std::vector<rs2::frame> bundle;

	rs2::processing_block bundler([&](rs2::frame f, rs2::frame_source& src) {
		bundle.push_back(f);     //add frame to vector
		if (bundle.size() == 2) { //or however many when full
			auto fs = src.allocate_composite_frame(bundle);
			src.frame_ready(fs);
			bundle.clear();
		}
		});
	bundler.start(q); //when enough frames are loaded, the queue is ready

	depth_sensor.start([&](rs2::frame f) {
		bundler.invoke(f);  //add frame to bundler
		});
	color_sensor.start([&](rs2::frame f) {
		bundler.invoke(f);
		});

	Color_Frame = getColor();
	Depth_Frame = getDepth();

	int ind = 0;

	while (isalive)
	{
		color_sensor.on_video_frame({ Color_Frame.frame.data(),[](void*) {},Color_Width * Color_Height,color_distance,(double)ind ,RS2_TIMESTAMP_DOMAIN_HARDWARE_CLOCK,ind,color_stream });

		depth_sensor.on_video_frame({ Depth_Frame.frame.data(),[](void*) {},Depth_Width * Depth_Height,Depth_distance,(double)ind ,RS2_TIMESTAMP_DOMAIN_HARDWARE_CLOCK,ind,depth_stream });

		ind++;

		rs2::frameset fset;
		rs2::frame depth = fset.first_or_default(RS2_STREAM_DEPTH);
		rs2::frame color = fset.first_or_default(RS2_STREAM_COLOR);

		//Point Cloud generation 

		if (depth && color)
		{
			if (auto as_depth = depth.as<rs2::depth_frame>())
				points = pc.calculate(as_depth);
			pc.map_to(color);

			// Upload the color frame to OpenGL
			app_state.tex.upload(color);
		}
		draw_pointcloud(app.width(), app.height(), app_state, points);

	}

	return EXIT_SUCCESS;
}
catch (const rs2::error& e)
{
	std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
	return EXIT_FAILURE;
}
catch (const std::exception& e)
{
	std::cerr << e.what() << std::endl;
	return EXIT_FAILURE;
}

`

Amayess, 2022-07-25:
any idea, please ? @MartyG-RealSense @dorodnic

MartyG-RealSense, 2022-07-25:
What FPS values are you using for Depth_Intr and Color_Intr please?

MartyG-RealSense, 2022-07-25:
I did some initial research and will continue looking at this question tomorrow.  Thanks very much for your patience!

Amayess, 2022-07-26:
Hi @MartyG-RealSense Thank you for your feedback. 

I tried to change de value of the **FPS** - the **format** of the frame -  to use the **post processing block** instead of **sync** as proposed here https://support.intelrealsense.com/hc/en-us/community/posts/360037090133-Construct-Frameset-Object 
but still no frame displayed on my window.

The size of the **fset** is 2 which is logical but the **data size** of each of Depth & Color frame is 0.

I am waiting for your feedback, Thank you.

MartyG-RealSense, 2022-07-26:
I studied the case further but did not make any progress.  I do sincerely apologize.

Amayess, 2022-07-26:
Ah.. Okay no problems.

I have two questions : 

- what are the potential reasons why the frames are always empty and do not receive any data ? 

- What exactly is the **processing block bundler** for? Why is it better than **Sync** ? 

My program recognises the existence of the two frames **Depth** and **Color** because with the following instruction : 

`std::cout << "I Found this Frame:" << fset[0].get_profile().stream_type() << "And this frame:" << fset[1].get_profile().stream_type() << std::endl; 
`
Gives me color and depth back. 

But the main problem is that the frames created do not contain any data :/

MartyG-RealSense, 2022-07-26:
I think that the best approach to your questions above will be to offer a couple of references as further reading material.

https://github.com/IntelRealSense/librealsense/issues/9144

https://support.intelrealsense.com/hc/en-us/community/posts/360037090133/comments/360011508794

Please do feel free to ask any follow-up questions that emerge from reading those references.

Amayess, 2022-07-27:
> #9144  We both used the same reference for writing the code.

> https://support.intelrealsense.com/hc/en-us/community/posts/360037090133/comments/360011508794
> This is exactly the reference I used to make my programme, I mentioned it here: https://github.com/IntelRealSense/librealsense/issues/10705#issuecomment-1195153958

But I have a feeling that the software device is more adequate when you have **two different cameras**. **My goal is just to create a frame with custom parameters and then generate a point cloud with it.**

I think the logic I followed in my code here : https://github.com/IntelRealSense/librealsense/issues/10705#issuecomment-1193844529 is good, but for some reason the frames I create **don't contain any data**. 

- I first thought it was a problem with the **auto exposure** so I put a loop for the **wait_for_frames()** to give the camera time to initialize properly but I get an error that says: **frame didn't arrive in time**, what might be the reasons, please ? 

- What is the real purpose of post processing bundle, please ?

Thank you.

MartyG-RealSense, 2022-07-27:
I think that what you are referring to with 'post processing bundle' is the **custom processing block** ?

If so, then the custom processing block allows you to manually define your own frameset instead of using the default frameset structure of the RealSense SDK.  This helps you to handle frames in the way that you prefer.

In regard to the **frame didn't arrive in time** error, this suggests that frames are not arriving quickly enough to avoid a timeout occurring.  You should be able to avoid the program exiting because of this error by replacing wait_for_frames() with **try_wait_for_frames()** as this version of the command allows the program to continue running if a timeout occurs.

MartyG-RealSense, 2022-08-01:
Hi @Amayess  Do you require further assistance with this case, please?  Thanks!

Amayess, 2022-08-01:
Hi @MartyG-RealSense 

You can close this case, it doesn't really fit in with what I want to do now.

I'm trying another method, hopefully it will work.

Thanks.

closed 2022-08-01 by Amayess

MartyG-RealSense, 2022-08-01:
Thanks very much @Amayess for the update and good luck with your new method!