realsenseai/librealsense #10707: L515 pugin to PC then the Viewer don't recognize

erwangccc, 2022-07-22:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {L500 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Win 10  |               
| Platform                                 | PC  |
| SDK Version                            | 2.50.0                       | 

### Issue Description
There are two windows PCs. 
I connect L515 to one of my PCs, it shows normally.
[image]
But i connect to another PC, it shows nothing, there is no button on the GUI. 

[image]

The USB ports of two windows PC are all USB 3.0 or USB 3.1.
Do i miss something?

MartyG-RealSense, 2022-07-22:
Hi @erwangccc  Please check in the camera privacy settings of Windows whether permission to access the camera and permission for applications to use the camera are both set to On using the instructions at https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789

erwangccc, 2022-07-25:
Thank you very much. It's the permission problem.

erwangccc, 2022-07-25:
Another probelm is the ghost image.
[image]
There are so much redundant data around the moving hand. It's harmful to the post-processing. 
But on another PC, it's not so much redundant data. Maybe there are some parameters i should modify, right?

MartyG-RealSense, 2022-07-25:
Hi @erwangccc  Please try resetting the RealSense Viewer to its default settings.  This is different from resetting the camera hardware.  Instructions for resetting the Viewer configuration can be found at the link below.

https://support.intelrealsense.com/hc/en-us/community/posts/360052013914-RGB-sensor-started-showing-strange-lines-in-images-D435-

erwangccc, 2022-07-26:
@MartyG-RealSense Hi, i have follow the instructions to reset the Viewer, but it still has ghost image around the moving hand. Maybe it's other reason for it, do you have other advices for it? Thanks

MartyG-RealSense, 2022-07-26:
Does the image improve if you unplug the micro-sized end of the USB cable from the base of the L515 camera, turn the connector around the other way and re-insert it into the base of the camera (USB Type-C cables are two way insertion at the micro-sized end)?

erwangccc, 2022-07-27:
@MartyG-RealSense The image didn't improve after i turn the connector.  This situation  did not occur at another windows PC.

MartyG-RealSense, 2022-07-27:
Please run the **rs-capture** example program to see whether the poor image also occurs in that program.  This will help to identify whether the problem on a particular PC is with the RealSense Viewer specifically.  Instructions for launching rs-capture on Windows can be found at https://github.com/IntelRealSense/librealsense/issues/8024#issuecomment-747630656

erwangccc, 2022-07-29:
@MartyG-RealSense Hi, the depth image is normal which is from **rs-capture** example program. Maybe it's the differnent setting at Viewer causing this problem, but how to adjust the parameters at Viewer?

MartyG-RealSense, 2022-07-29:
The RealSense Viewer applies a range of post-processing filters by default, whilst rs-capture does not.  If you left-click on the blue icon beside **Post-Processing** to disable all filters by turning the icon from blue (on) to red (off) then the resulting unfiltered image should help to confirm whether it is the filters that are creating the ghost image.

[image]

erwangccc, 2022-07-29:
@MartyG-RealSense Hi, thank you, it's caused by post-filter, it disappears after i close post-processing.
Close this issue.

closed 2022-07-29 by erwangccc

MartyG-RealSense, 2022-07-29:
It's great to hear that disabling the filters resolved your problem.  Thanks very much for the update!