realsenseai/librealsense #10708: D435 "No Frames Received" after sitting unused for a few months

RadCLarite, 2022-07-23:
Model D435 with latest Firmware on a Win 11 Laptop and most recent SDK Label. 

Just trying to run the D435 after it has been sitting unused for a few months (year?) and I get no frames received on the Viewer app. Interestingly, I seem to get one set of images from the Depth Quality Tool, then it starts flashing the No Frames received error on both the Depth stream and the Infrared stream. 

Any suggestions appreciated. 

-y

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

RadCLarite, 2022-07-23:
additional info: rs-capture just crashes after a few seconds after running it. Besides flashing up some blank windows, it doesn't do anything.

MartyG-RealSense, 2022-07-23:
Hi @RadCLarite  There have been a very small number of past cases where a 400 Series camera has displayed the described behaviour after having been stored unused in a place such as a cupboard or drawer for weeks or months.  In some of those cases, the problem self-corrected a few weeks after being taken out of storage.  

For example, in one case a camera was shipped for return to Intel from India to California and was working again by the time that it arrived.  This suggests that removal of condensation may help (google for the term **remove condensation from device**).

MartyG-RealSense, 2022-08-07:
https://github.com/IntelRealSense/librealsense/issues/10735 is a case where the RGB stream no longer displayed frames after the camera was stored for a long period.  The RealSense user in that case found that the RGB stream could appear if it was left running continuously for an hour, adding evidence to the trend of the camera's ability to self-recover from a storage-induced problem after a time.

closed 2022-08-20 by MartyG-RealSense