realsenseai/librealsense #10709: depth data extraction from depth frames of Intel Realsense camera

Pshcer, 2022-07-23:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (11)  | 

How to do the depth data extraction (from depth frames of Intel Realsense camera )?
Is there anyone out there who has done this earlier and is there any way to get it done?

MartyG-RealSense, 2022-07-23:
Hi @Pshcer  Are you aiming to retrieve the data stored in a depth frame and do something with the depth values? 

If you wish to access the depth frame contents then the SDK instruction **get_data()** is likely to be what you require.  For example in the simple snippet of C++ code below:

```
rs2::pipeline pipe;
pipe.start();
rs2::frameset frames = pipe.wait_for_frames();
rs2::frame frame = frames.first(RS2_STREAM_DEPTH);
if (frame)
frame.get_data(); // Pointer to depth pixels
```

In Python, get_data is typically used to store the contents of the frames in **numpy** arrays so that the stored values can be retrieved from the array and have further work done with it.  For example, the Python snippet below stores information from depth and color frames into separate numpy 'np' arrays.

```
depth_image = np.asanyarray(depth_frame.get_data())
color_image = np.asanyarray(color_frame.get_data())
```

Pshcer, 2022-07-25:
That helped thanks.

closed 2022-07-25 by Pshcer

MartyG-RealSense, 2022-07-25:
Thanks for the update, @Pshcer   :)