realsenseai/librealsense #10710: how to use rs-convert -d

Pshcer, 2022-07-23:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415} | 

### Issue Description
how to use the tool rs-convert -d

could anyone suggest the format for using it?

MartyG-RealSense, 2022-07-23:
Hi @Pshcer  The -d command does not represent a format to convert to.  It instructs rs-convert to only extract depth frames from the bag and not other types of frame such as color.  You would insert it into your conversion command.  For example:

**rs-convert.exe -i "bag file name" -d -v "path of output"**

The simplest way to construct a conversion instruction without using path names is to copy the bag file into the same folder as the rs-convert program.  On Windows, if you have installed the full RealSense SDK then this will be at the following location:

**C: > Program Files (x86) > Intel RealSense SDK 2.0 > tools**

You can find this folder quickly by right-clicking on the RealSense Viewer launch shortcut icon on the Windows desktop and selecting the menu option 'Open file location'.

If the bag file is pasted into the tools folder then you could perform a -v (csv) conversion with this instruction (change 'test.bag' to the name of your bag file).

**rs-convert.exe -i test.bag -d -v test**

If you do specify path names, do not put " " quotation marks around the path name.  For example, use `C:\Program Files (x86)\Intel RealSense SDK 2.0` and not "C:\Program Files (x86)\Intel RealSense SDK 2.0"

An example of a conversion command for Windows that only extracts the depth frames and specifies paths for rs-convert, the bag file and the location to place the extracted files could look like this:

**C:\Program Files (x86)\Intel RealSense SDK 2.0\tools\rs-convert.exe -i C:\Users\MartyG\Documents\test.bag -d -v C:\Users\MartyG\Documents\ply**

MartyG-RealSense, 2022-07-25:
You are very welcome, @Pshcer - that's excellent news.  Thanks very much for the update!

Pshcer, 2022-07-25:
While converting using "-v" i got a csv file with 0s and some numbers. What are those numbers. Where can i read the interpretation about that.

MartyG-RealSense, 2022-07-25:
A .csv file contains a textual representation of camera data that can be imported into software packages such as spreadsheets and databases.

Does your extracted .csv file resemble the one at https://github.com/IntelRealSense/librealsense/issues/2671#issuecomment-435773984

[image]

If you used -d to only extract depth frams to .csv then the values should represent the depth value at a particular width / height coordinate.  In the above image you can see that the bottommost row number that has values is '720', whilst the furthest column number at the top of the sheet is '1280'.  We can conclude from this that the numbers represent values on the coordinates of a 1280x720 resolution image.

[image]

Pshcer, 2022-07-26:
what does the numbers in spreadsheet indicates?

MartyG-RealSense, 2022-07-26:
An image is made up of horizontal rows and vertical columns of pixels, and the number of pixels in the image is determined by the width and height of the resolution that is being used (for example, an image with 1280x720 resolution is 1280 pixels wide and 720 pixels high.  You can therefore locate any pixel on an image using its width height location.  The center pixel coordinate of an image for a particular resolution will be half its resolution, such as 640x350 for the 1280x720 resolution.

The numbers on the csv spreadsheet therefore represent the depth value in meters for a specific coordinate on the depth image.  A zero value represents a coordinate where there is no depth data, either because the camera could not detect it or because there is no surface to read in that area of the real-world scene.

Pshcer, 2022-07-27:
THANKYOU SO MUCH. ISSUE RESOLVED

MartyG-RealSense, 2022-07-27:
You are very welcome, @Pshcer - I'm pleased that I could help.  :)

closed 2022-07-29 by Pshcer