realsenseai/librealsense #10711: How export ply as ASCII format in C#?

M-CAP7AIN, 2022-07-23:
`rs2::save_to_ply exporter(path, pc);
exporter.set_option(rs2::save_to_ply::OPTION_PLY_BINARY, 0.f);
exporter.export_to_ply(points, infrared);`

How use this code in **C# library**, I need export **PLY** file as **ASCII** format?

MartyG-RealSense, 2022-07-23:
Hi @M-CAP7AIN  There is not an available example of using the **save_to_ply** ply export command in the C# wrapper unfortunately.  There is one for **export_to_ply** at https://github.com/IntelRealSense/librealsense/issues/7362 though this will not export ASCII (binary = false).

Potentially the save_to_ply could be accessed in C++ through the C# wrapper via **NativeMethods**.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/csharp/Documentation/pinvoke.md

https://github.com/IntelRealSense/librealsense/issues/7848

**********

If it is possible for you to instead use the 3D mode of the RealSense Viewer tool to export a ply then the Viewer's **Export** option offers the ability to select a **Textual** ply export instead of binary.

[image]

MartyG-RealSense, 2022-07-29:
Hi @M-CAP7AIN  Do you require further assistance with this case, please?  Thanks!

M-CAP7AIN, 2022-07-29:
Hi @MartyG-RealSense
I have to change the source codes and get a new build to have ASCII output by default, or decode the binary output.
[https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md](url)

Thanks.

MartyG-RealSense, 2022-08-04:
Hi @M-CAP7AIN  Is your problem now solved or do you require further assistance, please?

M-CAP7AIN, 2022-08-04:
Hi @MartyG-RealSense , I use C++ for export PLY with texture and ASCII. [Sample](https://github.com/M-CAP7AIN/IntelRealSenseExportPLY)

`      auto frames = pipe.wait_for_frames();

        auto color = frames.get_color_frame();
        
        color = frames.get_infrared_frame();

        pc.map_to(color);

        auto depth = frames.get_depth_frame();

        points = pc.calculate(depth);

         rs2::save_to_ply exporter(inputArg, pc);
         //exporter.set_option(rs2::save_to_ply::OPTION_PLY_MESH, 1.f);
         exporter.set_option(rs2::save_to_ply::OPTION_PLY_BINARY, 0.f);
         //exporter.set_option(rs2::save_to_ply::OPTION_IGNORE_COLOR, 0.f);
         exporter.set_option(rs2::save_to_ply::OPTION_PLY_NORMALS, 1.f);
         exporter.process(frames);
`

closed 2022-08-04 by MartyG-RealSense