realsenseai/librealsense #10714: I can find the realsense device, but the page of the camera does not appear

CHMimilanlan, 2022-07-24:
| | |
|---|---|
|**librealsense**|2.50.0 RELEASE|
|**OS**|Linux|

As i activate realsense-viewer, the page of camera,which is supposed to automatically appear, does not appear, and the terminal write these down when i manage to reinsert the usb of realsense camera. 
23:05:13 [Info] /home/magik/下载/librealsense-2.50.0/src/context.cpp:382 - Found 1 RealSense devices (mask 0xff)

Do that mean my realsense camera is detected?  But the graphical interface does not appear, which confused me for a whole day.

MartyG-RealSense, 2022-07-24:
Hi @CHMimilanlan  Yes, **Found 1 RealSense devices (mask 0xff)** indicates that the camera has been detected.  The log says 'Found 1 RealSense devices' if one camera is attached when the Viewer is launched, 'Found 2' if two cameras are attached, etc.

Do the details of the camera appear in the side-panel of the Viewer or does the side-panel show "Connect a RealSense camera or Add Source"?

[image]

[image]

MartyG-RealSense, 2022-07-31:
Hi @CHMimilanlan  Do you require further assistance with this case, please?  Thanks!

CHMimilanlan, 2022-07-31:
> Hi @CHMimilanlan Do you require further assistance with this case, please? Thanks!

I forget to check my message, thanks for your help! I have solved this issue, the reason is my USB interface is not firm.
Sorry to bother

MartyG-RealSense, 2022-07-31:
It's no trouble at all - I'm pleased to hear that you found a solution.  Thanks very much for the update!  As you have resolved the issue, I will close this case.  Thanks again!

closed 2022-07-31 by MartyG-RealSense

rosedaleisaac, 2022-08-28:
> > Hi @CHMimilanlan Do you require further assistance with this case, please? Thanks!
> 
> I forget to check my message, thanks for your help! I have solved this issue, the reason is my USB interface is not firm. Sorry to bother

I am also having this issue. I have tried changing my USB ports and USB hubs, and it still does not work. What steps did you take to improve your USB interface?

CHMimilanlan, 2022-08-29:
> I am also having this issue. I have tried changing my USB ports and USB hubs, and it still does not work. What steps did you take to improve your USB interface?

At first I changed my USB ports, but it not work, which was same situation like you. After that I tried USB hubs, and it work. Make sure your USB ports have version of 3.0. You can tried to reinstall realsense-viewer, if that not work, The problem is possibly refered to your hardware, instead of software.