realsenseai/librealsense #10715: undefined reference

Yoki-pjx, 2022-07-24:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D455 } | 
| Firmware Version                   | (2.50) | 
| Operating System & Version |   Linux   | 
| Kernel Version (Linux Only)    |  Linux raspberrypi 5.10.103-v8+ #1529 SMP PREEMPT Tue Mar 8 12:26:46 GMT 2022 aarch64 GNU/Linux | 
| Platform                                 | Raspberry Pi 4B|
| SDK Version                            |  { legacy / 2.5.0 }                          | 
| Language                            |  {C++ }                          | 
| Segment			|               | 

### Issue Description
Hello, I am a freshman who is using the real sense camera for the first time. I have followed the [guide of installation for Raspberry Pi 4](https://github.com/datasith/Ai_Demos_RPi/wiki/Raspberry-Pi-4-and-Intel-RealSense-D435) to install the real sense library. My camera works with the realsense viwer and also works with the python example codes. However, it fails when I try to run the C++ example codes. It occurs errors as follow:

> /usr/bin/ld: CMakeFiles/rs-hello-realsense.dir/rs-hello-realsense.cpp.o: in function `rs2::error::error(rs2_error*)':
rs-hello-realsense.cpp:(.text._ZN3rs25errorC2EP9rs2_error[_ZN3rs25errorC5EP9rs2_error]+0x20): undefined reference to `rs2_get_error_message'
/usr/bin/ld: rs-hello-realsense.cpp:(.text._ZN3rs25errorC2EP9rs2_error[_ZN3rs25errorC5EP9rs2_error]+0x64): undefined reference to `rs2_get_failed_function'
/usr/bin/ld: rs-hello-realsense.cpp:(.text._ZN3rs25errorC2EP9rs2_error[_ZN3rs25errorC5EP9rs2_error]+0x78): undefined reference to `rs2_get_failed_function'
/usr/bin/ld: rs-hello-realsense.cpp:(.text._ZN3rs25errorC2EP9rs2_error[_ZN3rs25errorC5EP9rs2_error]+0xe0): undefined reference to `rs2_get_failed_args'
/usr/bin/ld: rs-hello-realsense.cpp:(.text._ZN3rs25errorC2EP9rs2_error[_ZN3rs25errorC5EP9rs2_error]+0xf4): undefined reference to `rs2_get_failed_args'
/usr/bin/ld: rs-hello-realsense.cpp:(.text._ZN3rs25errorC2EP9rs2_error[_ZN3rs25errorC5EP9rs2_error]+0x154): undefined reference to `rs2_get_librealsense_exception_type'
/usr/bin/ld: rs-hello-realsense.cpp:(.text._ZN3rs25errorC2EP9rs2_error[_ZN3rs25errorC5EP9rs2_error]+0x168): undefined reference to `rs2_free_error'
/usr/bin/ld: CMakeFiles/rs-hello-realsense.dir/rs-hello-realsense.cpp.o: in function `rs2::error::handle(rs2_error*)':
rs-hello-realsense.cpp:(.text._ZN3rs25error6handleEP9rs2_error[_ZN3rs25error6handleEP9rs2_error]+0x20): undefined reference to `rs2_get_librealsense_exception_type'
/usr/bin/ld: CMakeFiles/rs-hello-realsense.dir/rs-hello-realsense.cpp.o: in function `rs2::stream_profile::stream_profile(rs2_stream_profile const*)':
rs-hello-realsense.cpp:(.text._ZN3rs214stream_profileC2EPK18rs2_stream_profile[_ZN3rs214stream_profileC5EPK18rs2_stream_profile]+0xa0): undefined reference to `rs2_get_stream_profile_data'
/usr/bin/ld: rs-hello-realsense.cpp:(.text._ZN3rs214stream_profileC2EPK18rs2_stream_profile[_ZN3rs214stream_profileC5EPK18rs2_stream_profile]+0xb8): undefined reference to `rs2_is_stream_profile_default'
/usr/bin/ld: CMakeFiles/rs-hello-realsense.dir/rs-hello-realsense.cpp.o: in function `rs2::frame::~frame()':
rs-hello-realsense.cpp:(.text._ZN3rs25frameD2Ev[_ZN3rs25frameD5Ev]+0x24): undefined reference to `rs2_release_frame'
/usr/bin/ld: CMakeFiles/rs-hello-realsense.dir/rs-hello-realsense.cpp.o: in function `rs2::frame::get_profile() const':
rs-hello-realsense.cpp:(.text._ZNK3rs25frame11get_profileEv[_ZNK3rs25frame11get_profileEv]+0x24): undefined reference to `rs2_get_frame_stream_profile'
/usr/bin/ld: CMakeFiles/rs-hello-realsense.dir/rs-hello-realsense.cpp.o: in function `rs2::frame::add_ref() const':
rs-hello-realsense.cpp:(.text._ZNK3rs25frame7add_refEv[_ZNK3rs25frame7add_refEv]+0x1c): undefined reference to `rs2_frame_add_ref'
/usr/bin/ld: CMakeFiles/rs-hello-realsense.dir/rs-hello-realsense.cpp.o: in function `rs2::frame::reset()':
rs-hello-realsense.cpp:(.text._ZN3rs25frame5resetEv[_ZN3rs25frame5resetEv]+0x24): undefined reference to `rs2_release_frame'
/usr/bin/ld: CMakeFiles/rs-hello-realsense.dir/rs-hello-realsense.cpp.o: in function `rs2::video_frame::video_frame(rs2::frame const&)':
rs-hello-realsense.cpp:(.text._ZN3rs211video_frameC2ERKNS_5frameE[_ZN3rs211video_frameC5ERKNS_5frameE]+0x5c): undefined reference to `rs2_is_frame_extendable_to'
/usr/bin/ld: CMakeFiles/rs-hello-realsense.dir/rs-hello-realsense.cpp.o: in function `rs2::video_frame::get_width() const':
rs-hello-realsense.cpp:(.text._ZNK3rs211video_frame9get_widthEv[_ZNK3rs211video_frame9get_widthEv]+0x28): undefined reference to `rs2_get_frame_width'
/usr/bin/ld: CMakeFiles/rs-hello-realsense.dir/rs-hello-realsense.cpp.o: in function `rs2::video_frame::get_height() const':
rs-hello-realsense.cpp:(.text._ZNK3rs211video_frame10get_heightEv[_ZNK3rs211video_frame10get_heightEv]+0x28): undefined reference to `rs2_get_frame_height'
/usr/bin/ld: CMakeFiles/rs-hello-realsense.dir/rs-hello-realsense.cpp.o: in function `rs2::depth_frame::depth_frame(rs2::frame const&)':
rs-hello-realsense.cpp:(.text._ZN3rs211depth_frameC2ERKNS_5frameE[_ZN3rs211depth_frameC5ERKNS_5frameE]+0x5c): undefined reference to `rs2_is_frame_extendable_to'
/usr/bin/ld: CMakeFiles/rs-hello-realsense.dir/rs-hello-realsense.cpp.o: in function `rs2::depth_frame::get_distance(int, int) const':
rs-hello-realsense.cpp:(.text._ZNK3rs211depth_frame12get_distanceEii[_ZNK3rs211depth_frame12get_distanceEii]+0x38): undefined reference to `rs2_depth_frame_get_distance'
/usr/bin/ld: CMakeFiles/rs-hello-realsense.dir/rs-hello-realsense.cpp.o: in function `rs2::frameset::frameset(rs2::frame const&)':
rs-hello-realsense.cpp:(.text._ZN3rs28framesetC2ERKNS_5frameE[_ZN3rs28framesetC5ERKNS_5frameE]+0x64): undefined reference to `rs2_is_frame_extendable_to'
/usr/bin/ld: rs-hello-realsense.cpp:(.text._ZN3rs28framesetC2ERKNS_5frameE[_ZN3rs28framesetC5ERKNS_5frameE]+0xd4): undefined reference to `rs2_embedded_frames_count'
/usr/bin/ld: CMakeFiles/rs-hello-realsense.dir/rs-hello-realsense.cpp.o: in function `rs2::context::context()':
rs-hello-realsense.cpp:(.text._ZN3rs27contextC2Ev[_ZN3rs27contextC5Ev]+0x2c): undefined reference to `rs2_create_context'
/usr/bin/ld: rs-hello-realsense.cpp:(.text._ZN3rs27contextC2Ev[_ZN3rs27contextC5Ev]+0x38): undefined reference to `rs2_delete_context'
/usr/bin/ld: rs-hello-realsense.cpp:(.text._ZN3rs27contextC2Ev[_ZN3rs27contextC5Ev]+0x3c): undefined reference to `rs2_delete_context'
/usr/bin/ld: CMakeFiles/rs-hello-realsense.dir/rs-hello-realsense.cpp.o: in function `rs2::pipeline::pipeline(rs2::context)':
rs-hello-realsense.cpp:(.text._ZN3rs28pipelineC2ENS_7contextE[_ZN3rs28pipelineC5ENS_7contextE]+0x3c): undefined reference to `rs2_create_pipeline'
/usr/bin/ld: rs-hello-realsense.cpp:(.text._ZN3rs28pipelineC2ENS_7contextE[_ZN3rs28pipelineC5ENS_7contextE]+0x48): undefined reference to `rs2_delete_pipeline'
/usr/bin/ld: rs-hello-realsense.cpp:(.text._ZN3rs28pipelineC2ENS_7contextE[_ZN3rs28pipelineC5ENS_7contextE]+0x4c): undefined reference to `rs2_delete_pipeline'
/usr/bin/ld: CMakeFiles/rs-hello-realsense.dir/rs-hello-realsense.cpp.o: in function `rs2::pipeline::start()':
rs-hello-realsense.cpp:(.text._ZN3rs28pipeline5startEv[_ZN3rs28pipeline5startEv]+0x30): undefined reference to `rs2_pipeline_start'
/usr/bin/ld: rs-hello-realsense.cpp:(.text._ZN3rs28pipeline5startEv[_ZN3rs28pipeline5startEv]+0x3c): undefined reference to `rs2_delete_pipeline_profile'
/usr/bin/ld: rs-hello-realsense.cpp:(.text._ZN3rs28pipeline5startEv[_ZN3rs28pipeline5startEv]+0x40): undefined reference to `rs2_delete_pipeline_profile'
/usr/bin/ld: CMakeFiles/rs-hello-realsense.dir/rs-hello-realsense.cpp.o: in function `rs2::pipeline::wait_for_frames(unsigned int) const':
rs-hello-realsense.cpp:(.text._ZNK3rs28pipeline15wait_for_framesEj[_ZNK3rs28pipeline15wait_for_framesEj]+0x38): undefined reference to `rs2_pipeline_wait_for_frames'
/usr/bin/ld: CMakeFiles/rs-hello-realsense.dir/rs-hello-realsense.cpp.o: in function `void rs2::frameset::foreach_rs<rs2::frameset::first_or_default(rs2_stream, rs2_format) const::{lambda(rs2::frame)#1}>(rs2::frameset::first_or_default(rs2_stream, rs2_format) const::{lambda(rs2::frame)#1}) const':
[... cut ...]
set_target_properties (rs-hello-realsense PROPERTIES FOLDER "Examples")

install(TARGETS rs-hello-realsense RUNTIME DESTINATION ${CMAKE_INSTALL_BINDIR})

Under my understanding, I got errors because the cmake file did not find the rs2 library. I am not sure if my operation is collect. I did `cmake .` and `make`, then got errors. I find the executable files in /usr/local/bin, and they works. Sorry for raise a question as I am a freshman with Cmake, I have reinstalled it three times but it still doesn't work.

MartyG-RealSense, 2022-07-24:
Hi @Yoki-pjx  The set of official RealSense SDK example programs such as **rs-hello-realsense** are built differently from the method used to compile your own self-created scripts.  The SDK builds all of the examples instead of individual ones.  If one of the SDK examples is subsequently modified and the set of examples is built again, CMake only updates the programs that have been modified instead of building all of the programs again.  This is discussed at https://github.com/IntelRealSense/librealsense/issues/6797

When copying the example project code into your own script, there will therefore be dependencies that are unable to be found when the script is built. 

I recommend studying the **Getting Started** C++ tutorial project at the link below.  It has the same script as rs-hello-realsense but a more complex CMakeLists.txt file that enables the librealsense library to be found by the project when it is built.

https://github.com/zivsha/librealsense/tree/getting_started_example/examples/getting-started

Yoki-pjx, 2022-07-26:
Hi @MartyG-RealSense,

Thank you for replying. The getting started tutorial project helps a lot. I get some problem when I try to capture two images for color and depth as in the rs-capture example. It is different from the getting started tutorial project, it requires some functions for the example.hpp which also links to other head files like > third-party/stb_easy_font.h

The file structure layouts are as follow:
|-- build
|-- capture
|   |-- CMakeLists.txt
|   |-- example.hpp
|   |-- rs-capture.cpp
|-- CMakeLists.txt

I imitated and wrote two CMakeLists.txt as follows:
**./CMakeLists.txt**
```
#  minimum required cmake version: 3.1.0
cmake_minimum_required(VERSION 3.1.0)

project(RealsenseCapture)

# Save the command line compile commands in the build output
set(CMAKE_EXPORT_COMPILE_COMMANDS 1)

# View the makefile commands during build
#set(CMAKE_VERBOSE_MAKEFILE on)

include(CheckCXXCompilerFlag)
CHECK_CXX_COMPILER_FLAG("-std=c++11" COMPILER_SUPPORTS_CXX11)
CHECK_CXX_COMPILER_FLAG("-std=c++0x" COMPILER_SUPPORTS_CXX0X)
if(COMPILER_SUPPORTS_CXX11)
    set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11")
elseif(COMPILER_SUPPORTS_CXX0X)
    set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++0x")
endif()

# Simple non robust way to find the librealsense library
find_library(REALSENSE2_FOUND realsense2 HINTS ${LIBRARY_DIR} REQUIRED)
if(NOT REALSENSE2_FOUND)
    SET(REALSENSE2_FOUND "realsense2")
    message(WARN "Failed to find_library(realsense2)")
endif()

if(BUILD_GRAPHICAL_EXAMPLES)
    find_package(OpenGL)
    if(NOT OPENGL_FOUND)
        message(FATAL_ERROR "\n\n OpenGL package is missing!\n\n")
    endif()

    set(DEPENDENCIES realsense2 ${OPENGL_LIBRARIES})

    if(WIN32)
        list(APPEND DEPENDENCIES glfw3)
    else()
        # Find glfw header
        find_path(GLFW_INCLUDE_DIR NAMES GLFW/glfw3.h
            PATHS /usr/X11R6/include
                  /usr/include/X11
                  /opt/graphics/OpenGL/include
                  /opt/graphics/OpenGL/contrib/libglfw
                  /usr/local/include
                  /usr/include/GL
                  /usr/include
        )
        # Find glfw library
        find_library(GLFW_LIBRARIES NAMES glfw glfw3
                PATHS /usr/lib64
                      /usr/lib
                      /usr/lib/${CMAKE_LIBRARY_ARCHITECTURE}
                      /usr/local/lib64
                      /usr/local/lib
                      /usr/local/lib/${CMAKE_LIBRARY_ARCHITECTURE}
                      /usr/X11R6/lib
        )
        list(APPEND DEPENDENCIES m ${GLFW_LIBRARIES} ${LIBUSB1_LIBRARIES})
        include_directories(${GLFW_INCLUDE_DIR})
    endif()
else()
    set(DEPENDENCIES realsense2)
    if(NOT WIN32)
        list(APPEND DEPENDENCIES m ${LIBUSB1_LIBRARIES})
    endif()
endif()

add_subdirectory(capture)
```
**./capture/CMakeLists.txt**
```
#  minimum required cmake version: 3.1.0
cmake_minimum_required(VERSION 3.1.0)

project(RealsenseExamplesCapture)

# Save the command line compile commands in the build output
set(CMAKE_EXPORT_COMPILE_COMMANDS 1)

# Simple non robust way to find the librealsense library
find_library(REALSENSE2_FOUND realsense2 HINTS ${LIBRARY_DIR} REQUIRED)
if(NOT REALSENSE2_FOUND)
    SET(REALSENSE2_FOUND "realsense2")
    message(WARN "Failed to find_library(realsense2)")
endif()

#Rs-capture
add_executable(rs-capture rs-capture.cpp example.hpp)
target_link_libraries(rs-capture ${ADDITIONAL_INCLUDE_DIRS})
include_directories(rs-capture ${REALSENSE2_FOUND} "/home/pi/librealsense/third-party")
set_target_properties (rs-capture PROPERTIES FOLDER Examples)

install(
    TARGETS

    rs-capture

    RUNTIME DESTINATION
    ${CMAKE_INSTALL_PREFIX}/bin
)
```
I put the example.hpp and the rs-capture.cpp in the same folder and include the third-party head files in the CMakeLists.txt as 
`include_directories(rs-capture ${REALSENSE2_FOUND} "/home/pi/librealsense/third-party")`

However, I still got errors after
```
mkdir build && cd build
cmake ..
make
```
[image]

Could you help to point out where I'm going wrong with it? Or any other tutorials that call the functions?

MartyG-RealSense, 2022-07-26:
The CMakeLists.txt file for the rs-capture example can be found here:

https://github.com/IntelRealSense/librealsense/blob/master/examples/capture/CMakeLists.txt

Your CMakeLists.txt file seems to be missing some of the instructions for rs-capture that are in the above file.

closed 2022-08-04 by Yoki-pjx

MartyG-RealSense, 2022-08-04:
You're very welcome @Yoki-pjx and I am pleased that I could help.  Thanks for the update!