realsenseai/librealsense #10717: Can‘t read IMU data with the latest Firmware version

knight-arturia, 2022-07-25:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 5.12.11 - 5.13 | 
| Operating System & Version |  Win 10 | 
| Platform                                 | PC  |
| SDK Version                           |  2.50                    | 
| Language                               |  python  3.8.6                     |  

### Issue Description
Hi, I was trying to get accel and gyro data from a D435i camera. The enable stream function looks like the below.
```
imu_config.enable_stream(rs.stream.accel, rs.format.motion_xyz32f, 250)
imu_config.enable_stream(rs.stream.gyro, rs.format.motion_xyz32f, 200)
```
With the latest package version Pyrealsense 2.50, when I run with a Firmware version later than 5.12.10, it will always call a ```RuntimeError: Couldn't resolve requests``` at line ```profig = pipeline.start(imu_config)```.  But it works well on the Firmware version before 5.12.11. And this problem also happed on older version of Pyrealsense

Did you guys update the way to get IMU data in the latest Firmware?

MartyG-RealSense, 2022-07-25:
Hi @knight-arturia  There have not been changes to how IMU data is retrieved.

May I first confirm that in the line before your 'imu_config.enable_stream' instructions, you defined the corret config name:

**imu_config = rs.config()**

Please also confirm whether or not the problem is in the imu_config instructions by removing imu_config from the pipeline start line's brackets so that the script ignores those instructions and uses the camera's default configuration.

**profig = pipeline.start()**

knight-arturia, 2022-07-25:
Hi, @MartyG-RealSense  Thanks for your quick answer!

The whole structure looks like this:
```
# Configure streams
if enable_imu:
    imu_pipeline = rs.pipeline()
    imu_config = rs.config()
    if None != device_id:
        imu_config.enable_device(device_id)
    imu_config.enable_stream(rs.stream.accel, rs.format.motion_xyz32f, 63) # acceleration
    imu_config.enable_stream(rs.stream.gyro, rs.format.motion_xyz32f, 200)  # gyroscope
    imu_profile = imu_pipeline.start(imu_config)
```
I just follow the code from other Issues. I think It was not a programming fault, since it works well on some early Firmware versions.

MartyG-RealSense, 2022-07-25:
The **if None != device_id** code is not typically used in config instructions.

```
if None != device_id:
imu_config.enable_device(device_id)
```

Does the script run if you comment those two lines out?

Please also test the Python IMU script at https://github.com/IntelRealSense/librealsense/issues/3409#issuecomment-471673994 to see whether it has the same **Couldn't resolve requests** error.

knight-arturia, 2022-07-26:
Hi, After trying I just found out the Problem is at Jupyter Notebook Kernel. If I run the code directly via ```python``` command there would not be any problems. XD
Thank you for your help! I think this Issue could be closed now.

MartyG-RealSense, 2022-07-26:
That's great news that you found the answer, @knight-arturia - thanks very much for the update   :)  As you suggested, I will close the issue.  Thanks again!

closed 2022-07-26 by MartyG-RealSense