realsenseai/librealsense #10718: T265: RuntimeError: No device connected

Micbut, 2022-07-25:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { t265 } | 

| Operating System & Version |   {Win (10) | 
| Platform                                 | PC |
| SDK Version                            |  {  2.5 }                          | 
| Language                            |  {python}                          | 
| Segment			|  {others }                          | 

### Issue Description
Hi,
I connected the T265 camera to my PC and the Intel Realsense Viewer detects the camera and allows the tracking to happen.
However, when working with python (through google Colab)  the code tells me that there is no device connected
[image]

Similarly, if I try to get the ID of the device, the command query_devices() doesn't show any device
[image]
.

patatessever, 2022-08-06:
It's about a pipeline error I assume.  When you use it on RealSense viewer you occupy the pipeline; thus, python does not see your device. Do not use RealSense viewer, unplug and plug the device and try your code again.
,

Micbut, 2022-08-16:
> 
Thank you for your answer - unfortunately the problem remains. I also forgot to say that I can also do the traking simply for the command prompt, without using the Viewer. And still python doesn't work

Micbut, 2022-08-31:
PROBLEM SOLVED: Google Colab doesn't have access to the USB connected on the local device. Working on anaconda/spyder solved the issue

closed 2022-08-31 by Micbut