realsenseai/librealsense #10719: Camera Orientation Angles on D435

dennewbie, 2022-07-25:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras). 
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)? 
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue. 

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.13.0.50 | 
| Operating System & Version |  MacOS  | 
| Platform                                 | PC  |
| SDK Version                            |  2.50.0 }                          | 
| Language                            |  C++, OpenCV                          | 

### Issue Description
Hi, I would like to get the camera orientation angle from D435. Output example:
[https://github.com/IntelRealSense/librealsense/issues/7907#issuecomment-737780812](GitHub Issue #7907)

 I know that since there is no IMU module, it's not possible to use the approach used here:
[https://github.com/IntelRealSense/librealsense/issues/7907](How to get rotation matrix or angles #7907)
[https://github.com/IntelRealSense/librealsense/issues/4391](D435i IMU problem #4391)

Furthermore, I don't need just an angle between the 2 planes. So I can't use the approach used here:
[https://github.com/IntelRealSense/librealsense/issues/4440](How does Depth Quality Tool know the angle without IMU #4440)

Is there any solution to achieve the output shown before?
Thanks in advance,
Denny

MartyG-RealSense, 2022-07-25:
Hi @dennewbie  If a RealSense T265 Tracking Camera was placed on a mount with your D435 then it could obtain the device orientation from the T265's **pose data** stream.  If the mount is moved and the T265 and D435 were both pointing in the same direction on the mount then the T265's orientation angle would also be the D435's.

If only the D435 can be used then typically the recommendation would be to obtain the camera angles from a **plane fit** as the links that you quoted suggest.  If a plane fit is not suitable for your particular application then finding the angle between the camera and a pixel may be appropriate, as discussed at https://github.com/IntelRealSense/librealsense/issues/5553

dennewbie, 2022-07-26:
Perfect, thank you @MartyG-RealSense

closed 2022-07-26 by dennewbie