realsenseai/librealsense #10720: Calibration parameters conversion without Intel® RealSense™ D400 Series Dynamic Calibration Tool

rasmusgo, 2022-07-25:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400  | 
| Firmware Version                   | - | 
| Operating System & Version |  Ubuntu  | 
| Kernel Version (Linux Only)    |  -                                         | 
| Platform                                 | PC  |
| SDK Version                            |  -                          | 
| Language                            |  C++                          | 
| Segment			|  others                          | 

### Issue Description
Please add simple functions to convert between raw calibration data and calibration parameters into librealsense. These conversions can be done by using the [Intel® RealSense™ D400 Series Dynamic Calibration Tool](https://www.intel.com/content/www/us/en/download/645988/intel-realsense-d400-series-dynamic-calibration-tool.html?) but that introduces a lot of unnecessary complexity for our use case. It would be much preferable to avoid adding that dependency.

### Background
We currently have about 3000 foot scanners deployed globally, each running an Intel NUC with four SR300 cameras attached. We are working on transitioning from SR300 cameras to D415 cameras and see the need for calibrating the D415 cameras during production after the camera modules have been installed. We have a custom evaluation procedure that we have been using for SR300 cameras that we are modifying to use for calibrating the D415 cameras. Introducing a dependency on the dynamic calibration tool adds unwanted complexity especially as that tool is not available on github and comes with a separate license that needs consideration.

MartyG-RealSense, 2022-07-25:
Hi @rasmusgo  Thanks very much for your question!

RealSense has an alternative official set of calibration tools that were introduced in 2019.  The **On-Chip** (depth quality), **Tare** (depth ground-truth accuracy) and **Focal Length** calibration tools are built into the RealSense camera firmware driver and so do not require installation of a separate software tool lke the Dynaic Calibration tool does.  They can also be accessed with librealsense SDK programming instructions.

An Intel white-paper guide document for the calibration tools can be found at the link below.

https://dev.intelrealsense.com/docs/self-calibration-for-depth-cameras

Would these firmware-based calibration tools meet your needs, please?

rasmusgo, 2022-07-25:
Thank you for your fast response!

Unfortunately, they are not suitable for our application. I have tried to use them but we are operating the cameras at such a short distance that we need to use disparity shift. These tools do are not compatible with disparity shift from what I can tell.

MartyG-RealSense, 2022-07-25:
If you have not yet purchased the D415 cameras then it may be worth considering the new **D405** model if you are depth sensing at a range between 7 cm and 50 cm, so disparity shift would not be necessary.  The D405 is designed specially to provide high accuracy images at close range.

https://store.intelrealsense.com/buy-intel-realsense-depth-camera-d405.html

[image]

rasmusgo, 2022-07-25:
There are still some details hidden in the reserved and rect_params fields though:

```cpp
        struct coefficients_table
        {
            table_header        header;
            float3x3            intrinsic_left;             //  left camera intrinsic data, normilized
            float3x3            intrinsic_right;            //  right camera intrinsic data, normilized
            float3x3            world2left_rot;             //  the inverse rotation of the left camera
            float3x3            world2right_rot;            //  the inverse rotation of the right camera
            float               baseline;                   //  the baseline between the cameras in mm units
            uint32_t            brown_model;                //  Distortion model: 0 - DS distorion model, 1 - Brown model
            uint8_t             reserved1[88];
            float4              rect_params[max_ds5_rect_resolutions];
            uint8_t             reserved2[64];
        };
```

MartyG-RealSense, 2022-07-25:
I will consult with my Intel RealSense colleagues about your programming question.  Thanks very much for your patience!

rasmusgo, 2022-07-25:
> If you have not yet purchased the D415 cameras then it may be worth considering the new **D405** model if you are depth sensing at a range between 7 cm and 50 cm. Its lower minimum distance means that disparity shift would not be necessary. The D405 is designed to provide high accuracy images at close range.

It looks interesting but I think we need to have an infrared projector to get depth data from evenly colored regions.

MartyG-RealSense, 2022-07-25:
If an external projector is an option for your setup then a **visible pattern projector** should be able to provide texture for the D405 camera to analyze instead of an IR dot pattern, as described by Intel's projector white-paper in the section linked to below.

https://dev.intelrealsense.com/docs/projectors#6-the-consumer-front-projector

Alternatively, the **D435** has a minimum distance of only 0.1 meters, making it also suited to close-range sensing, and is equipped with a built-in projector.

MartyG-RealSense, 2022-07-26:
Hi again @rasmusgo  Because of the volume of camera units that you are seeking to upgrade, my Intel RealSense colleagues request that you open a **Zendesk** support ticket for your RealSense query.

MartyG-RealSense, 2022-08-06:
Hi @rasmusgo  Were you able to open a Zendesk support ticket, please?  Thanks!

MartyG-RealSense, 2022-08-14:
Hi @rasmusgo  Do you have an update about this case that you can provide, please?  Thanks!

rasmusgo, 2022-08-14:
Sorry for not responding. I have been in contact with support and they directed me back to the binaries in the dynamic calibration tool. I'm trying to make it work but it wont be pretty. It would be a lot better if I could at least have a static library version of the dynamic calibration API.

MartyG-RealSense, 2022-08-15:
Please make your request to support on the Zendesk for a static library version of the dynamic calibration API.  Thanks!

MartyG-RealSense, 2022-08-26:
Hi @rasmusgo  Were you able to make a request on Zendesk for a static library version of the calibration API, please?

MartyG-RealSense, 2022-09-01:
Hi @rasmusgo  Do you require further assistance with this case, please?  Thanks!

rasmusgo, 2022-09-01:
Hi @MartyG-RealSense, I asked for a statically linked version now, thanks for following up!

MartyG-RealSense, 2022-09-01:
You are very welcome!  As your issue will be handled on your private Zendesk ticket now, I will close this public GitHub case.  Thanks again!

closed 2022-09-01 by MartyG-RealSense