realsenseai/librealsense #10721: first frame numbers

AdoNunes, 2022-07-25:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D455                                                      |
| Firmware Version                   |  05.12.12.100                                          | 
| Operating System & Version |  Win 10                                                   | 
| Platform                                 | PC.                                                          |
| SDK Version                           |  2.44.0.3056                                            | 
| Language                              | python                                                    | 
| Segment			        |  others                                                    | 

### Issue Description
I use pyrealsense to capture the frame number to synchronize the bag file to other data streams.  However, the first frame numbers are not correct. I see that the bag timestamps are off by .3 seconds and I wonder if the cause is the first frame numbers.

Below is the method to capture the frame numbers after starting the pipeline:
```
    @catch_exception
    def record(self):
        self.recording = True
        self.frame_counter = 0
        self.pipeline.start(self.config)
        
        # Avoid autoexposure frame drops
        dev = self.pipeline.get_active_profile().get_device()
        sens = dev.first_color_sensor()
        sens.set_option(rs.option.auto_exposure_priority, 0.0)
        
        while self.recording:
            frame = self.pipeline.wait_for_frames()
            self.n = frame.get_frame_number()
            self.outlet.push_sample([self.frame_counter, self.n])

            self.frame_counter += 1

        self.pipeline.stop()
```

Here are the captured frame numbers for three intel D455 cameras, each column is a camera:
[image]
As can be seen in camera 2, the first frame number is 15, the second is 1. For camera 3, the first 13 frames are 15, then starts from 1 and increments one frame. However, when reading the .bag file, the first frame is 1.

Do you know the reason for this behavior?

MartyG-RealSense, 2022-07-26:
Hi @AdoNunes  Please try changing wait_for_frames() to **poll_for_frames()** 

Does this change improve the results, please?  poll_for_frames is guaranteed to provide each new frame only once to avoid duplicates.

AdoNunes, 2022-07-27:
Hi @MartyG-RealSense 
I tried with poll_for_frames() however the problem persists, as you can see in the screenshot below, the first frame numbers are the same, after frame number 0 the increment is +1:
[image]
Why pyrealsense gives the same frame number at the beginning? is it related to the start of the stream recording?
I tried recording only from on camera but it makes no difference.

MartyG-RealSense, 2022-07-28:
At https://github.com/IntelRealSense/librealsense/issues/8949 there was a case of a RealSense Python user whose script with a frame-count printing mechanism worked in SDK version 2.35.0 and then they upgraded to 2.44.0 (the version that you are using) and it no longer worked correctly.  A RealSense team member tested the user's script in SDK 2.49.0 and found that the problem no longer occurred in that SDK version.

MartyG-RealSense, 2022-08-03:
Hi @AdoNunes  Was the advice in the comment above helpful to you, please?  Thanks!

AdoNunes, 2022-08-03:
I upgraded the SDK version to 2.50, which let me upgrade the intel firmware to 05.12.00.50.  However, I still get initial frames greater than one. 
The screenshot below each column is the obtained frame number for each of the three intel cameras.
[image]

AdoNunes, 2022-08-03:
I managed to improve synchronization by getting the frame timestamp during the recording and when reading the bag file I get the bag frame timestamp and match it to the closest frame timestamp obtained while recording. 

[image]
In the image, the first column is the loop index number (number of times frame number was requested since the start of recording), the second is the API frame number, the third is its associated timestamp.  It seems that the timestamp is correct and it can be used to map recording frames and frames from bag file.

MartyG-RealSense, 2022-08-03:
That's awesome that you made such significant progress.  Congratulations!  :)

closed 2022-08-21 by MartyG-RealSense