realsenseai/librealsense #10722: Does this library support jetson orin？

zwjgit, 2022-07-26:
Does this library support jetson orin？

MartyG-RealSense, 2022-07-26:
Hi @zwjgit  There have not yet been reports of use of the librealsense SDK with the new Jetson Orin at the time of writing this.  Typically though with Jetson boards, the same librealsense installation process can work with multiple Jetson models.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

MartyG-RealSense, 2022-08-01:
Hi @zwjgit  Do you require further assistance with this case, please?  Thanks!

zwjgit, 2022-08-01:
> Hi @zwjgit Do you require further assistance with this case, please? Thanks!

No~

zwjgit, 2022-08-01:
> Hi @zwjgit Do you require further assistance with this case, please? Thanks!

No,Maybe you can try source code compilation? There is no other good solution~

MartyG-RealSense, 2022-08-01:
There is a guide at https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049for building from I source code on Jetson with the RSUSB backend installation method.

https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049

MartyG-RealSense, 2022-08-07:
Case closed due to no further assistance required.

closed 2022-08-07 by MartyG-RealSense

Yadunund, 2022-08-24:
Hi @MartyG-RealSense 

When following the Jetson installation instructions you linked above for the Orin or an AGX with the latest jetpack version, I am unable to successfully run the `./scripts/patch-realsense-ubuntu-L4T.sh ` script as it throws an `Unsupported JetPack revision 34.1.1 aborting script` error.
The kernel version used on these systems is `5.10.65-tegra`

Are you able to comment on whether
1) The kernel patch is still need for these jetson devices
2) If so, when can we expect the path to work?

Many thanks for your time with answering these questions.

MartyG-RealSense, 2022-08-24:
The most recent officially supported JetPack version in the RealSense SDK is 4.6 (L4T 32.6.1), as described in the line of the L4T script linked to below.

https://github.com/IntelRealSense/librealsense/blob/master/scripts/patch-realsense-ubuntu-L4T.sh#L60  

The 34.1.1 version mentioned by the error message is JetPack 5.  Whilst support for newer JetPack versions does get added over time, I do not have a time estimate for the next update of supported JetPacks.

You would not have to apply a patch script if building from source code with the **RSUSB backend** method.  The Jetson instruction page recommends building librealsense from source code with the **libuvc_installation.sh** build script, like the 'libuvc backend' procedure described at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

iraadit, 2022-08-29:
Hi @Yadunund 

As an info, I successfully installed librealsense on Jetpack 5 using the RSUSB backend method.

Hoping for an update of the kernel method soon too :)

MartyG-RealSense, 2022-08-29:
Thanks very much @iraadit for your feedback to @Yadunund  :)

Yadunund, 2022-08-29:
Thanks a lot @MartyG-RealSense and @iraadit for this valuable information.

I was able to install `librealsense` on the Orin with both approaches, ie 1) Following the native backend instructions but skipping the kernel patch step and also 2) Following the libuvc instructions. I guess it wasn't clear if the patch was an essential step. 

fyi, I've also tested the `realsense-ros` driver (which links to this lib) on ROS 2 `humble` and I'm able to stream all the sensor data.

MartyG-RealSense, 2022-08-29:
That's great news, @Yadunund - thanks for the confirmation of your success with Orin with both librealsense and ROS2 Humble!

iraadit, 2022-08-29:
Hi @Yadunund and @MartyG-RealSense 

I've been able to install following method 1 without kernel patching too.