realsenseai/librealsense #10725: Extracting depth information

L-xn, 2022-07-27:
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (/10) / Linux (Ubuntu 22  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python }                          | 

    I've been doing keystroke tracking.I'm working on a two-dimensional keypoint on the depth map.I now want to get the corresponding Z-axis distance information.But I find it hard to match.
    I recorded a 5 minute 5 second video and put it in the BAG at 30 frames.I got 2239 depth maps and 2238 color maps using the RS-convert tool.I read the bag through Python and got 4460 depth maps and 4460 color maps.
    I didn't use rs-convert because the depth map converted by this tool is the original image, whereas I can convert the filtered depth map using Python.I was wondering if I could use Python to parse the bag to get a depth file, which happens to be 4460 files, so I could match the depth map to the depth file and get the 3D coordinates.
    The following two images are the original depth map transformed by RS-convert and the depth map processed by Python filtering.
[image]
[image]

MartyG-RealSense, 2022-07-27:
Hi @L-xn  Are you aiming to use Python to extract a separate file for each of the 4460 depth frames in the bag?  If you are then using the SDK instruction **save_single_frameset()** may work for you.  As described at https://github.com/IntelRealSense/librealsense/issues/2588#issuecomment-453757755 - if you use your original bag file as the data source for the script instead of a live camera, save_single_frameset should extract a new bag file for each individual frameset.

A RealSense user shared a Python save_single_frameset() script at the link below.

https://github.com/soarwing52/RealsensePython/blob/master/separate%20functions/single_frameset.py

L-xn, 2022-07-27:
Thanks for your quick reply!I don't have a need for that yet.What I want to do now is put the depth information into CSV via Python.

MartyG-RealSense, 2022-07-27:
There is not a reference available for saving depth to csv using Python (there are example of C++ than can do so).  For Python it may be better to save the depth to an **npy** file as an array of scaled matrices instead, like the script referred to at https://github.com/IntelRealSense/librealsense/issues/9563#issuecomment-892013926

L-xn, 2022-08-01:
I tried to save the depth information as NPY by Pyhton, but failed because the NPY array was too large and the computer memory was insufficient.
[image]
I would like to ask why the picture extracted from BAG by RS-convert has a serious frame loss.

MartyG-RealSense, 2022-08-01:
The performance of rs-convert in regards to the number of frames lost tends to depend on the particular computer being used, so there is not a single set of advice that can be given about improving the tool's frame extraction.

closed 2022-08-20 by MartyG-RealSense