realsenseai/librealsense #10726: Unable to import pyrealsesnse2 from Intel RealSense SDK 2.0 windows 10

YadneshD, 2022-07-27:
I have installed Intel Realsense Viewer and SDK version - 2.50.0.3785. The installation included python and C/C++ API's. I am unable to import pyrealsense2 from py3.8 and py3.9 while it properly works with py3.6. In my Program Files (x86)\Intel RealSense SDK 2.0\bin\x64, I can get following pyd files - 

pyrealsense2.pyd
pyrealsense2.cp36-win_amd64.pyd

When I import pyrelasense2 from the bin/x64 folder in python 3.8 and 3.9, it gives me errors. While in python 3.6 it works well. Screenshot is attached -

-------------------------------------------------------------------------------------------------
| Operating System & Version -   Windows 10
| SDK Version                            - 2                         
| Language                          -   python 3.x

[image]

MartyG-RealSense, 2022-07-27:
Hi @YadneshD  The number in a pyrealsense2 wrapper file name, such as '36', indicates which Python 3 version that file was built for.  '36' is Python 3.6, and Python 3.8 and 3.9 would have filenames with '38' and '39' in them.

The 'pyrealsense2.pyd' file in the bin folder is for use with Python 2.7, whilst Python 3 pyd files have the more complex names based on the particular Python 3 version that is installed on the computer.

Whilst the Windows version of the RealSense SDK only installs pyd files for Python 3.6, it is possible to install the Python wrapper for 3.8 and 3.9 on Windows by performing a 'pip install' from PyPi packages with the instruction **pip install pyrealsense2**

https://pypi.org/project/pyrealsense2/#files

At https://github.com/IntelRealSense/librealsense/issues/6402#issuecomment-629763906 I wrote a guide to performing a pip install of pyrealsense2 on Windows that may be a useful reference.

Alternatively, you can build the pyrealsense2 wrapper from source code for modern Python versions, including ones not supported currently by pip packages such as Python 3.10.  The wrapper can be built separately from source code or by building the Windows version of librealsense from source code with CMake and including the **BUILD_PYTHON_BINDINGS** build flag for the Python wrapper.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md

YadneshD, 2022-07-27:
@MartyG-RealSense 
Thanks a lot for your explanation. I am closing this issue.

closed 2022-07-27 by YadneshD