realsenseai/librealsense #10727: d455 camera dropped while using，prompt no device connect

ssshello, 2022-07-27:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | d455 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  Linux tx2-nx 4.9.201-tegra             | 
| Platform                                 | NVIDIA Jetson tx2  |
| SDK Version                            |  2.50.0                          | 
| Language                            |  python                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I used an nvidia-docker to build a container to use the camera, but the camera drops during use, and the camera device cannot be queried, but the host can recognize the device.
[image]
The situation is that it happens during the use process. It can be used normally in the docker at the beginning, and the disconnection occurs after a period of time.
Running the python program directly in the host computer will also cause the camera to drop. It can be used normally at first, but after a period of time, it will prompt "No device connected", reboot does not work, reinserting the type-c header does not work, only reinserting usb plug to continue normal use.

MartyG-RealSense, 2022-07-27:
Hi @ssshello  It does not sound as though there is a problem with the camera hardware if it can run correctly for a period of time and can re-connect when reset with a USB unplug-replug when a connection drop occurs.

Does the camera's casing feel hot to the touch if you touch it a few minutes after you begin streaming from a cold start, please?

ssshello, 2022-07-28:
> 你好@ssshello 如果它可以正常运行一段时间并且在发生连接断开时使用USB拔出重新插入重置时可以重新连接，则听起来相机硬件没有问题。
> 
> 如果您在从冷启动开始流式传输几分钟后触摸相机外壳，摸起来会不会感觉很热，好吗？

Hello, the camera feels a little hot to the touch after using it for a while, but is there a way to make the host re-read the camera device without re-inserting the usb? Is there any python code or shell script available? or something like that?

ssshello, 2022-07-28:
> 
Hello, the camera feels a little hot to the touch after using it for a while, but is there a way to make the host re-read the camera device without re-inserting the usb? Is there any python code or shell script available? or something like that?

ssshello, 2022-07-28:
> Hi @ssshello It does not sound as though there is a problem with the camera hardware if it can run correctly for a period of time and can re-connect when reset with a USB unplug-replug when a [...]

Hello, the camera feels a little hot to the touch after using it for a while, but is there a way to make the host re-read the camera device without re-inserting the usb? Is there any python code or shell script available? or something like that?

MartyG-RealSense, 2022-07-28:
If the camera is taking a while to heat up instead of within the first few minutes then it is likely not experiencing overheating (which can occur due to a bad USB cable, a glitch on the USB port or overheating of a single-board computer that the camera is attached to).

A camera can be reset without an unplug-replug using the SDK instruction **hardware_reset()**.  A simple Python example of code to initiate this is below: 

```
ctx = rs.context()
devices = ctx.query_devices()
for dev in devices:
dev.hardware_reset()
```

If a specific camera in a set of multiple cameras needs to be reset then https://github.com/IntelRealSense/librealsense/issues/5428 describes how to perform this in Python using a camera serial number.

Another way to reset is to use a Linux **bash** script to reset the entire USB port instead of just the camera, like the script at https://github.com/IntelRealSense/librealsense/issues/8393

ssshello, 2022-07-29:
> If the camera is taking a while to heat up instead of within the first few minutes then it is likely not experiencing overheating (which can occur due to a bad USB cable, a glitch on the USB port or [...]
hi,@MartyG-RealSense ,thanks for the reply
Usually the result returned by query_devices() when the camera is offline is an empty list so hardware_reset() cannot be used, and the bash script in the referenced link doesn't seem to work for my device because calling the lspci command can't find the usb controller, I tried Similar commands such as:
echo 2-2.3 > /sys/bus/usb/drivers/usb/unbind
echo 2-2.3 > /sys/bus/usb/drivers/usb/bind
But this method doesn't work
Use the command `ls /sys/bus/usb/drivers/usb/` to see the device number that exists in usb3, such as 2-2.x(the serial number type of the usb3 device on the industrial computer I am using), but the lsusb command cannot query the device
The actual operation is to re-insert the camera so that lsusb can rediscover the camera. So I try to search for some bash command or code that can replace this operation.Because I think the camera is connected to the machine, but the machine does not re-read the camera device.

MartyG-RealSense, 2022-07-29:
You mentioned in the above comment that you are using an industrial computer.  That could be significant information, as the SDK may sometimes experience issues on an industrial computer with a more exotic hardware / software configuration than a typical consumer PC.  In that situation, building the SDK with the **RSUSB backend = true** installation method can resolve problems.

This is because an RSUSB-based build of the SDK bypasses the Linux kernel.  It is therefore not dependent on Linux versions or kernel versions and does not require patching.

Instructions for installing the SDK from source code in RSUSB = true mode on Jetson can be found at https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049

closed 2022-08-20 by MartyG-RealSense