realsenseai/librealsense #10728: Can you extract by measuring the distance to the object in front?

jiminiscat, 2022-07-27:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400. 455 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Linux (Ubuntu 18.04)}  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

After extracting the distance to the object (person) in front, we want to use the TensorFlow example, the human height estimation example. 
The TensorFlow human height estimation example has high accuracy at a 'specific distance', so we want to measure a person's height only at that distance.
Is there any code or example where I can extract the distance value of the object in front of the camera?

I want to extract distance measurements. And I want to measure the person height from that value.

MartyG-RealSense, 2022-07-27:
Hi @jiminiscat  You could perhaps apply a **Threshold Filter** to exclude depth data that is outside of a defined minimum and maximum depth range, like in the Python script at https://github.com/IntelRealSense/librealsense/issues/8170#issuecomment-761930301

A minimum distance could be defined by adding the line below to the script and changing '1' to the minimum distance in meters of your choice:

`threshold_filter.set_option(rs.option.min_distance, 1)`

Like in the linked-to script, the threshold filter could be inserted into the **Example 2**height estimation script at the line after the pipeline start instruction.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/tensorflow/example2%20-%20person%20height.py#L18

sosoyeon1125, 2022-07-28:
Is this the right way to do it? but there was an error ... ;(
Can you help me? thank you...

[image]

MartyG-RealSense, 2022-07-28:
In the original 'Example 2' script, **aligned_stream = rs.align(rs.stream.color)** and **point_cloud = rs.pointcloud()** are not within the Try / While True section but outside of it, just after the pipeline start instruction:

```
pipeline.start(config)

aligned_stream = rs.align(rs.stream.color) # alignment between color and depth
point_cloud = rs.pointcloud()
```

sosoyeon1125, 2022-08-01:
SyntaxError: invalid syntax 
-> When i corrected the error, there was a problem that was not in the example2.
    I don't know what to do next. 
    Adding ':' results in an error ... ;(

[image]
[Full code]
import pyrealsense2 as rs
import numpy as np
import cv2
import tensorflow as tf

W = 640
H = 480

# Configure depth and color streams
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, W, H, rs.format.z16, 30)
config.enable_stream(rs.stream.color, W, H, rs.format.bgr8, 30)

print("[INFO] start streaming...")

pipeline.start(config)
aligned_stream = rs.align(rs.stream.color) # alignment between color and depth
point_cloud = rs.pointcloud()

pc = rs.pointcloud()
points = rs.points()
pipe = rs.pipeline()

config = rs.config()
config.enable_stream(rs.stream.depth)

pipe.start(config)

threshold_filter = rs.threshold_filter()
threshold_filter.set_option(rs.option.max_distance, 1.6)
threshold_filter.set_option(rs.option.min_distance, 1.3)

try:
   frames = pipe.wait_for_frames()
   frames_filtered = threshold_filter.process(frames)

print("[INFO] loading model...")
PATH_TO_CKPT = r"frozen_inference_graph.pb"
# download model from: https://github.com/opencv/opencv/wiki/TensorFlow-Object-Detection-API#run-network-in-opencv

# Load the Tensorflow model into memory.
detection_graph = tf.Graph()
with detection_graph.as_default():
    od_graph_def = tf.compat.v1.GraphDef()
    with tf.compat.v1.gfile.GFile(PATH_TO_CKPT, 'rb') as fid:
        serialized_graph = fid.read()
        od_graph_def.ParseFromString(serialized_graph)
        tf.compat.v1.import_graph_def(od_graph_def, name='')
    sess = tf.compat.v1.Session(graph=detection_graph)

# Input tensor is the image
image_tensor = detection_graph.get_tensor_by_name('image_tensor:0')
# Output tensors are the detection boxes, scores, and classes
# Each box represents a part of the image where a particular object was detected
detection_boxes = detection_graph.get_tensor_by_name('detection_boxes:0')
# Each score represents level of confidence for each of the objects.
# The score is shown on the result image, together with the class label.
detection_scores = detection_graph.get_tensor_by_name('detection_scores:0')
detection_classes = detection_graph.get_tensor_by_name('detection_classes:0')
# Number of objects detected
num_detections = detection_graph.get_tensor_by_name('num_detections:0')
# code source of tensorflow model loading: https://www.geeksforgeeks.org/ml-training-image-classifier-using-tensorflow-object-detection-api/

while True:
    frames = pipeline.wait_for_frames()
    frames = aligned_stream.process(frames)
    depth_frame = frames.get_depth_frame()
    color_frame = frames.get_color_frame()
    points = point_cloud.calculate(depth_frame)
    verts = np.asanyarray(points.get_vertices()).view(np.float32).reshape(-1, W, 3)  # xyz

    # Convert images to numpy arrays
    color_image = np.asanyarray(color_frame.get_data())
    scaled_size = (int(W), int(H))
    # expand image dimensions to have shape: [1, None, None, 3]
    # i.e. a single-column array, where each item in the column has the pixel RGB value
    image_expanded = np.expand_dims(color_image, axis=0)
    # Perform the actual detection by running the model with the image as input
    (boxes, scores, classes, num) = sess.run([detection_boxes, detection_scores, detection_classes, num_detections],
                                             feed_dict={image_tensor: image_expanded})

    boxes = np.squeeze(boxes)
    classes = np.squeeze(classes).astype(np.int32)
    scores = np.squeeze(scores)

    print("[INFO] drawing bounding box on detected objects...")
    print("[INFO] each detected object has a unique color")

    for idx in range(int(num)):
        class_ = classes[idx]
        score = scores[idx]
        box = boxes[idx]
        print(" [DEBUG] class : ", class_, "idx : ", idx, "num : ", num)

        if score > 0.8 and class_ == 1: # 1 for human
            left = box[1] * W
            top = box[0] * H
            right = box[3] * W
            bottom = box[2] * H

            width = right - left
            height = bottom - top
            bbox = (int(left), int(top), int(width), int(height))
            p1 = (int(bbox[0]), int(bbox[1]))
            p2 = (int(bbox[0] + bbox[2]), int(bbox[1] + bbox[3]))
            # draw box
            cv2.rectangle(color_image, p1, p2, (255,0,0), 2, 1)

            # x,y,z of bounding box
            obj_points = verts[int(bbox[1]):int(bbox[1] + bbox[3]), int(bbox[0]):int(bbox[0] + bbox[2])].reshape(-1, 3)
            zs = obj_points[:, 2]

            z = np.median(zs)

            ys = obj_points[:, 1]
            ys = np.delete(ys, np.where(
                (zs < z - 1) | (zs > z + 1)))  # take only y for close z to prevent including background

            my = np.amin(ys, initial=1)
            My = np.amax(ys, initial=-1)

            height = (My - my)  # add next to rectangle print of height using cv library
            height = float("{:.2f}".format(height))
            print("[INFO] object height is: ", height, "[m]")
            height_txt = str(height) + "[m]"

            # Write some Text
            font = cv2.FONT_HERSHEY_SIMPLEX
            bottomLeftCornerOfText = (p1[0], p1[1] + 20)
            fontScale = 1
            fontColor = (255, 255, 255)
            lineType = 2
            cv2.putText(color_image, height_txt,
                        bottomLeftCornerOfText,
                        font,
                        fontScale,
                        fontColor,
                        lineType)

    # Show images
    cv2.namedWindow('RealSense', cv2.WINDOW_AUTOSIZE)
    cv2.imshow('RealSense', color_image)
    cv2.waitKey(1)

# Stop streaming
pipeline.stop()

sosoyeon1125, 2022-08-01:
try:
frames = pipe.wait_for_frames()
frames_filtered = threshold_filter.process(frames)
<- It seems to be the part where the error occurs.... 
    used 'try:',
    'except'<- need?

sosoyeon1125, 2022-08-01:
This code add
[image]

New error...
Traceback (most recent call last):
  File "try.py", line 29, in <module>
    pipe.start(config)
RuntimeError: xioctl(VIDIOC_S_FMT) failed Last Error: Device or resource busy

[image]

MartyG-RealSense, 2022-08-01:
At the top of the script in the above image you have a pipeline.start(config) instruction and then the same line further down after config.enable_stream.  You cannot start a pipeline that has already been started as it will be busy.  Once the pipeline has been started then you have to stop() the pipeline first before you can use the start instruction again.

sosoyeon1125, 2022-08-07:
1. error
[image]

MartyG-RealSense, 2022-08-07:
The error is at line 71 of try.py and relates to the instruction **frames = pipeline.wait_for_frames()**

It indicate that this instruction was used whilst the pipeline was stopped and the pipeline has to have been started before it can work.

It looks as though line 71 is further down the script than the section of code that is shown in the image above.

sosoyeon1125, 2022-08-07:
It indicate that this instruction was used whilst the pipeline was stopped and the pipeline has to have been started before it can work. -> isn't this the pipe.start ??
[image]

MartyG-RealSense, 2022-08-07:
The highlighted line is the pipe start.  I cannot see the entire script though (the image has around the first 40 lines and error is at line 71) so there may a Stop() further down the script.  Could you show me more of the script below the image's bottom line please?

sosoyeon1125, 2022-08-07:
All code

[image]
[image]
[image]

MartyG-RealSense, 2022-08-07:
I think it is because you are using two pipeline definitions, 'pipeline' and 'pipe'.  You called Stop() on pipeline and then started pipe.

On line 71, you use **frames = pipeline.wait_for_frames()**.  But you stopped 'pipeline' and started 'pipe'.  So I believe that you should change the wait_for_frames instruction to reference the pipeline that is currently open.

**frames = pipe.wait_for_frames()**

sosoyeon1125, 2022-08-07:
another error-> I tensorflow/core/platform/cpu_feature_guard.cc:142] This TensorFlow binary is optimized with oneAPI Deep Neural Network Library (oneDNN) to use the following CPU instructions in performance-critical operations:  AVX2 FMA
To enable them in other operations, rebuild TensorFlow with the appropriate compiler flags.
Traceback (most recent call last):
  File "try.py", line 79, in <module>
    color_image = np.asanyarray(color_frame.get_data())
RuntimeError: null pointer passed for argument "frame_ref"

[image]

MartyG-RealSense, 2022-08-07:
The simplest way to handle this may be to first delete the **config.enable_stream(rs.stream.depth)** instruction at the line immediately before pipeline.Stop()

Then change pipe.start(config) to **pipe.start()** with empty brackets.  When the pipeline is restarted, the script should then apply the camera's default profile, which should enable both the depth and color streams at their default resolution and FPS values.

sosoyeon1125, 2022-08-07:
#config.enable_stream(rs.stream.depth)
pipeline.stop()
pipe.start()     -> OK

' When the pipeline is restarted, the script should then apply the camera's default stream profile, which should enable both the depth and color streams at their default resolution and FPS values. '
-> How do i do this? when i ran the code i fixed, it turned blue

MartyG-RealSense, 2022-08-07:
The depth and color streams are enabled **automatically** when the pipe.start brackets are empty.  You do not have to write any code to enable them.

sosoyeon1125, 2022-08-07:
and 
No boxes are created in the human position. 
A min 1.3M and  max 1.6M also seems useless............

[image]

MartyG-RealSense, 2022-08-07:
An alternative to using the default stream configuration is to put **config** back in the pipe.start brackets and then copy the two config.enable_stream lines from near the beginning of your script and paste them at the line above pipe.start(config) so that 'pipe' uses the same stream configuration that 'pipeline' did.

sosoyeon1125, 2022-08-07:
threshold_filter = rs.threshold_filter()
threshold_filter.set_option(rs.option.max_distance, 1.6)
threshold_filter.set_option(rs.option.min_distance, 1.3)

try:
  frames = pipe.wait_for_frames()
  frames_filtered = threshold_filter.process(frames)

->Where should "frames_filtered"  be located??? 
 "frames_filtered" there is mix1.6M & min 1.3M value inside, but the code seems to be not working....

sosoyeon1125, 2022-08-07:
another error....

Traceback (most recent call last):
  File "try5.py", line 73, in <module>
    depth_frame = frames_filtered.get_depth_frame()
AttributeError: 'pyrealsense2.pyrealsense2.frame' object has no attribute 'get_depth_frame'

MartyG-RealSense, 2022-08-07:
Please try changing frames_filtered = threshold_filter.process(frames) to this:

**frames_filtered = threshold_filter.process(frames_filtered)**

Does the threshold filter then work correctly?

Is **try5.py** a different script to try.py?  If it is then please post images of the try5.py script.

sosoyeon1125, 2022-08-07:
try5 is try: <-put in the code
other than that, it is the same as the original code.
[image]
 
"Please try changing frames_filtered = threshold_filter.process(frames) to this:
frames_filtered = threshold_filter.process(frames_filtered)
Does the threshold filter then work correctly?"
-> same ..
[image]

[image]

MartyG-RealSense, 2022-08-07:
It appears that the try-except mechanism in your threshold code is printing "Error" because when the script tries to apply the threshold filter, an error occurs but the except instruction handles it and prints "Error".

In the initial 'pipeline' section of code, a pipeline called 'pipeline' is opened, depth is aligned to color using **align_stream** and a pointcloud is generated, and then 'pipeline' is closed.

Later in the second pipeline called 'pipe', align_stream from the first pipeline is called in the line **frames = aligned_stream.process(frames)**

But 'frames' is defined in the previous line as pointing to the second 'pipe' pipeline instead of the first 'pipeline' pipeline where the aligned depth-color image was created.

This may have been why originally you were using **frames = pipeline.wait_for_frames()** on this line until I suggested changing it to 'pipe' in https://github.com/IntelRealSense/librealsense/issues/10728#issuecomment-1207359499 because at that time I did not see how the second 'pipe' pipeline needed to access aligned-frame data from the first 'pipeline' pipeline.  I do apologize.

sosoyeon1125, 2022-08-07:
[**try5**.py All code]
[image]

[image]

**error**
Traceback (most recent call last):
  File "try5.py", line 74, in <module>
    depth_frame = frames_filtered.get_depth_frame()
AttributeError: 'pyrealsense2.pyrealsense2.frame' object has no attribute 'get_depth_frame'
[image]

MartyG-RealSense, 2022-08-08:
The script may benefit from being checked section by section to confirm what each part is doing and whether it is needed.

For example, in lines 20 to 23 of pipeline 'pipeline' a pointcloud is set up and stored in 'points'.  But the pointcloud is not generated from 'points' until line 76 of pipeline 'pipe'.  

Splitting the script code between two pipelines and having instructions in pipeline 2 depend on instructions in pipeline 1, and also inserting TensorFlow code in-between the librealsense code, increases the complexity of debugging.

sosoyeon1125, 2022-08-08:
error
Traceback (most recent call last):
File "try5.py", line 74, in
depth_frame = frames_filtered.get_depth_frame()
AttributeError: 'pyrealsense2.pyrealsense2.frame' object has no attribute 'get_depth_frame'
Screenshot from 2022-08-08 02-03-55

->https://github.com/IntelRealSense/librealsense/issues/8170#issuecomment-761930301
The below application successfully filtered the point cloud to under the configured distance option (1). The units seem to be in meters. I don't know what defines this, but assume it ties back to [option.depth_units](https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.option.html?highlight=option#pyrealsense2.option.depth_units). Further, the filter would only accept the frames type and not a frames.get_depth_frame() type. I'm still new to rs and cannot explain why.

->"frames.get_depth_frame() type" I can not ?????

sosoyeon1125, 2022-08-08:
Do you have any other examples of reference?

MartyG-RealSense, 2022-08-08:
Yes  by default real world distance is measured in meters in the RealSense SDK.  You could take the distance value provided by the SDK and perform a calculation on it to convert it into another unit of measurement such as mm though (for example, distance value in m x 1000 = mm)

Yes, depth unit scale affects distance scale.  The default depth scale is 0.001, which is millimeter scale.  A scale of 0.01 is centimeter scale.  The bottom of the section of Intel's **Projection** documentation linked to below confirms this.

https://dev.intelrealsense.com/docs/projection-in-intel-realsense-sdk-20#depth-image-formats

********

What happens if you change line 67 to this:

frames = pipe.wsit_for_frames()

Line 68 to this:

**frames_filtered = threshold.filter.process(frames)**

Then comment out line 72, which is not needed as it is the same as the new line 67

sosoyeon1125, 2022-08-08:
same error..

-> catkin_ws/src/librealsense/wrappers/tensorflow$ python3 try5.py
[INFO] start streaming...
[INFO] loading model...
2022-08-09 03:02:11.849028: I tensorflow/core/platform/cpu_feature_guard.cc:142] This TensorFlow binary is optimized with oneAPI Deep Neural Network Library (oneDNN) to use the following CPU instructions in performance-critical operations:  AVX2 FMA
To enable them in other operations, rebuild TensorFlow with the appropriate compiler flags.
1
threshold_filter <pyrealsense2.pyrealsense2.threshold_filter object at 0x7fd7cc965308>
Traceback (most recent call last):
  File "try5.py", line 74, in <module>
    depth_frame = frames_filtered.get_depth_frame()
AttributeError: 'pyrealsense2.pyrealsense2.frame' object has no attribute 'get_depth_frame'

[image]

MartyG-RealSense, 2022-08-09:
Thanks again for your patience!

If the section of code in lines 37 to 62 (containing TensorFlow code) was moved further down the script of Try5.py to below the current line 85 **# Perform the actual detection by running the model with image as input** then you could prove whether or not the block of TensorFlow code is interfering with the processing of the librealsense code in a way that causes **File "try5.py", line 74, in depth_frame = frames_filtered.get_depth_frame()  AttributeError: 'pyrealsense2.pyrealsense2.frame' object has no attribute 'get_depth_frame'**

jiminiscat, 2022-08-12:
Solved. thank you
I solved it using the depth example.

closed 2022-08-18 by MartyG-RealSense