realsenseai/librealsense #10729: Missing modules & installation issues

agangavaram, 2022-07-27:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   |  | 
| Operating System & Version |   Ubuntu 20.04.4 LTS | 
| Kernel Version (Linux Only)    |  5.17     | 
| Platform                                 | PC |
| SDK Version                            |  2.50 }                          | 
| Language                            |  {}                          | 
| Segment			|  { }                          | 

### Issue Description

I currently am trying to install the SDK in a docker container that is running Ubuntu 20.04.4 with the kernel version of 5.17. This docker container contains another SDK that I also need for my application. I am running into a couple of issues when following the main installation guidelines (using the package managers).

[image]

I understand that this issue is due to 5.17 not being supported, but how can I go about resolving it and ensuring that the modules are loaded into the kernel.

Thanks!

MartyG-RealSense, 2022-07-27:
Hi @agangavaram  If you are unable to change your kernel version due to elements of your project that depend on 5.17 then you could consider the possibility of building the librealsense SDK from source code with CMake using the **RSUSB backend** installation method.  

An RSUSB build bypasses the kernel and so is not dependent on Linux versions or kernel versions and does not require patching.

If you do not need to install additional wrappers for librealsense such as pyrealsense2 then the simple source code build method in the link below would be recommendable to try first.

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md

agangavaram, 2022-07-27:
Thank you so much for responding so quickly! I'm trying to integrate the RealSense camera with NVIDIAs Deepstream SDK i.e. have the RealSense camera be a source for a GStreamer pipeline.

My existing application runs in a Docker container which is based on the NVIDIA Deepstream image. I have a couple of questions regarding the integration between RealSense and other docker containers.

1. Is the SDK absolutely necessary to be able to use the RealSense camera in the way that I have described?
2. If I want my end Docker container to be as lightweight and flexible as possible, should I use the RSUSB build/the source code build method in the link that you sent?

The reason I am asking is because the libuvc installation and build has been taking quite some time, and I want to make my end docker build as fast as possible. Thanks again!

agangavaram, 2022-07-27:
Additionally, on completion of installation I am receiving the following error when I try rs-enumerate-devices.
[image]

MartyG-RealSense, 2022-07-27:
1.  Typically, if you plan to stream depth then the RealSense SDK is going to need to be involved.

If you are only streaming RGB then RealSense cameras can be accessed like ordinary webcams.

The link below has a RealSense plugin for GStreamer that can stream depth and color, align them and stream depth only or color only if you wish.

https://github.com/WKDSMRT/realsense-gstreamer

2.  When building from source code with CMake, you can significantly reduce the build size by stripping out debug symbols using the information at https://github.com/IntelRealSense/librealsense/issues/7982

Also, when building from source you can use a **-jx** command to accelerate the build time by allowing the process to make use of the CPU's number of **cores**.

The 'x' of -jx to use will depend on how many **threads** the CPU of the computer that you are building on has.  The thread number is 2x the number of cores.  So an appropriate j-number to use on a 4-core (quad core) CPU would be -j8 whilst on a higher-end CPU with 8 or more cores you could use -j16 

More information about 'make -jx' values is at the source-code installation documentation link below.  -j8 is a good value to use though.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#building-librealsense2-sdk

************

The **Failed to set power state** error can indicate that there was a problem with accessing the camera, either because it could not be detected (perhaps because of insufficient power supplied to it) or another application has already accessed the camera and is actively using it, putting the camera in a 'busy' state.

agangavaram, 2022-07-27:
@MartyG-RealSense Thank you once again! I have looked into the RealSense plugin, and plan to use it. The build optimizations are also extremely helpful!

As for the Failed to set power state, I've verified that the camera is not being accessed by any other applications. Additionally, I have tried every USB port and am still receiving the error. The camera is being detected by the v4l2 utility, so I am not sure how to resolve this problem. I appreciate all your help in this matter!

MartyG-RealSense, 2022-07-28:
Thanks again for your patience as I researched this case!

The project requirement for kernel 5.17 creates challenges for installing librealsense, as we cannot install from Debian packages (as the most recent kernel supported by the packages is 5.4) and the most recent kernel supported for a source code build is 5.15 (via the unofficial test patch).

The librealsense SDK can work with unsupported kernels, though there may be unpredictable consequences in regard to stability.

Would it be possible to test whether the other SDK that you have installed can work with kernel 5.15?  That may simplify things by allowing the RealSense SDK's new 5.15-supporting kernel patch to be applied.  Given that your camera can be detected by the V4L2 utility, building the SDK from source code with RSUSB = false (also known as the **V4L2 backend**) may increase the chances of the camera being detected.

Another option would be to try a source code build with the RSUSB backend that is not dependent on the kernel.  The **libuvc backend** installation script that you tried earlier is actually the direct predecessor of the RSUSB method.  Although they work similarly in principle, it may be a good idea to try the actual modern RSUSB procedure to confirm whether or not RSUSB can work with your Docker container.  Instructions for an RSUSB build can be found at https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692

Thanks again!

agangavaram, 2022-07-28:
@MartyG-RealSense. I switched to the 5.15 kernel and built the SDK from source code with RSUSB = false, and rs-enumerate-devices is able to detect the camera! However, when I try to open realsense-viewer from the command line, I get no response from the terminal. I want to ensure the firmware is updated in the camera and view the camera in realsense-viewer. 
[image]

What could be causing this?

MartyG-RealSense, 2022-07-28:
If the flags **-dbuild_examples=true -dbuild_graphical_examples=true** were included in the CMake build instruction then graphics support and built executable versions of graphical examples and tools such as realsense-viewer should be included in the build.in the usr/local/bin location referenced in the image above after youni out the whereis query.

agangavaram, 2022-07-29:
Ok makes sense, everything works on the RealSense side - I just want to understand a few things.

Does the RealSense SDK use the uvc API or the v4l2 API? How does the SDK work on the lowest level? Additonally, does the viewer communicate with uvc or v4l2? My feed looks good through the viewer, but when using a GStreamer pipeline, the colors are distorted. Additionally, I'm running into errors when trying to update firmware through the viewer. I suspect that this has to do with running the viewer in the docker container, and the lack of direct interface between the container and the RealSense camera.

Again, thank you for all your help! I'd love to learn more about how the RealSense SDK works.

MartyG-RealSense, 2022-07-29:
*  RealSense 400 Series cameras are compliant with UVC 1.5, as advised at https://github.com/IntelRealSense/librealsense/issues/3416#issuecomment-471720605

*  On Linux, V4L2 is the default backend of the librealsense SDK.  More information about V4L2 vs RSUSB can be found at https://github.com/IntelRealSense/librealsense/issues/7904

On Windows, **Microsoft Media Foundation** is the default backend and building the Windows version of the SDK from source code with the RSUSB backend set to true will configure librealsense to use a Windows backend based on a WinUSB rewrite of the UVC protocol instead of Media Foundation.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md

*  As mentioned above, standard Linux tools can access the camera when the SDK is built for the V4L2 backend, though https://github.com/IntelRealSense/librealsense/issues/6841#issuecomment-660859774 advises that the results may not be nicely colorized.

*  The SDK has a **Low-Level Device API** that can provide direct access to the camera sensor hardware.  RealSense SDK programs typically run in the High-Level Pipeline API.

https://dev.intelrealsense.com/docs/api-architecture

*  Another RealSense user at https://github.com/IntelRealSense/librealsense/issues/4652 also had problems with updating firmware from within a Docker container.  They were advised that it may be possible to do so on an RSUSB = true build.  In your particular situation, it may be easier to update outside of Docker.  
 
The firmware driver is stored inside the camera hardware and not on the computer, so you can update on another computer (even a Windows one) and the firmware driver will be carried within the camera to whatever computer you attach it to.

agangavaram, 2022-07-29:
My host machine is running Fedora 35, and I installed the realsense SDK through rpm. I am able to get the viewer up and running, but when I try to install the firmware, I run into the same error, null pointer passed for argument, "device." 

Thank you for the different resources you sent me. From what I understand, there are 2 ways of building the SDK - using a package manager and the manual build. Through the package manager, the SDK will have a V4L2 backend and requires patching of the kernels. This is impossible in a Docker container because of the lack of access to kernel space. However, the manual build allows for the SDK to be built with the RSUSB backend which lives entirely in user space. 

My goal application requires everything to be in a Docker container - so I'm trying to understand the best build for this method. Why does setting RSUSB=false, and using a V4L2 backend work in the container?

Thank you

MartyG-RealSense, 2022-07-30:
A package build of librealsense is usually based on V4L2, yes.

However, at https://github.com/IntelRealSense/realsense-ros/issues/1515#issuecomment-744223832 the developer of the RealSense ROS1 wrapper advised that from December 2020 onwards, building librealsense and the RealSense ROS wrapper together at the same time from packages with the 'Method 1' installation process using the apt-get command installs a package version of librealsense2 that uses **RSUSB** instead of V4L.  So if you needed an RSUSB-based librealsense package build then that could be a route to consider if you do not mind having the ROS wrapper sitting unused on your computer.

Intel's official Docker guide for librealsense 2.50.0 and Ubuntu 20.04 does not make mention of V4L2 or RSUSB.  Instead, it takes the approach of pulling a pre-built librealsense Docker saved in Docker Hub.

https://github.com/IntelRealSense/librealsense/tree/master/scripts/Docker

I am not familiar enough with Docker or Fedora to offer in-depth advice about those subjects though unfortunately.

MartyG-RealSense, 2022-08-05:
Hi @agangavaram  Do you require further assistance with this case, please? Thanks!

agangavaram, 2022-08-08:
I was able to resolve the issues! Thanks!

closed 2022-08-08 by agangavaram

MartyG-RealSense, 2022-08-08:
That's excellent news, @agangavaram - thanks very much for the update. :)