realsenseai/librealsense #10730: How to improve speed of receiving image data with Wi-fi connection

jhforstudy, 2022-07-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |  Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  4.9.253-tegra                                        | 
| Platform                                 | Jetson nano  |
| SDK Version                            |  { legacy / 2.50.0 }                          | 
| Language                            |  {Python }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
Hi, I want to ask more about wi-fi connection of d435 camera.

This is my previous question : https://github.com/IntelRealSense/librealsense/issues/10697
``realsense-viewer`` is finally working well, and it looks like real-time video stream.

https://user-images.githubusercontent.com/48710703/181461776-8f989261-5fa2-4828-9fe9-bd5e3b5e32f7.mp4

`But the problem is`, ``pyrealsense2_net`` module is not supported in Python now, so I cannot use it in my Python script.
Therefore, I changed my project to use ROS topic of depth camera :
``/camera/color/image_raw``, ``/camera/depth/image_rect_raw``

I tried this method:

* In Jetson : 
change ``ROS_MASTER_URI`` in ``~/.bashrc``,
and run ``roslaunch realsense2_camera rs_camera.launch``

* In my laptop :
run ``roscore`` & receive camera topic from Jetson

However, the speed of topic received is very slow. (about 15 seconds per 1 frame)

I think the difference is that,
in ``realsense-viewer`` I used independent ``rs-server`` so it was very fast
But in ROS communication, I didn't use any independent server so it was very slow.

I don't have any idea to improve the topic-received speed using Wifi connection.

Could you give me any advice for me?? Thanks in advance. :+1:

MartyG-RealSense, 2022-07-28:
Hi @jhforstudy  In the couple of cases that I have seen where the rs-server networking system was adapted to use ROS protocols instead of the default RTSP protocol - such as https://github.com/IntelRealSense/realsense-ros/issues/1808 - the best speed that was achieved was 10 Hz on a wired ethernet connection compared to 30 FPS in the original system setup.  

There was a discussion from July 2020 where someone used ethernet with realsense2_camera by configuring ROS_MASTER_URI and also **ROS_IP**.

https://answers.ros.org/question/357185/intel-realsense-nodelets-running-with-remote-roscore-are-dying-bond-broken/

When Intel's networking white-paper mentions ROS, it has the number [10] beside it that refers to a RealSense ROS guide listed in the references at the bottom of the paper for using 3 cameras across 2 computers:

https://github.com/IntelRealSense/realsense-ros/wiki/showcase-of-using-3-cameras-in-2-machines

This suggests to me that you are on the right track in using ROS_MASTER_URI

https://github.com/IntelRealSense/realsense-ros/issues/1673 has a similar situation to yours, where wi-fi streaming in ROS with a Jetson Nano (and a D455 instead of D435) was being attemped.  The RealSense user in that particular case shared their eventual solution at https://github.com/IntelRealSense/realsense-ros/issues/1673#issuecomment-785752752

At https://github.com/IntelRealSense/realsense-ros/issues/2052 there is a lengthy discussion about wireless use of the camera in ROS and drops in performance.  That RealSense ROS user found at https://github.com/IntelRealSense/realsense-ros/issues/2052#issuecomment-923120865 that their infrared topic frequency was plummeting from 30 Hz to 5 Hz if they used a wireless mouse but had no problem with a wired mouse.

That case also references https://github.com/IntelRealSense/realsense-ros/issues/2039 where a RealSense ROS user with a wireless LAN (WLAN) was experiencing very high delay with communicating with a mobile scale-model car.  The delay started at zero and grew over time.  That user's particular problem was resolved by reinstalling the RealSense ROS camera package.

https://github.com/IntelRealSense/realsense-ros/issues/2052#issuecomment-910059689 references how some users with very slow performance have achieved gains by disabling wi-fi power saving.  Disabling X11 forwarding has also helped some people - https://github.com/IntelRealSense/realsense-ros/issues/2052#issuecomment-906586743

jhforstudy, 2022-07-29:
Thanks @MartyG-RealSense ! I'll try those methods.

closed 2022-07-29 by jhforstudy