realsenseai/librealsense #10731: Which distance is measured and output of the camera - D435i

JCPSinIT, 2022-07-28:
Dear community,

i am currently working with a D435i and i am curious about the depth information. Is this information always the distance between the camera and the object (opt 1) or is it the distance between the camera and the perpendicular level of the object (opt 2)? 
In tests, the output distance is independend from the position in the frame. It does not matter, if the object is in the center of the picture or near the edge. This would prefer opt 2. But I can not find any inforamtions about this.

So is there any (valid) source which can verify this?

Thanks for your help in advance.

MartyG-RealSense, 2022-07-28:
Hi @JCPSinIT  Page 17 of the data sheet documents for the 400 Series cameras states that "the depth pixel value is a measurement from the parallel plane of the imagers and not the absolute range" and provides the accompanying diagram below.

https://dev.intelrealsense.com/docs/intel-realsense-d400-series-product-family-datasheet

[image]

JCPSinIT, 2022-07-29:
Thanks for your superfast answer. Helps me alot!

closed 2022-07-29 by JCPSinIT