realsenseai/librealsense #10734: RS Tools cannot be run in AGX Xavier

tugbakara, 2022-07-30:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | 5.13.0.50 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  4.9.253       | 
| Platform                                 | NVIDIA Jetson AGX Xavier (Auvidea X221 mother board) |
| SDK Version                            |  2.50.0                        | 
| Language                            |                           | 
| Segment			|  Robot                          | 

### Issue Description
Firstly, we have followed [this instructions ](https://dev.intelrealsense.com/docs/nvidia-jetson-tx2-installation#building-from-source-using-native-backend) we didn't take any error but [in the picture](https://files.readme.io/33c9697-jetson_l4t_kernel_patches.png) of instruction we just saw uvc video, nothing more. We suspected about hid sensor didn't exist etc. But we skipped this part. We start realsense-viewer in terminal. We took datas of accel and gyro separately, we didn't see both simultaneously. When we run  tools like rs-capture we couldn't take any data from it, we tried rs-pointcloud nothing came, rs-data-collect took emptiness only. Any number was constructed in .xml file. We tried python3 rs-imu-cailbration.py ( we took pyrealsense2 error that's why we passed python3 instead python) but in the output we just waited to see the data but nothing came. Sometimes rs-motion  works quite well sometimes nice. We suspected also power issues. Also we searched all issues from github, intel support and nvidia forum. We tried lots of combination. If you want to some outputs from our terminal  we can provide.

MartyG-RealSense, 2022-07-30:
Hi @tugbakara  Which RealSense 400 Series camera model are you using, please?  RealSense cameras that are **hid** devices are ones equipped with an IMU component (D435i and D455).

When you say that you took data separately from the Accel and Gyro in realsense-viewer, do you mean the overlay of values that appears when you move the mouse cursor over the Accel or Gyro panel?  If so then it is normal for the data overlay on one of these two panels to disappear as the cursor is hovered over the other.

IMU data in realsense-viewer will only be shown in the **2D** mode (accessed by left-clicking on the '2D' option in the top corner of the window) and does not appear in the '3D' pointcloud mode.

*************

The rs-camera example program does not provide an overlay of values like the one in realsense-viewer and just displays the Accel and Gyro ring of axes when the program is used with a RealSense camera model equipped with an IMU.

A RealSense team member created a version of rs-capture at the link below that can save data to the **.csv** file format.

https://github.com/dorodnic/librealsense/tree/text_sample/examples/capture

***********

If you are not seeing data visually rendered on-screen in rs-capture or rs-pointcloud, this could be due to your computer having a problem with a graphics rendering component of the RealSense SDK called **GLSL**.  

This could be tested by going to the '3D' pointcloud mode of the RealSense Viewer and attempting to view a depth pointcloud.  If there is a conflict with GLSL then the pointcloud will typically flicker and then completely disappear a few seconds after the depth stream is enabled.

A fix for this can be to disable the GLSL options in the 'Settings' interface of realsense-viewer.  Instructions for doing so can be found at https://github.com/IntelRealSense/librealsense/issues/8110#issuecomment-754705023

*********

Problems with graphics should not affect the rs-data-collect program though as it is a text-based tool.

*********

I appreciate that there is a lot of informaton to digest in the above comments.  So let's focus on these for the moment and then deal with the pyrealsense2 and IMU calibration issues afterwards.  Thanks again!

TossoEngineering, 2022-07-30:
Hello . I'm teammate of @tugbakara . We'r using D435i and we'll follow your suggestions immediately and let you know. Thanks alot for your quick response!

tugbakara, 2022-08-01:
> Which RealSense 400 Series camera model are you using, please? RealSense cameras that are **hid** devices are ones equipped with an IMU component (D435i and D455).

Firstly, when installation happened, as I mentioned before, we didn't see any hid_sensor_accel_3d , hid_sensor_gyro_3d, hid_sensor_trigger, hid_sensor_iio_common . If they are driver, we couldn't install.

> When you say that you took data separately from the Accel and Gyro in realsense-viewer, do you mean the overlay of values that appears when you move the mouse cursor over the Accel or Gyro panel? If [...]

This is not our problem , it happens but we are not worry about it, we know it is normal. We are not talking about it.

> IMU data in realsense-viewer will only be shown in the **2D** mode (accessed by left-clicking on the '2D' option in the top corner of the window) and does not appear in the '3D' pointcloud mode.

We cannot see IMU datas in 2D mode.

> The rs-camera example program does not provide an overlay of values like the one in realsense-viewer and just displays the Accel and Gyro ring of axes when the program is used with a RealSense camera model equipped with an IMU.

When we run rs-camera in terminal we got:
```
$ rs-camera
bash: rs-camera: command not found
```

> A RealSense team member created a version of rs-capture at the link below that can save data to the **.csv** file format.

when we run rs-capture in terminal only black screen is shown. 

> A fix for this can be to disable the GLSL options in the 'Settings' interface of realsense-viewer. Instructions for doing so can be found at [#8110 (comment)](https://github.com/IntelRealSense/librealsense/issues/8110#issuecomment-754705023)

it is already disabled.
[image]

I will provide the other answers for your reply later. Could you help us for these for now?

tugbakara, 2022-08-01:
> roblems with graphics should not affect the rs-data-collect program though as it is a text-based tool.

when we run in terminal :

```
$ rs-data-collect -c ./data_collect.cfg -f ./log.csv -t 60 -m 1000
Running rs-data-collect: -c ./data_collect.cfg -f ./log.csv -t 60 -m 1000 

Request added for DEPTH,640,480,30,Z16,0
Request added for INFRARED,640,480,30,Y8,1
Request added for INFRARED,640,480,30,Y8,2
Request added for COLOR,640,480,30,RGB8,0
Request added for ACCEL,1,1,100,MOTION_XYZ32F
Request added for GYRO,1,1,200,MOTION_XYZ32F

Device selected: 
	Intel RealSense D435I. USB Type: 3.2
	S.N: 153122072006
	FW Ver: 05.13.00.50

User streams requested: 6, actual matches: 6

Data collection started.... 

Collecting data for 1 sec

```
 goes up to 60 as expected.
then created csv is :
```

Configuration:
Stream Type,Stream Name,Format,FPS,Width,Height
Infrared 1,Infrared,Y8,30,640,480
Infrared 2,Infrared,Y8,30,640,480
Depth,Depth,Z16,30,640,480
Color,Color,RGB8,30,640,480
Accel,Accel,MOTION_XYZ32F,100
Gyro,Gyro,MOTION_XYZ32F,200

Stream Type,Index,F#,HW Timestamp (ms),Host Timestamp(ms)

Depth,0,3,1659343385468.338,11.314
Depth,0,4,1659343385501.682,12.506
Depth,0,6,1659343385601.714,26.342
Depth,0,0,1659343385790.538,204.610
Depth,0,1,1659343385826.793,237.976
Depth,0,2,1659343385861.931,274.799
Depth,0,3,1659343385897.139,306.992
Depth,0,4,1659343385932.418,340.607
Depth,0,5,1659343385967.767,373.148
Depth,0,6,1659343386003.188,406.772

Stream Type,Index,F#,HW Timestamp (ms),Host Timestamp(ms)

Color,0,0,1659343385718.785,169.853
Color,0,1,1659343385754.548,197.774
Color,0,2,1659343385790.558,230.591
Color,0,3,1659343385826.812,266.089
Color,0,4,1659343385861.950,297.598
Color,0,5,1659343385897.158,331.636
Color,0,6,1659343385932.437,362.810
Color,0,7,1659343385967.786,398.902
Color,0,8,1659343386003.207,431.587
Color,0,9,1659343386038.136,463.486

Stream Type,Index,F#,HW Timestamp (ms),Host Timestamp(ms)

Infrared,1,3,1659343385468.338,12.006
Infrared,1,5,1659343385601.714,27.138
Infrared,1,6,1659343385635.058,60.418
Infrared,1,0,1659343385861.931,273.220
Infrared,1,1,1659343385897.139,306.455
Infrared,1,2,1659343385932.418,340.455
Infrared,1,3,1659343385967.767,372.762
Infrared,1,4,1659343386003.188,406.758
Infrared,1,5,1659343386038.118,439.395
Infrared,1,6,1659343386073.065,472.836

Stream Type,Index,F#,HW Timestamp (ms),Host Timestamp(ms)

Infrared,2,3,1659343385468.338,12.044
Infrared,2,5,1659343385601.714,27.158
Infrared,2,6,1659343385635.058,60.462
Infrared,2,0,1659343385861.931,273.252
Infrared,2,1,1659343385897.139,306.499
Infrared,2,2,1659343385932.418,340.557
Infrared,2,3,1659343385967.767,372.813
Infrared,2,4,1659343386003.188,406.797
Infrared,2,5,1659343386038.118,439.441
Infrared,2,6,1659343386073.065,472.875

```

MartyG-RealSense, 2022-08-01:
The realsense-viewer image shows that the camera is being correctly detected as a D435i device with an IMU (motion module) components.

Regarding 'rs-camera' not working, that was an auto-correction mistake by my computer and it should have said **rs-capture**.  I do apologize and have edited that statement.

Regarding only getting a black screen from rs-capture, that does not sound correct.  It should automatically display your D435i camera's depth. RGB and Motion Module streams when the program is run.

As you are receiving data from the IMU with the text-based tool **rs-data-collect**, it shows that the IMU is detected and working.

If rs-capture has a black screen then perhaps there is a problem in the SDK arising from the use of the Auvidea X221 motherboard.   I am not familiar with this board.  It's configuration may be unusual enough that there is a conflict between it and the SDK that may be solved by building the SDK on Jetson from source code with the **RSUSB backend** installation method, as described for Jetson at https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049
.

tugbakara, 2022-08-02:
Thanks, I will try this. if it  works I will give a feedback.

tugbakara, 2022-08-03:
Hi @MartyG-RealSense, we reinstall AGX's firmware and start from zero and we followed the all steps in issue #10304 , and we got it work all the tools, but we have doubts about IMU datas are correct or not. To be sure we want to calibrate our IMU, first of all do we really need to? If yes, when source installation is made, [calibration code](https://github.com/IntelRealSense/librealsense/blob/master/tools/rs-imu-calibration/rs-imu-calibration.py) comes but we follow the apt installation and we don't have IMU calibration code and we really don't want to messed up everything again. What can we do to go forward from this point? 
Have a nice day! :)

MartyG-RealSense, 2022-08-03:
The IMU can work without using the IMU calibration tool, though it may not be able to achieve the ideal target value of 9.80 for Y Accel and instead have a Y Accel value of around 9.4 or 9.6.  Y Accel may also continue to change when the camera is stationary on a flat surface.

Calibrating the IMU and saving the calibration to the camera hardware allows a data correction feature to be enabled that 'fixes' the raw data so that Y Accel is closer to 9.80.  It is not compulsory to perform a calibration though.

tugbakara, 2022-08-04:
Thank you @MartyG-RealSense , we did it too. Have a nice day.

MartyG-RealSense, 2022-08-14:
Hi @tugbakara  Do you require further assistance with this case, please?  Thanks!

tugbakara, 2022-08-16:
No, thanks @MartyG-RealSense , I just want to note here we reinstalled 2.50 and got it work thanks to uninstalling ros-melodic-librealsense2 after ros installation like in the [issue](https://github.com/IntelRealSense/librealsense/issues/10304) that we used and solved problem : `sudo apt purge ros-melodic-librealsense2`

closed 2022-08-16 by tugbakara

MartyG-RealSense, 2022-08-17:
Thanks very much @tugbakara for the update and for sharing the details of your solution!