realsenseai/librealsense #10738: L515 missing depth on black/dark objects

FeiSong123, 2022-08-02:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515| 
| Firmware Version                   | 01.05.08.01 | 
| Operating System & Version |  Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  5.4.0-122-generic | 
| Platform                                 | PC |
| SDK Version                            |  2.50.0              | 
| Language                            |  C++           | 
| Segment			|  Robot       | 

### Issue Description
Sorry to interrupt. Today I used L515 to look at five different color golf ball(pink, green, blue, black, white), and found that the depth of black is incomplete as others complete. Is there any way to improve it, or maybe it is just a problem of TOF camera?

MartyG-RealSense, 2022-08-02:
Hi @FeiSong123  Depth cameras in general (not only RealSense) have difficulty analyzing dark grey or black surfaces unless a strong light source is cast into them to bring out their depth.  More information about this can be found at https://github.com/IntelRealSense/librealsense/issues/6757

FeiSong123, 2022-08-02:
OK, I see. Thank you for your reply.

MartyG-RealSense, 2022-08-09:
Hi @FeiSong123  Do you require further assistance with this case, please?  Thanks!

FeiSong123, 2022-08-10:
No,

closed 2022-08-10 by FeiSong123