realsenseai/librealsense #10742: Installation problem

Olaniyi117, 2022-08-02:
| | |
|---|---|
|**librealsense**|2.50.0 RELEASE|
|**OS**|Windows|

Please provide a description of the problem

I install but it still asks me to add the real sense which is already plugged in via usb

[image]
[image]

MartyG-RealSense, 2022-08-03:
Hi @Olaniyi117  Please check in the Windows camera privacy settings whether permission to access the camera hardware and permission for applications to access the camera are both set to 'On'.  You can do this using the instructions at https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789

Olaniyi117, 2022-08-03:
> to 'On'. You can do this using the instructions at #3381 (comment)

yes, the window camera privacy is turn on and still I have the same problem

MartyG-RealSense, 2022-08-03:
Do you experience the same problem if you install the previous RealSense Viewer version **2.49.0** please?

You can install a standalone executable version of this Viewer by visiting the link below, going to the 'Assets' file list and then
downloading and running the **Intel.RealSense.Viewer.exe** file.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.49.0

You can either run this program by left-clicking on its download tab at the bottom of the browser once download has completed or by going to the Downloads folder of your user documents on Windows and launching it with a double-leftclick (the launch shortcut icon on the Windows desktop will launch the 2.50.0 Viewer instead of the downloaded 2.49.0 one).

MartyG-RealSense, 2022-08-10:
Hi @Olaniyi117  Do you require further assistance with this case, please?  Thanks!

Olaniyi117, 2022-08-10:
Yes i experience the same problem even with 2.49.0

MartyG-RealSense, 2022-08-11:
What is your RealSense camera model name (D415, D435, D455, etc)

Olaniyi117, 2022-08-11:
L515

MartyG-RealSense, 2022-08-11:
Thank you.  Is the camera able to display streams if you try the **rs-capture** example program, please?  If you have installed the full RealSense SDK on Windows by downloading and running the installer file **Intel.RealSense.SDK-WIN10** then you can run a pre-made executable version of rs-capture using the instructions at https://github.com/IntelRealSense/librealsense/issues/8024#issuecomment-747630656

closed 2022-08-29 by MartyG-RealSense