realsenseai/librealsense #10744: Distortion Model in D435i

JCPSinIT, 2022-08-03:
Hello together,

i am currently working with D435i and got the following camera intrinsics for the Color Frame

[ 1280x720  p[642.395 347.588]  f[905.706 904.496]  Inverse Brown Conrady [0 0 0 0 0] ]

I already read here , that throug the Inverse Brown Conrady Distortion Model, the distortion in the frame should be reduced.

But when I try to measure a symetric ball in a corner of the image, the values for the width and height of the object are not equal. 
The values for width are 85pxl and for height 77pxl which leads to a difference of 10,4%, when it should be 0%.

So my question is:
Is this the maximum performance of the model, is it working at all or are there other possibilities to reduce the distortion?

If possible, with a valid reference.

Thanks for your help in advance.

MartyG-RealSense, 2022-08-03:
Hi @JCPSinIT   Do you experience the same difference in X and Y if you use Y16 as the RGB format, as Y16 is unrectified, unlike the other stream formats which are rectified.

JCPSinIT, 2022-08-04:
Hi @MartyG-RealSense , I am experiencing this difference using Y16 as well. 
So is there any possibility to define / activate the distortion model?

Thanks for your help.

MartyG-RealSense, 2022-08-04:
I would suspect that something other than the distortion model is the likely cause of your XY measurement problem.  For example, when performing depth to color alignment and using depth instrinsics, using **aligned intrinsics** instead of depth intrinsics can be more accurate.

The RealSense SDK does not have the ability to change the distortion model or toggle it on / off, though OpenCV has an **undistort** function to remove the model.

https://docs.opencv.org/4.5.2/dc/dbb/tutorial_py_calibration.html

A RealSense user at https://github.com/IntelRealSense/librealsense/issues/3880 tried it.

closed 2022-08-21 by MartyG-RealSense