realsenseai/librealsense #10746: nodejs librealsense install fails (MacOS Monterey, node v17.7.2)

callumhay, 2022-08-03:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | N/A | 
| Operating System & Version |   MacOS (Monterey 12.5)  | 
| Platform                                 | Mac  |
| SDK Version                            |  "node-librealsense": 0.2130.0      | 
| Language                            |  nodejs                         | 

### Issue Description
I've followed the prerequisite installations as described in the repo but despite this, I'm unable to install the nodejs npm module, when I attempt to install it I get the following errors:

```
npm ERR! code 1
npm ERR! path /Users/beowulf/projects/realsense_test/node_modules/node-librealsense
npm ERR! command failed
npm ERR! command sh -c node-gyp rebuild
npm ERR! CXX(target) Release/obj.target/node_librealsense/src/addon.o
npm ERR! gyp info it worked if it ends with ok
npm ERR! gyp info using node-gyp@8.4.1
npm ERR! gyp info using node@17.7.2 | darwin | x64
npm ERR! gyp info find Python using Python version 3.9.12 found at "/opt/anaconda3/bin/python3"
npm ERR! gyp info spawn /opt/anaconda3/bin/python3
npm ERR! gyp info spawn args [
npm ERR! gyp info spawn args   '/usr/local/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py',
npm ERR! gyp info spawn args   'binding.gyp',
npm ERR! gyp info spawn args   '-f',
npm ERR! gyp info spawn args   'make',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/Users/beowulf/projects/realsense_test/node_modules/node-librealsense/build/config.gypi',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/usr/local/lib/node_modules/npm/node_modules/node-gyp/addon.gypi',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/Users/beowulf/Library/Caches/node-gyp/17.7.2/include/node/common.gypi',
npm ERR! gyp info spawn args   '-Dlibrary=shared_library',
npm ERR! gyp info spawn args   '-Dvisibility=default',
npm ERR! gyp info spawn args   '-Dnode_root_dir=/Users/beowulf/Library/Caches/node-gyp/17.7.2',
npm ERR! gyp info spawn args   '-Dnode_gyp_dir=/usr/local/lib/node_modules/npm/node_modules/node-gyp',
npm ERR! gyp info spawn args   '-Dnode_lib_file=/Users/beowulf/Library/Caches/node-gyp/17.7.2/<(target_arch)/node.lib',
npm ERR! gyp info spawn args   '-Dmodule_root_dir=/Users/beowulf/projects/realsense_test/node_modules/node-librealsense',
npm ERR! gyp info spawn args   '-Dnode_engine=v8',
npm ERR! gyp info spawn args   '--depth=.',
npm ERR! gyp info spawn args   '--no-parallel',
npm ERR! gyp info spawn args   '--generator-output',
npm ERR! gyp info spawn args   'build',
npm ERR! gyp info spawn args   '-Goutput_dir=.'
npm ERR! gyp info spawn args ]
npm ERR! gyp info spawn make
npm ERR! gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
npm ERR! ../src/addon.cpp:30:20: error: no matching member function for call to 'Get'
npm ERR!     return v8obj_->Get(Nan::New(name).ToLocalChecked());
npm ERR!            ~~~~~~~~^~~
npm ERR! /Users/beowulf/Library/Caches/node-gyp/17.7.2/include/node/v8-object.h:290:43: note: candidate function not viable: requires 2 arguments, but 1 was provided
npm ERR!   V8_WARN_UNUSED_RESULT MaybeLocal<Value> Get(Local<Context> context,
npm ERR!                                           ^
npm ERR! /Users/beowulf/Library/Caches/node-gyp/17.7.2/include/node/v8-object.h:293:43: note: candidate function not viable: requires 2 arguments, but 1 was provided
npm ERR!   V8_WARN_UNUSED_RESULT MaybeLocal<Value> Get(Local<Context> context,
npm ERR!                                           ^
npm ERR! ../src/addon.cpp:38:13: error: no matching member function for call to 'Set'
npm ERR!     v8obj_->Set(Nan::New(name).ToLocalChecked(), Nan::Undefined());
npm ERR!     ~~~~~~~~^~~
npm ERR! /Users/beowulf/Library/Caches/node-gyp/17.7.2/include/node/v8-object.h:244:37: note: candidate function not viable: requires 3 arguments, but 2 were provided
npm ERR!   V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context,
npm ERR!                                     ^
npm ERR! /Users/beowulf/Library/Caches/node-gyp/17.7.2/include/node/v8-object.h:247:37: note: candidate function not viable: requires 3 arguments, but 2 were provided
npm ERR!   V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context, uint32_t index,
npm ERR!                                     ^
npm ERR! ../src/addon.cpp:42:13: error: no matching member function for call to 'Delete'
npm ERR!     v8obj_->Delete(Nan::New(name).ToLocalChecked());
npm ERR!     ~~~~~~~~^~~~~~
[... 140 lines cut ...]
npm ERR!     ~~~~~~~~~~~~~^~~~~
npm ERR! ../../nan/nan_persistent_12_inl.h:28:15: note: candidate template ignored: could not match 'Local' against 'MaybeLocal'
npm ERR!   inline void Reset(const v8::Local<S> &other) {
npm ERR!               ^
npm ERR! ../../nan/nan_persistent_12_inl.h:33:15: note: candidate template ignored: could not match 'PersistentBase' against 'MaybeLocal'
npm ERR!   inline void Reset(const v8::PersistentBase<S> &other) {
npm ERR!               ^
npm ERR! ../../nan/nan_persistent_12_inl.h:25:15: note: candidate function not viable: requires 0 arguments, but 1 was provided
npm ERR!   inline void Reset() { v8::PersistentBase<T>::Reset(); }
npm ERR!               ^
npm ERR! fatal error: too many errors emitted, stopping now [-ferror-limit=]
npm ERR! 2 warnings and 20 errors generated.
npm ERR! make: *** [Release/obj.target/node_librealsense/src/addon.o] Error 1
npm ERR! gyp ERR! build error 
npm ERR! gyp ERR! stack Error: `make` failed with exit code: 2
npm ERR! gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)
npm ERR! gyp ERR! stack     at ChildProcess.emit (node:events:527:28)
npm ERR! gyp ERR! stack     at Process.ChildProcess._handle.onexit (node:internal/child_process:291:12)
npm ERR! gyp ERR! System Darwin 21.6.0
npm ERR! gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
npm ERR! gyp ERR! cwd /Users/beowulf/projects/realsense_test/node_modules/node-librealsense
npm ERR! gyp ERR! node -v v17.7.2
npm ERR! gyp ERR! node-gyp -v v8.4.1
npm ERR! gyp ERR! not ok
```

MartyG-RealSense, 2022-08-03:
Hi @callumhay  I note that you are using NodeJs 17.7.2. The RealSense SDK currently supports up to NodeJs 14.

callumhay, 2022-08-03:
I tried with node 18.7.0 and I'm getting the same issue:

```
npm ERR! code 1
npm ERR! path /Users/beowulf/projects/realsense_test/node_modules/node-librealsense
npm ERR! command failed
npm ERR! command sh /var/folders/wc/wl4w92ts1978fq13g19708680000gn/T/install-0eeffb07.sh
npm ERR! CXX(target) Release/obj.target/node_librealsense/src/addon.o
npm ERR! gyp info it worked if it ends with ok
npm ERR! gyp info using node-gyp@9.1.0
npm ERR! gyp info using node@18.7.0 | darwin | x64
npm ERR! gyp info find Python using Python version 3.9.12 found at "/opt/anaconda3/bin/python3"
npm ERR! gyp info spawn /opt/anaconda3/bin/python3
npm ERR! gyp info spawn args [
npm ERR! gyp info spawn args   '/usr/local/lib/node_modules/npm/node_modules/@npmcli/run-script/node_modules/node-gyp/gyp/gyp_main.py',
npm ERR! gyp info spawn args   'binding.gyp',
npm ERR! gyp info spawn args   '-f',
npm ERR! gyp info spawn args   'make',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/Users/beowulf/projects/realsense_test/node_modules/node-librealsense/build/config.gypi',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/usr/local/lib/node_modules/npm/node_modules/@npmcli/run-script/node_modules/node-gyp/addon.gypi',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/Users/beowulf/Library/Caches/node-gyp/18.7.0/include/node/common.gypi',
npm ERR! gyp info spawn args   '-Dlibrary=shared_library',
npm ERR! gyp info spawn args   '-Dvisibility=default',
npm ERR! gyp info spawn args   '-Dnode_root_dir=/Users/beowulf/Library/Caches/node-gyp/18.7.0',
npm ERR! gyp info spawn args   '-Dnode_gyp_dir=/usr/local/lib/node_modules/npm/node_modules/@npmcli/run-script/node_modules/node-gyp',
npm ERR! gyp info spawn args   '-Dnode_lib_file=/Users/beowulf/Library/Caches/node-gyp/18.7.0/<(target_arch)/node.lib',
npm ERR! gyp info spawn args   '-Dmodule_root_dir=/Users/beowulf/projects/realsense_test/node_modules/node-librealsense',
npm ERR! gyp info spawn args   '-Dnode_engine=v8',
npm ERR! gyp info spawn args   '--depth=.',
npm ERR! gyp info spawn args   '--no-parallel',
npm ERR! gyp info spawn args   '--generator-output',
npm ERR! gyp info spawn args   'build',
npm ERR! gyp info spawn args   '-Goutput_dir=.'
npm ERR! gyp info spawn args ]
npm ERR! gyp info spawn make
npm ERR! gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
npm ERR! ../src/addon.cpp:30:20: error: no matching member function for call to 'Get'
npm ERR!     return v8obj_->Get(Nan::New(name).ToLocalChecked());
npm ERR!            ~~~~~~~~^~~
npm ERR! /Users/beowulf/Library/Caches/node-gyp/18.7.0/include/node/v8-object.h:290:43: note: candidate function not viable: requires 2 arguments, but 1 was provided
npm ERR!   V8_WARN_UNUSED_RESULT MaybeLocal<Value> Get(Local<Context> context,
npm ERR!                                           ^
npm ERR! /Users/beowulf/Library/Caches/node-gyp/18.7.0/include/node/v8-object.h:293:43: note: candidate function not viable: requires 2 arguments, but 1 was provided
npm ERR!   V8_WARN_UNUSED_RESULT MaybeLocal<Value> Get(Local<Context> context,
npm ERR!                                           ^
npm ERR! ../src/addon.cpp:38:13: error: no matching member function for call to 'Set'
npm ERR!     v8obj_->Set(Nan::New(name).ToLocalChecked(), Nan::Undefined());
npm ERR!     ~~~~~~~~^~~
npm ERR! /Users/beowulf/Library/Caches/node-gyp/18.7.0/include/node/v8-object.h:244:37: note: candidate function not viable: requires 3 arguments, but 2 were provided
npm ERR!   V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context,
npm ERR!                                     ^
npm ERR! /Users/beowulf/Library/Caches/node-gyp/18.7.0/include/node/v8-object.h:247:37: note: candidate function not viable: requires 3 arguments, but 2 were provided
npm ERR!   V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context, uint32_t index,
npm ERR!                                     ^
npm ERR! ../src/addon.cpp:42:13: error: no matching member function for call to 'Delete'
npm ERR!     v8obj_->Delete(Nan::New(name).ToLocalChecked());
npm ERR!     ~~~~~~~~^~~~~~
npm ERR! /Users/beowulf/Library/Caches/node-gyp/18.7.0/include/node/v8-object.h:328:37: note: candidate function not viable: requires 2 arguments, but 1 was provided
npm ERR!   V8_WARN_UNUSED_RESULT Maybe<bool> Delete(Local<Context> context,
npm ERR!                                     ^
npm ERR! /Users/beowulf/Library/Caches/node-gyp/18.7.0/include/node/v8-object.h:333:37: note: candidate function not viable: requires 2 arguments, but 1 was provided
npm ERR!   V8_WARN_UNUSED_RESULT Maybe<bool> Delete(Local<Context> context,
npm ERR!                                     ^
npm ERR! ../src/addon.cpp:46:13: error: no matching member function for call to 'Set'
npm ERR!     v8obj_->Set(Nan::New(name).ToLocalChecked(),
npm ERR!     ~~~~~~~~^~~
npm ERR! /Users/beowulf/Library/Caches/node-gyp/18.7.0/include/node/v8-object.h:244:37: note: candidate function not viable: requires 3 arguments, but 2 were provided
[... 130 lines cut ...]
npm ERR!     ~~~~~~~~~~~~~^~~~~
npm ERR! ../../nan/nan_persistent_12_inl.h:28:15: note: candidate template ignored: could not match 'Local' against 'MaybeLocal'
npm ERR!   inline void Reset(const v8::Local<S> &other) {
npm ERR!               ^
npm ERR! ../../nan/nan_persistent_12_inl.h:33:15: note: candidate template ignored: could not match 'PersistentBase' against 'MaybeLocal'
[... cut ...]
npm ERR! gyp ERR! stack     at ChildProcess.emit (node:events:513:28)
npm ERR! gyp ERR! stack     at ChildProcess._handle.onexit (node:internal/child_process:291:12)
npm ERR! gyp ERR! System Darwin 21.6.0
npm ERR! gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/@npmcli/run-script/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
npm ERR! gyp ERR! cwd /Users/beowulf/projects/realsense_test/node_modules/node-librealsense
npm ERR! gyp ERR! node -v v18.7.0
npm ERR! gyp ERR! node-gyp -v v9.1.0
npm ERR! gyp ERR! not ok
```

callumhay, 2022-08-03:
Could be helpful to add that to "engines" in the package.json. I had checked and there was nothing in there so I figured node version wouldn't impact the install...
Are there any plans for supporting higher nodejs versions (it's at 18.7 now)?

MartyG-RealSense, 2022-08-03:
Support for NodeJs 14 as the newest version supported was added in SDK 2.44.0 (see the 'New Features' section of the release notes link below).

https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#release-2440

In a previous discussion with my Intel RealSense colleagues about updating the newest supported NodeJs version, it was advised that RealSense community members can submit their own updates for the NodeJs wrapper as **pull requests**.  

There is a detailed discussion about use of newer versions at https://github.com/IntelRealSense/librealsense/issues/10322

MartyG-RealSense, 2022-08-14:
Hi @callumhay  Do you require further assistance with this case, please?  Thanks!

callumhay, 2022-08-16:
I just want to voice my frustration with librealsense in general. I paid over $600 CAD for a device I thought would be well supported across platforms and I'm getting segmentation faults or permissions issues or power-related error codes every time I attempt to use it. I've recompiled all of librealsense from source and stepped through the code only to find mismatched calls across threads between libusb_init, device usage via libusb, and libusb_exit across the library (most of these issues can be tracked back to [https://github.com/IntelRealSense/librealsense/blob/master/src/libusb/context-libusb.cpp](https://github.com/IntelRealSense/librealsense/blob/master/src/libusb/context-libusb.cpp), but I don't know enough about the library (nor do I have the time to devote the many hours it would take) to fix it. The cpp bridge code doesn't work for nodejs on any remotely recent version of node and from what I've seen digging through the source, the code hasn't really been modified or modernized for many years.

Regardless, I would expect that for such a expensive product that there would be better support in the libraries ... heck, I could go out and buy an old kinect 2 and have orders of magnitude better support for it at 1/10th the cost - sure I'd be sacrificing resolution, but at least it would work on my system. I think Intel should consider this moving into the future and up their game when it comes to maintaining their own libraries - especially when it comes to arguably luxury products like the realsense.

MartyG-RealSense, 2022-08-17:
I'm genuinely sorry to hear about the problems that you have been experiencing.  The most recent MacOS version that is officially supported is High Sierra (10.13), and although the RealSense SDK continued to work well in newer MacOS releases such as Catalina and Big Sur, compatibility with Monterey has been problematic.

Although official MacOS updates in the SDK have ceased after High Sierra, submission to the SDK of updates from RealSense's Mac community of developers are welcomed.

Similarly with the NodeJs wrapper, it is maintained but not updated and RealSense NodeJS users are encouraged to submit community-created updates for the wrapper.

The next version of librealsense is currently targeted for release around Q3 2022.

MartyG-RealSense, 2022-08-28:
Case closed due to no further comments received.  I sincerely apologize again for the installation difficulties that you experienced.

closed 2022-08-28 by MartyG-RealSense