realsenseai/librealsense #10747: Out of range value for argument 'def' in RealSense ethernet demo

Kueshen, 2022-08-03:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400
| Firmware Version                   |5.13.0.50 | 
| Operating System & Version |  Ubuntu 20.04 | 
| Kernel Version (Linux Only)    |  (5.15.13)                                         | 
| Platform                                 | Gaming Laptop (Alder Lake + nvidia rtx3070ti)  |
| SDK Version                            |  { 2.0 }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {VR/AR}                          | 

### Issue Description
Evaluation RealSense capabilities to run Ethernet demo using D455. Following https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras to bring it up. When running realsense-viewer I get the error "RealSense error calling: out of range value for argument "def. This error is similar to the one reported in #7459. However, I've not been able to fix the issue with the recommended course of action.  The librealsense.log file on server side thrwos this info:

3/08 16:21:34,608 WARNING [139655557470016] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.3.auto
 03/08 16:21:34,608 WARNING [139655557470016] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.2.auto
 03/08 16:21:34,810 WARNING [139655557470016] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.3.auto
 03/08 16:21:34,810 WARNING [139655557470016] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.2.auto
 03/08 16:21:35,023 WARNING [139655557470016] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.3.auto
 03/08 16:21:35,024 WARNING [139655557470016] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.2.auto
 03/08 16:21:35,231 WARNING [139655557470016] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.3.auto
 03/08 16:21:35,232 WARNING [139655557470016] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.2.auto
 03/08 16:21:35,432 WARNING [139655557470016] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.3.auto
 03/08 16:21:35,432 WARNING [139655557470016] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.2.auto
 03/08 16:21:35,637 WARNING [139655557470016] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.3.auto
 03/08 16:21:35,637 WARNING [139655557470016] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.2.auto
 03/08 16:21:35,846 WARNING [139655557470016] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.3.auto
 03/08 16:21:35,846 WARNING [139655557470016] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.2.auto
 03/08 16:21:36,057 WARNING [139655557470016] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.3.auto
 03/08 16:21:36,057 WARNING [139655557470016] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.2.auto
 03/08 16:21:36,269 WARNING [139655557470016] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.3.auto
 03/08 16:21:36,269 WARNING [139655557470016] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.2.auto
 03/08 16:21:49,549 WARNING [139655557470016] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.3.auto
 03/08 16:21:49,549 WARNING [139655557470016] (backend-hid.cpp:1203) Failed to read busnum/devnum. Custom HID Device Path: /sys/bus/platform/drivers/hid_sensor_custom/HID-SENSOR-2000e1.2.auto
220803162149.581 [librealsense] I: Init sensor 1, option 'Backlight Compensation', value 0
220803162149.582 [librealsense] E: [getOption] error: 500 Invalid Option - 500
220803162149.582 [librealsense] I: Init sensor 1, option 'Brightness', value 0
220803162149.583 [librealsense] E: [getOption] error: 500 Invalid Option - 500
220803162149.583 [librealsense] I: Init sensor 1, option 'Contrast', value 50
220803162149.584 [librealsense] E: [getOption] error: 500 Invalid Option - 500
220803162149.584 [librealsense] I: Init sensor 1, option 'Exposure', value 33000
220803162149.585 [librealsense] E: [getOption] error: 500 Invalid Option - 500
220803162149.585 [librealsense] I: Init sensor 1, option 'Gain', value 16
220803162149.586 [librealsense] E: [getOption] error: 500 Invalid Option - 500
220803162149.586 [librealsense] I: Init sensor 1, option 'Gamma', value 300
220803162149.587 [librealsense] E: [getOption] error: 500 Invalid Option - 500
220803162149.587 [librealsense] I: Init sensor 1, option 'Hue', value 0
220803162149.587 [librealsense] E: [getOption] error: 500 Invalid Option - 500
220803162149.587 [librealsense] I: Init sensor 1, option 'Saturation', value 64
220803162149.588 [librealsense] E: [getOption] error: 500 Invalid Option - 500
220803162149.588 [librealsense] I: Init sensor 1, option 'Sharpness', value 50

MartyG-RealSense, 2022-08-04:
Hi @Kueshen  I note that you are using kernel 5.15.13.  The newest supported kernel versions at the time of writing this are 5.4 if building the SDK from Debian packages or 5.8 or 5.11 if building from source code.

An unofficial test patch has been made available though at https://github.com/IntelRealSense/librealsense/issues/10439#issuecomment-1171965106 that adds support for kernels 5.13 and 5.15.

The SDK can work with unsupported kernels but there may be unpredictable consequences in regards to stability.  Alternatively, the kernel can be bypassed when building from source code with the **RSUSB backend** so that the SDK build is not dependent on Linux versions or kernel versions and does not require kernel patching.

MartyG-RealSense, 2022-08-14:
Hi @Kueshen  Do you require further assistance with this case, please?  Thanks!

yurrodtl, 2022-08-16:
Just had this issue when trying to run RealSense over network. The error occurs on the anti-flickering parameter. On client side, there are 3 values for it:
0 - None
1 - 50Hz
2 - 60Hz

Yet, the rs-server sends your a value of 4, which is "Auto". Client side doesn't have this value and when checks the ranges of received values, on anti-flickering one, it indicates the value received from server being out of range. I made a quick fix for myself, that puts received value back into known range, so, when 4 comes, I set it to 3, as it is the maximum value.

Also, I did go through all the rs-server profiles and put the anti-flickering value to be 1 (50Hz), rebuilt the server, yet value of 4 remains.

A good and proper fix would be really nice.

MartyG-RealSense, 2022-08-17:
Hi @yurrodtl  I will discuss your reported issue with my Intel RealSense colleagues.  Thanks very much for your patience!

Nir-Az, 2022-08-25:
> Just had this issue when trying to run RealSense over network. The error occurs on the anti-flickering parameter. On client side, there are 3 values for it: 0 - None 1 - 50Hz 2 - 60Hz
> 
> Yet, the rs-server sends your a value of 3, which is "Auto". Client side doesn't have this value and when checks the ranges of received values, on anti-flickering one, it indicates the value received from server being out of range. I made a quick fix for myself, that puts received value back into known range, so, when 3 comes, I set it to 2, as it is the maximum value.
> 
> Also, I did go through all the rs-server profiles and put the anti-flickering value to be 1 (50Hz), rebuilt the server, yet value of 3 remains.
> 
> A good and proper fix would be really nice.

Hi @yurrodtl ,
Do you get the same behavior when running with USB?
I am trying to clarify if it's a network device issue or a general one.
Thanks

yurrodtl, 2022-08-25:
Hi,

Direct USB connection works fine, it occurs only when trying to pair a
camera over the network.
There is another hint to the issue, when trying to view a camera over
network, the PointCloud function doesn't generate proper points. So, even
when I avoid that 'def' error, I still cannot get a 3D object. There are
some rare runs when PointCloud actually works, but it happens in about 1-2%
of the runs and the reason for it to work (or not to work) is unclear to me.
I didn't bring it up in the thread, because I see it as another issue,
which I might open later on.

MartyG-RealSense, 2023-03-17:
I am closing this issue as the rs-server ethernet networking system is removed in the next librealsense SDK release.  This is because it was an experimental proof of concept rather than an official, validated Intel RealSense feature.  This networking tool will therefore not receive further fixes / updates in subsequent SDK releases.

closed 2023-03-17 by MartyG-RealSense