realsenseai/librealsense #10748: How can I Using  multiple pipelines with one realsense device  on the rock pi rk3399 ?

itgo067, 2022-08-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.12.13.50 | 
| Operating System & Version |  Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |  Linux 5.10.103+ aarch64                              | 
| Platform                                 |Rock Pi rk3399|
| SDK Version                            |  2.50              | 
| Language                            | cpp                         | 
| Segment			|  Robot                        | 

### Issue Description
I am using the realsense d435i camera in ubuntu on a Rock Pi 4B Plus, and I want to read the Infrared data stream and the color data stream using two separate pipelines.

But on the Rock Pi I get an error, only one pipeline started ok ,and working well. But on my PC in windows 10, it works perfectly.

The code is as follows.  

https://gist.github.com/itgo067/11be993ff30a98fe1f9fc9efe81486a5

 When I call `run_stereo` and `run_color` function with aync thread . It will be run errors. The errors are as follows. 

```cpp
color 2 error calling rs2_pipeline_start_with_config(pipe:0xaaaad11ed8e0, config:0xaaaad11ed8a0):
    No device connected
```

In additonal , when I run `run_color_stereo` function  . it using one pipeline , every stream working well . but frame rate will down . I set Infrared stream fps 90 , color stream  30 fps . but  I get frame rate 60 fps .

Now I want run Infrared stream 90fps , and color stream 30 fps on my project .

I cann't found any infos for this problem . 

how can I set some option to run two pipelines with one realsense device on rock pi platform . Or how to set some option to run one pipeline but frame rate  keep it the same as I set ?

MartyG-RealSense, 2022-08-04:
Hi @itgo067  Let's start with maintaining the same FPS rate on one pipeline.  If auto-exposure is enabled and a separate RGB option called **auto-exposure priority** is disabled then the SDK should try to force FPS to be maintained at a constant rate instead of being allowed to vary.  An example of C++ code for disabling auto-exposure priority is at https://github.com/IntelRealSense/librealsense/issues/7285#issuecomment-693179828

itgo067, 2022-08-04:
@MartyG-RealSense  Hello,  I test disable auto exposure priority option in my code . but also get 60 fps .  when I disable color stream , the ir stream can become 90 fps.

```cpp
#include <thread>
#include <librealsense2/rs.hpp>
#include <opencv2/opencv.hpp>

cv::Size color_size(1280, 720);
int color_frameRate = 30;

cv::Size stereo_size(640, 480);
int stereo_frameRate = 90;

rs2::context ctx;
std::shared_ptr<rs2::config> cfg_color_stereo;

void run_color_stereo()
{
  rs2::colorizer color_map;
  rs2::pipeline pipe_color_stereo(ctx);
  auto last_fps_time = std::chrono::system_clock::now();
  auto last_fps_time_color = last_fps_time;
  auto last_fps_time_stereo = last_fps_time;

  int fps_num_color = 0;
  int fps_num_stereo = 0;

  int show_fps_color = 0;
  int show_fps_stereo = 0;

  std::cout << "cfg  color stereo" << std::endl;

  try
  {
    cfg_color_stereo = std::make_shared<rs2::config>();
    // cfg_color_stereo->enable_stream(RS2_STREAM_COLOR, color_size.width,
    //                                 color_size.height, RS2_FORMAT_BGR8,
    //                                 color_frameRate);
    cfg_color_stereo->enable_stream(RS2_STREAM_INFRARED, 1, stereo_size.width,
                                    stereo_size.height, RS2_FORMAT_Y8,
                                    stereo_frameRate);
    cfg_color_stereo->enable_stream(RS2_STREAM_INFRARED, 2, stereo_size.width,
                                    stereo_size.height, RS2_FORMAT_Y8,
                                    stereo_frameRate);
    cfg_color_stereo->enable_device(sn_realsense);
    
    auto profile_color_stereo = pipe_color_stereo.start(*cfg_color_stereo);

    auto sensor1 = profile_color_stereo.get_device().first<rs2::depth_sensor>();
    sensor1.set_option(RS2_OPTION_ENABLE_AUTO_EXPOSURE, 1);

    rs2::color_sensor color_sensor = profile_color_stereo.get_device().query_sensors()[1];
    color_sensor.set_option(RS2_OPTION_ENABLE_AUTO_EXPOSURE, 1);
    color_sensor.set_option(RS2_OPTION_AUTO_EXPOSURE_PRIORITY, 0);

    bool stop_color = false;
    while (1)
    {
      rs2::frameset frames = pipe_color_stereo.wait_for_frames();
      // if (!stop_color) {
      //   rs2::frame color = frames.get_color_frame().apply_filter(color_map);
      //   const int w2 = color.as<rs2::video_frame>().get_width();
      //   const int h2 = color.as<rs2::video_frame>().get_height();
      //   cv::Mat color_image(cv::Size(w2, h2), CV_8UC3,
      //                       (void *)color.get_data(), cv::Mat::AUTO_STEP);
      //   if (!color_image.empty())
      //   {
      //     fps_num_color++;
      //     cv::putText(color_image, "FPS: " + std::to_string(show_fps_color),
      //                 cv::Point(20, 50), cv::FONT_HERSHEY_COMPLEX, 2.,
[... 60 lines cut ...]
          last_fps_time_stereo = current_fps_time;
          fps_num_stereo = 0;
        }

        left_image.release();
        right_image.release();
      }
    }
  }
  catch (const rs2::error &e)
  {
    std::cerr << "color stereo error calling " << e.get_failed_function() << "("
              << e.get_failed_args() << "):\n    " << e.what() << std::endl;
    // return EXIT_FAILURE;
  }
  catch (const std::exception &e)
  {
    std::cerr << "color stereo " << e.what() << std::endl;
    // return EXIT_FAILURE;
  }

  //pipe_color_stereo.stop();
}

```

itgo067, 2022-08-05:
@MartyG-RealSense thanks . But in my project , i cloudn't use any other gpu etc. 

Now I want use two pipeline to process IR stream and color stream .  Can you give me some advice? 

I use two pipeline working well on  my pc . but  on the Rock Pi , I can run only one pipeline to work .

MartyG-RealSense, 2022-08-05:
If the script works fine on PC but not on the Rock Pi then that suggests that the code is correct and the problem may be with the **5.10.103+ aarch64** kernel that is used (one that is not officially supported by the librealsense SDK).

If librealsense is built from source code with the **RSUSB backend** installation method then the SDK bypasses the kernel, so that it is not dependent on Linux versions or kernel versions and does not require kernel patching.

**********

Before creating an RSUSB build though, you could first try adding a **hardware_reset()** instruction before the pipe start to reset the camera when the program runs, as described by the C++ script at https://github.com/IntelRealSense/librealsense/issues/2161#issuecomment-408779150

closed 2022-08-20 by MartyG-RealSense