realsenseai/librealsense #10749: Motion Module failure! [d435i nx ]

Abner0907, 2022-08-05:
Hi, when I run `realsense-viewer`,
and switch to the 2D screen,  open the Motion module, it will not show the imu data, and shows the errors: Motion Module failure!

How ever, when I run `sudo realsense-viewer`. all works normally.

FYI: the stereo module and RGB CAMERA can work all the time with or without sudo.

Is there any device need supper permission?

I have debug this issue a long time and can not find the reason. hope help.

MartyG-RealSense, 2022-08-05:
Hi @Abner0907   Does the 'nx' in this issue's title mean that you are using an Nvidia Jetson Xavier NX computing board, please?  If you are, an NX user at https://github.com/IntelRealSense/realsense-ros/issues/2213#issuecomment-1008126279 resolved the motion module failure error by flashing their Jetson to JetPack 4.5.1.

Abner0907, 2022-08-05:
@MartyG-RealSense Hi, so nice of  you for your quick response.
 yes, I have use the   Nvidia Jetson NX computing board.
 
 I have refer your link #2213 issue, very helpful. but not fix my issues. 
 I have solved it by remove all the realsense packages include librealsense modules and ros sides modules.
 and reinstall a different version. which solves my problems. I replaced the default librealsense 2.50.0 by 2.49.0. now the `realsense-viewer ` works good. 
 
 However, I still have another issues, when I run the ros command, it will not publish the `/camera/imu` topic. 
 Do you have any suggestions?

MartyG-RealSense, 2022-08-05:
Is the 'imu' topic published if the instruction below is used?

roslaunch realsense2_camera rs_camera.launch enable_depth:=false enable_color:=false enable_accel:=true enable_gyro:=true accel_fps:=100 gyro_fps:=200 unite_imu_method:="copy" enable_infra1:=true enable_infra2:=true infra1_width:=640 infra1_height:=480 infra1_fps:=30 infra2_width:=640 infra2_height:=480 infra2_fps:=30

Abner0907, 2022-08-05:
@MartyG-RealSense  I have try below, there is no difference. 
roslaunch realsense2_camera rs_camera.launch enable_depth:=false enable_color:=false enable_accel:=true enable_gyro:=true accel_fps:=63 gyro_fps:=200 unite_imu_method:="copy" enable_infra1:=true enable_infra2:=true infra1_width:=640 infra1_height:=480 infra1_fps:=30 infra2_width:=640 infra2_height:=480 infra2_fps:=30
The outputs is: 
[image]

MartyG-RealSense, 2022-08-05:
The log states that the camera is on a USB 2.1 connection instead of USB 3.  Infra2 is not available on USB 2.1, so the infra2 references will have to be removed from the roslaunch instruction.

Let's also see if changing 'copy' for 'linear,_interpolation' makes a difference.

roslaunch realsense2_camera rs_camera.launch enable_depth:=false enable_color:=false enable_accel:=true enable_gyro:=true accel_fps:=63 gyro_fps:=200 unite_imu_method:="linear_interpolation" enable_infra1:=true infra1_width:=640 infra1_height:=480 infra1_fps:=30

Abner0907, 2022-08-05:
@MartyG-RealSense I have try both, and no difference.

I am doubt the ros wraper's issues, what I use now is Realsense Ros v2.3.2. 
Do you think  a new version is good?

MartyG-RealSense, 2022-08-05:
2.3.2 is the newest wrapper available for ROS1.  Changing to an older wrapper would require installing an older version of the librealsense SDK.

Xavier NX has a USB 3 port, so the camera should be detected as USB 3 in the launch log instead of the slower USB 3.1 if the camera is attached directly to the NX's USB port.

Are you using your own choice of USB cable instead of the official one supplied with the camera?

Is the IMU topic published if you add **initial_reset:=true** to the launch instruction to reset the camera at launch?

Abner0907, 2022-08-05:
@MartyG-RealSense  I have changed to a usb3 line and test the result. It is the same. 
Now I have switch to another ros rapper, 2.2.16.  
All the results are the same. 

[image]

Abner0907, 2022-08-05:
@MartyG-RealSense 
I have another NX PC board, it can work normally with all the function.
I have found the only difference is  librealsense version is 2.48.0. So I will switch and test in the new board and let you know the results. 
[image]

Abner0907, 2022-08-05:
@MartyG-RealSense 
Hi, Can you let me know how to switch the Librealsense version. 
you can see below picture, the version is different, How can i check to v2.48.0. 
[image]

Abner0907, 2022-08-05:
hi, thank you.
I think I have solved all the issues by now.
do as belows:
1: remove the ros realsense package.
2: clone the realsense ros package and rebuild it.

now I use the realsense ro v2.2.16.
built realsense with librealsense v2.48.0
and running with librealsense v2.48.0 

all the functions is perfect.

closed 2022-08-05 by Abner0907

MartyG-RealSense, 2022-08-05:
It's excellent to hear that you found a solution, @Abner0907 - thanks very much for the update and for sharing your solution with the RealSense community!

Abner0907, 2022-08-06:
yes, it is my pleasure. 
BTW, I think the librealsense v2.49.0 and librealsense v2.50.0 is not compatible with ubutun nx 18.04 very friendly. I have waste two days at this.