realsenseai/librealsense #10750: Noise pattern

Obrepal, 2022-08-05:
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (5.13.0.50) | 
| Operating System & Version |    MacOS  | 
| Platform                                 | PC  |
| SDK Version                            |   2.5                     | 
| Language                            |  python                          | 

### Issue Description
So I would like to know what is the reason of noise pattern. 
 I took 100 photos of same object, then add values to each other and in the and I calculated variance of each point. 
 A clear  pattern can be seen and I am wondering about its reason but cannot find sufficient answer. 
 
 According to https://github.com/IntelRealSense/librealsense/issues/3411 it might be stereoscopic system design
  
 
[image]

MartyG-RealSense, 2022-08-05:
Hi @Obrepal  Does the pattern disappear if you disable the camera's infrared dot pattern projection using the Python code at https://github.com/IntelRealSense/librealsense/issues/10065#issuecomment-994776524

Obrepal, 2022-08-05:
I think it might because this pattern is very similar to emitted one, but I just want to know what is the reason. I do not need to change it.

MartyG-RealSense, 2022-08-05:
The projector that casts the infrared dot pattern is a separate component from the infrared sensors and will project the pattern even if the infrared streams are not enabled unless the **IR Emitter** is turned off or the **Laser Power** option's value is reduced to make the pattern less visible.

Obrepal, 2022-08-05:
Ok, I get it. But why this noise pattern looks this way?

MartyG-RealSense, 2022-08-05:
Would it be possible to first disable the IR Emitter in your script to confirm whether or not the noise is being caused by the emitter or has another cause?

closed 2022-08-20 by MartyG-RealSense