realsenseai/librealsense #10752: Four D435i setup causes frequent freezing/crashing on Ubuntu 18.04

pmnev2022, 2022-08-05:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i x4 | 
| Firmware Version                   | 5.13.0.50 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  5.4.0-122-generic            | 
| Platform                                 | PC / Laptop  |
| SDK Version                            |  2.50.0                          | 
| Language                            |   C++ (ROS1 Melodic)                          | 
| Segment			|  Other                          | 

### Additional Installation Information

The current install method that was used is to install the ROS packages with Method 1 as referenced in the readme of the realsense-ros GitHub page. The overall system installation is fairly minimal, with few other packages installed.

Previous attempts to install from source also had the same issue, but may have been done incorrectly. The most similar issue was #10159 , but they use a different kernel version and installation approach.  

### Hardware Setup

4x D435i connected to four separate USB ports on the PC

--- or ---

4x D435i connected to a single [Powered USB 3.1 hub](https://www.amazon.com/Powered-Rosonway-Aluminum-Splitter-RSH-A10S/dp/B09NNQFZ2S) that is then connected to the computer. 

---

Tested on two systems:

1. PC: Intel i7-4790K, 16GB RAM
2. Laptop: intel i7-6600U, 16GB RAM

### Issue Description

Hello, the issue that we are experiencing is with running four cameras simultaneously. Occasionally, the system freezes and has to be reset. The cause has yet to be isolated, as freezes have been found to occur under various circumstances. Here is a list of some situations that crashes have occurred in:

1. Running RealSense Viewer and starting 4x Depth Stream (640 x 480, 30fps) and 4x RGB Stream (1280x720, 30fps or 640x480, 30fps)
4. Using a custom version of rs_multicamera.launch to start 4 cameras (realsense-ros nodes) . Several stream configurations were tried: 4x RGB only (640x480, 30fps), 4x RGB (640x480, 30fps) and 4x Depth (640x480, 30fps), and finally: 4x RGB (640x480, 30fps), 4x Depth (640x480, 30 fps), and Point Clouds enabled (align depth disabled). initial_reset is set to true. Notably, a delay was introduced between the starting of different cameras, to help lighten processing load. serial_no is specified to ensure consistent behavior.
5. In addition to the launch file, a custom program to process the point clouds.
6. In addition to the launch file, RViz to simply view the point clouds.
7. Running rs-pointcloud example program with no additional programs.

Single camera operations have not been documented to cause crashes. Two camera configurations have been tested and do not cause crashes. Three camera configurations have not been tested. 

If this is a software error, what information would you require to isolate it, and what steps would you recommend for us to take? The terminal often outputs information about the camera operations, but there does not seem to be a correlation between a particular output and crash. The syslog has been examined, but it is unclear and inconsistent across crashes. It can be posted here if it would help you diagnose the issue. Perhaps a more stable USB backend may be used? A different Linux kernel version?

If this is a hardware error or limitation, what information, e.g. CPU monitor at time of crash or USB bandwidth monitor, hardware specification, could help diagnose this? Perhaps a more powerful CPU is required? Maybe the USB controller is out of spec?

MartyG-RealSense, 2022-08-05:
Hi @pmnev2022  An i7 processor is recommendable for use of four RealSense 400 Series camera's simultaneously.  Whilst you have two i7 computers, the CPU codenames indicate that they are old technology.  

The first number after the 'ix' number is the generation of the CPU.  i7-4790K is 4th Generation (Haswell) dating to around 2014.  i7-6600U is 6th Generation (Skylake) dating from around 2016.  For comparison, the original 400 Series cameras (D415 and D435) were launched in late 2017 and the D435i in late 2018.

In regard to delaying multiple camera launches in ROS to improve multicam launch reliability, the feature **usb_port_id** was created as an alternative to serial_no so that cameras could be identified by the USB port that they are attached to instead of the camera serial number.  The disadvantage of this approach is that specific cameras in a multicam set would always have to be plugged into the same USB port if you need to access a particular camera by that port ID.

Since 640x480 resolution is being used, it is unlikely that enough bandwidth is being consumed to exceed a maximum bandwidth allowance (such as 5 Gbps / 5000 Mbps on a typical USB hub).  Charts supplied by Intel at the link below provide bandwidth estimates for multiple cameras on direct-connection and USB hub setups.

https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration#2-multi-camera-considerations

pmnev2022, 2022-08-05:
Thank you for the quick response!

We can consider alternative processor options and evaluate their performance, but we would like to confirm that this is not a software issue prior to upgrading the processor. Could you help us do this?

Additionally, while we are using 640x480 resolution, this is a measure that was taken to have four cameras work at all. The multiple cameras do all work simultaneously, but the system crashes/freezes. We attempted to increase the resolution, which would be preferred, on both the depth and RGB streams, but this resulted in being unable to start four streams at all, so it did not get to the point of crashing the system. 

The error that occurred when starting the fourth camera the node reports the following (three cameras are already running):

` 05/08 11:57:28,388 WARNING [140487937750784] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b`
` 05/08 11:57:28,478 WARNING [140487415547648] (messenger-libusb.cpp:92) usb_request_queue returned error, endpoint: 132 error: Cannot allocate memory, number: 12 `
` 05/08 11:57:28,478 ERROR [140487415547648] (dispatcher.cpp:35) Dispatcher [0x7fc5d02b4d28] exception caught: failed to submit UVC request while start streaming
`

The resolution specified here is 1280x720, 30fps RGB, 848x480, 30FPS Depth. This error was after simply manually starting four cameras in different terminals by specifying different camera names and serial numbers, can try by specifying the USB port instead, but that does not appear to be the issue at hand here. This error is consistent. The computer does not crash while doing this, but there is no visualization or processing going on. 

Additionally, based on the output of program "usbtop", USB bandwidth does not appear to be the problem, as each camera is only using 28.8 megabits/s. This is strange however, as according to the multi-camera-considerations post, the configuration we are using would be expected to use at least 221 megabits/s per camera (640x360 30 fps Depth and RGB). The bandwidth does not appear to be limiting our ability to handle the larger resolution.

MartyG-RealSense, 2022-08-05:
In https://github.com/IntelRealSense/realsense-ros/issues/1515#issuecomment-744223832 the developer of the ROS1 wrapper advised that since December 2020 the **apt-get install ros-melodic-realsense2-camera** installation method (Method 1) installs a version of the librealsense SDK that is based on the **RSUSB backend** instead of the V4L backend.  The RSUSB backend is not as suited to use with multiple cameras as the V4L backend (a build based around a kernel patch).

You could setup a V4L-based librealsense build by installing the SDK from packages using the instructions at the link below and then building the ROS wrapper from source code after librealsense has been successfully installed.

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

The most recent kernel version supported by a package build with the above instructions is kernel 5.4.  A source code build of SDK 2.50.0 supports 5.8 and 5.11, with the ability to add 5.13 and 5.15 support with an unofficial test patch at https://github.com/IntelRealSense/librealsense/issues/10439#issuecomment-1171965106

pmnev2022, 2022-08-05:
Thank you so much for the instructions on how to properly install the packages for multi camera operations. The crashing seems to have been completely eliminated and the realsense-viewer is now much more responsive. There appear to be issues with our "3.1" hub, as we are unable to make use of the bandwidth of our computer's 3.1 port effectively while having this hub plugged in - it gets recognized as a 3.0 hub and the cameras stop sending data after enabling streams. The hub works when connected to a 3.0 port on the computer, but there is insufficient bandwidth on the 3.0 port to run 4 cameras at our desired resolution.

Do you have any recommendations for tested hubs that may work better?

MartyG-RealSense, 2022-08-06:
It's great to hear that performance has improved significantly for you!

With the USB bandwidth charts from Intel's multiple camera white paper document that were quoted earlier, the paper states that these tests were performed with Amazon's own-brand **AmazonBasics** mains powered USB 3.1 hub.  I use this hub brand myself based on the paper's recommendation and have experienced no problems.

pmnev2022, 2022-08-08:
Thank you for the recommendation. I will close the issue since there are no additional questions. 

For future reference, the solution that worked for solving the primary issue in this post was to first uninstall the existing realsense SDK, install the version of the SDK with the V4L USB backend as recommended, and then building the realsense-ros packages from source. After this, there have been no more crashes.

closed 2022-08-08 by pmnev2022

MartyG-RealSense, 2022-08-08:
Thanks very much for the update and the sharing of what worked for you!

corwin0815, 2022-10-31:
Hello, I was facing a similiar issue and I was using two D435i.
And no matter which version of the librealsense installed can't solve it.
After I set both D435i's resolution from 1920x1080 to 1280x720, both cameras work well simultaneously.
So I believe the error `(messenger-libusb.cpp:92) usb_request_queue returned error, endpoint: 132 error: Cannot allocate memory` means that the data transfering is too big that my device can't deal with it.
Just offer a solution for others who may face the same problem.

MartyG-RealSense, 2022-10-31:
Thanks so much @corwin0815 for sharing your experience with the RealSense community!