realsenseai/librealsense #10754: Could D400 series Camera identify info for an AprilTag when camera is moving？

yuey05, 2022-08-08:
Could D400 series Camera identify info for an AprilTag when camera is moving？

MartyG-RealSense, 2022-08-08:
Hi @yuey05   Apriltags are almost never used with RealSense camera models other than the T265 Tracking Camera.  It is possible to use a different type of fiducial image marker called **ArUco** with the 400 Series cameras though.

yuey05, 2022-08-09:
1.Why couldn't use Apriltags use with RealSense camera models other than the T265 Tracking Camera? Cos D400 could recognize the Apriltags, just could not recognize it in real time. 
2.Also, you recommend ArUco. Why could I use it to get the info of 6 DoF when moving the camera? Are there some examples?
3.I need to identify it using Raspberry Pi, but the speed of recognize is too slow. Are there any solution?

MartyG-RealSense, 2022-08-09:
1.  As far as I am aware there is not a technical reason preventing use of Apriltags with RealSense 400 Series cameras but there are a greater number of references for ArUco.  Some Apriltag projects are designed to work with webcams in general and happen to work with RealSense cameras too.

An Apriltag project that has been confirmed as RealSense compatible by its users is **apriltag_ros**

https://github.com/AprilRobotics/apriltag_ros

https://blog.actorsfit.com/a?ID=01700-6a8e167e-0c0e-4a0b-8802-0bdd947f1b4e

There is also an Apriltag project for Unity at the link below whose developer said that they had used with a RealSense integration project.

https://www.npmjs.com/package/ababilinski.jp.keijiro.apriltag

2.  **roboy_realsense** is a C++ ArUco project designed for RealSense and ROS, though it's very old now.

https://github.com/Roboy/roboy_realsense

**Aruco_Tracker** is a newer project for Python, though designed for webcams in general rather than RealSense specifically.

https://github.com/njanirudh/Aruco_Tracker

You can also find demonstrations of RealSense ArUco marker detection on YouTube by searching it for the term **realsense aruco 6dof**

3.  As ArUco detection is a feature of OpenCV, that may be the most processing-efficient means of using it on Raspberry Pi.  A YouTube video of such a project is at https://www.youtube.com/watch?v=cIVZRuVdv1o

The official OpenCV documentation for ArUco detection is hee:

https://docs.opencv.org/4.x/d5/dae/tutorial_aruco_detection.html

MartyG-RealSense, 2022-08-14:
Hi @yuey05  Was the information provided above helpful to you, please?  Thanks!

closed 2022-08-26 by MartyG-RealSense