realsenseai/librealsense #10755: Ubuntu Linux Server 2204 install

cirobr, 2022-08-08:
Cheers! Can you please provide guidance for installation at the aforementioned OS?

Thanks in advance.

MartyG-RealSense, 2022-08-08:
Hi @cirobr  Details for installing the librealsense SDK (and the RealSense wrapper for ROS2 Humble if you also wish to) for Ubuntu 22.04 Jammy can be found by visiting https://github.com/IntelRealSense/librealsense/issues/10439#issuecomment-1171965106 and reading down through the messages in that discussion.

MartyG-RealSense, 2022-08-19:
Hi @cirobr  Do you require further assistance with this case, please?  Thanks!

cirobr, 2022-08-19:
Hello, thanks for the response and follow up. Have managed to set my system up, thanks for that.

 

Regards,

 

Ciro.

Enviada em: sexta-feira, 19 de agosto de 2022 02:43
Para: IntelRealSense/librealsense ***@***.***>
Cc: Ciro B Rosa ***@***.***>; Mention ***@***.***>

MartyG-RealSense, 2022-08-19:
I'm very pleased to hear that you were successful, @cirobr - thanks very much for the update!

closed 2022-08-31 by MartyG-RealSense