realsenseai/librealsense #10756: Error: Exception was thrown during user processing callback

GHCK57, 2022-08-09:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } -> D415| 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   {Win (10) |                                       | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python}                          | 

### Issue Description
Hi to everyone,

I bought a new one realsense D415 camera. When I open it with intel realsense viewer, everything looks fine. But, there is an error occured in logger. **Exception was thrown during user processing callback: Out of frame resources.** This error comes frequently. Also there are lots of warning. 

What is the reason of this error and warnings. I searched this problem on google and didn't found any solution related with  this error.

This is not my first time using camera. I worked with it before. I didn't faced the error like this. 

The error occurs when I change the "preset" configuration

**synthetic-stream.cpp:48 - Exception was thrown during user processing callback: Error occured during execution of the processing block! See the log for more info**

**synthetic-stream.cpp:48 - Exception was thrown during user processing callback: Out of frame resources!**

Screen record is in link.

I need your support. 
Best regars

https://user-images.githubusercontent.com/97222370/183583800-ed4b9d05-bcdf-49de-aff5-62d88be1a427.mp4

MartyG-RealSense, 2022-08-09:
Hi @GHCK57  I note that you are using Viewer version **2.48.0**.  Do the errors still occur if you use version **2.50.0** of the Viewer, please?

You can test this by going to the link below and downloading and running the file **Intel.RealSense.Viewer.exe** from the 'Assets' file list at the bottom of the page.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.50.0

GHCK57, 2022-08-09:
Hi,

I downloaded the 2.5 version of Realsense Viewer but nothing has changed. 

Note: I have D435 camera. It gives the same error too.

MartyG-RealSense, 2022-08-09:
Thank you for testing with 2.50.0.

Have your D415 and D435 been placed unused in storage for a period of weeks / months since the last time when they were working without errors?

Can the Viewer display streams correctly if the preset or the options are not changed?

GHCK57, 2022-08-09:
**Have your D415 and D435 been placed unused in storage for a period of weeks / months since the last time when they were working without errors?**

No I was using D435 daily but in my own program. I bought a new D415 camera. It arrived to me yesterday. When I tried D415 camera on Realsense viewer I saw that error. After that I tried D435 camera , I faced with the same error.

 **Can the Viewer display streams correctly if the preset or the options are not changed?**

No, sometimes the error is published to logger screen.

MartyG-RealSense, 2022-08-09:
Let's check whether the problem is limited to the Viewer only by running the example program **rs-capture**.  A pre-made executable version of it should already be on your computer if you have installed the full RealSense SDK on Windows using the **Intel.RealSense.SDK_WIN10** automated installer program in the Assets list.

Once the full SDK is installed, you can find the rs-capture.exe executable quickly by right-clicking on the RealSense Viewer launch icon on the Windows desktop and select **Open File Location** from its menu.to open the SDK's Tools folder.

GHCK57, 2022-08-09:
Okey,

What we should see in there ? 

The streams are : 

[image]

MartyG-RealSense, 2022-08-09:
In addition to the RGB and Depth streams, I would expect to also see an **infrared** stream displayed by rs-capture when a D415 is attached, which would suggest that there is a problem with your RGB and Depth camera drivers on Windows that is causing infrared to be missing.

If the infrared streams are currently inaccessible then you may be able to resolve the problem by performing a complete uninstall and reinstall of the two drivers in the Windows **Device Manager** interface.  Instructions for doing so can be found at the link below.

https://support.intelrealsense.com/hc/en-us/community/posts/4419989666323/comments/4431239847443

GHCK57, 2022-08-10:
I tried it. Still same..

MartyG-RealSense, 2022-08-10:
Do the errors in the log continue if you disable the two **GLSL** options in the Viewer's settings interface using the instructions at https://github.com/IntelRealSense/librealsense/issues/8110#issuecomment-754705023

GHCK57, 2022-08-11:
Hi,

Yes, The error in the log continue when the GLS options has disabled.

MartyG-RealSense, 2022-08-11:
Next, please download and run the **Intel.RealSense.SDK-WIN10** SDK installer file again.  This will completely overwrite the previous SDK installation on your Windows computer and should remove any hidden problems that there might have been in the previous installation.  Then launch the Viewer with the launch shortcut icon on the Windows desktop.

GHCK57, 2022-08-11:
Hi,
I followed the instructions but nothing changed

MartyG-RealSense, 2022-08-12:
*  Do the errors in the Viewer still occur if you only enable depth and use **640x480** resolution and the default **Custom** preset?

*  Do the same errors occur in the **Depth Quality Tool**, which can also be launched with a shortcut icon from the Windows desktop?  The design and programming of this tool is similar to the Viewer, and so if the Depth Quality Tool has similar problems then it would indicate that there is something about this type of program that your particular Windows PC has a conflict with.

*. Does a much earlier standalone version of Intel.RealSense.Viewer.exe such as **2.21.0** have the log errors?  If it does not then it could indicate that the conflict is with something introduced in a later version.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.21.0

Thanks again for your patience!

MartyG-RealSense, 2022-08-18:
Hi @GHCK57  Do you require further assistance with this case, please?  Thanks!

GHCK57, 2022-08-22:
@MartyG-RealSense 

Thanks for your support. No, I don't need assistance. Thanks a lot.

MartyG-RealSense, 2022-08-22:
Thanks very much @GHCK57 for the update.  As you do not require further assistance, I will close this case.  Thanks again!

closed 2022-08-22 by MartyG-RealSense