realsenseai/librealsense #10757: Memory leak while doing pipeline start - stop

niho518, 2022-08-09:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | - | 
| Operating System & Version |   Win (10)  | 
| Kernel Version (Linux Only)    |  -  | 
| Platform                                 | PC |
| SDK Version                            |  2.50.0 }                          | 
| Language                            |  {C }                          | 
| Segment			|  {others }                          | 

### Issue Description

Hi,
I noticed significant increase of memory usage over time while performing start/stop IMU stream via RaelSense Viewer. Same is with start/stop of depth or rgb streams.
I tried to debug the code and came to the conclusion that if I comment out "glCopyTexImage2D" function call (file rendering.h, inside function draw_motion_data) memory increase  stops. Is this issue known?

MartyG-RealSense, 2022-08-09:
Hi @niho518  The top line of the What's New section of the release notes for RealSense SDK version 2.50.0 describe how repeated stop-start (like in a stress test) can cause intermittent failures if firmware version **5.13.0.50** is currently installed in the camera.

https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#whats-new

A fix for memory leaks during stop-start of the IMU was also implemented in version **2.48.0** so it may be worth testing 2.48.0 to see whether your issue might be a regression in 2.50.0 from the previous fix.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.48.0

You can obtain a standalone version of the 2.48.0 Viewer from the 'Assets' file list at the above link by downloading and running the file **Intel.RealSense.Viewer.exe**

niho518, 2022-08-10:
Hi @MartyG-RealSense,
with 2.48.0 and D415 camera I can reproduce a memory leak while starting and stopping RGB stream (will test IMU on D435i later during the day).

MartyG-RealSense, 2022-08-10:
Thanks very much for the information.  I look forward to your later D435i test results.

niho518, 2022-08-10:
Tested with D435i (wtih RealSense SDK version 2.48.0) memory leak can be reproduced. Simple continuous start/stop of IMU stream in RealSense viewer.

MartyG-RealSense, 2022-08-10:
Thanks for the results.  Which tool are you using to track the memory leak, please?  For example, it was found that detecting leaks with Valgrind may result in false positive results of leaks, as discussed at https://github.com/IntelRealSense/librealsense/issues/9141#issuecomment-853658523

niho518, 2022-08-11:
I'm using visual studio 'Diagnostic Tools' and windows task manager (simply observe memory usage increase over time).

MartyG-RealSense, 2022-08-11:
As you found that the leaks stopped if you commented out **glCopyTexImage2D**, does the leak still occur if you update the graphics drivers on your Windows computer and therefore update the OpenGL driver too in doing so?

MartyG-RealSense, 2022-08-18:
Hi @niho518  Do you require further assistance with this case, please?  Thanks!

niho518, 2022-08-18:
Hi @MartyG-RealSense, my gpu drivers are up to date so the initial problem still exist.

MartyG-RealSense, 2022-08-19:
Which firmware version are you using, please?  Firmware **5.13.0.50** included a fix for intermittent failures during repeated depth stop-start. 

https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#whats-new

[image]

closed 2022-08-31 by MartyG-RealSense