realsenseai/librealsense #10760: Windows10 thinks realsense2 is a trojan

TBS1996, 2022-08-09:
https://i.imgur.com/P0qgRvm.png

SDK downloaded: Intel.RealSense.SDK-WIN10-2.50.0.3785

MartyG-RealSense, 2022-08-10:
Hi @TBS1996  Was the RealSense SDK installer file Intel.RealSense.SDK-WIN10-2.50.0.3785 downloaded from the 'Assets' file list of the SDK's **Releases** page, please?

https://github.com/IntelRealSense/librealsense/releases/tag/v2.50.0

TBS1996, 2022-08-10:
Yes, it was.

MartyG-RealSense, 2022-08-10:
I have not seen this exe file trigger a Trojan warning in protective software before as it is a safe download.  Do you have the option under the Actions drop-down menu to ignore the warning and proceed with running the file?

TBS1996, 2022-08-10:
it didnt flag it during the installation, I just noticed the warning afterwards, I had some linker issues but not sure if thats due to certain files being quarantined or not

MartyG-RealSense, 2022-08-10:
Does the camera seem to perform normally in the **RealSense Viewer** tool that can be launched with the shortcut icon that the installer should have placed on the Windows desktop?

TBS1996, 2022-08-10:
Yes the camera works without an issue in the realsense viewer, so I dont know if anything actually broke because of this trojan detection

MartyG-RealSense, 2022-08-10:
If the Viewer works normally then it is likely that other RealSense programs will work correctly too.  Because you have installed the full RealSense SDK, you can test with a range of pre-built executable example programs in the SDK's 'Tools' folder.  You can find this easily by right-clicking on the Viewer shortcut icon on the Windows desktop and selecting **Open file location** from its menu.  The programs **rs-capture.exe** and **rs-pointcloud.exe** are excellent test programs to start with.

MartyG-RealSense, 2022-08-19:
Hi @TBS1996  Do you require further assistance with this case, please?  Thanks!

TBS1996, 2022-08-20:
Hello, sorry for late reply, no i don't need any assistance, I never had a real problem in the first place I just wanted to notify of this warning. I'll close this issue, thank you for the attention you've been giving me.

closed 2022-08-20 by TBS1996

MartyG-RealSense, 2022-08-20:
You are very welcome.  Thanks again for highlighting the warning that you experienced!