realsenseai/librealsense #10761: How to align depth stream to color stream without open color stream?

Huskywyh, 2022-08-10:
How to align depth stream(640*480) to color stream(640*480) without open color stream when the color camera is used by another application？
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i  | 
| Operating System & Version | Win10 | 
| Platform                                 | PC |
| SDK Version                            | 2.50 | 
| Language                            |  C#/unity/C++/python }                          |

MartyG-RealSense, 2022-08-10:
Hi @Huskywyh  If the color sensor has been "claimed" first by another application and cannot be accessed until that application releases it, a workaround may be to combine depth and **infrared** to create a colorized image that resembles an aligned depth and color image.  The RealSense SDK has a C++ OpenCV example script and demonstration image at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/stepbystep/getting_started_with_openCV.md#displaying-infrared-frame

If you would prefer not to use OpenCV, an alternative may be to align depth and infrared using a Python script at https://github.com/IntelRealSense/librealsense/issues/5093#issuecomment-544354522 that aligns depth, RGB and infrared and commenting out the color stream from the script.  The infrared image would be monochrome though.

If your project has the option to change camera model then the D415 and D455 models can stream color RGB from the left infrared sensor, leaving the RGB sensor available.

Huskywyh, 2022-08-10:
Sorry, I want to align depth stream(640 480) to color stream(640 480) without opening color stream instead of aligning depth to infrared. Do you know how to do it ?

MartyG-RealSense, 2022-08-10:
The problem is that according to the **Multi-Streaming Model** of the RealSense SDK described at the link below, once a stream is accessed by one application then that particular stream is blocked from being accessed on the same camera by a second application until the first application stops using that stream.

https://github.com/IntelRealSense/librealsense/blob/master/doc/rs400_support.md#multi-streaming-model

An example of this principle would be enabling streams in the RealSense Viewer and then launching another RealSense program such as rs-capture that wants to access the same streams.  The second program may not be able to complete its launch and just show a black window.

Streaming full RGB and RGB from the infrared sensor (which is an RGB format, not infrared) is a valid way to stream two RGB images simultaneously, but unfortunately the D435i model does not support this feature.

closed 2022-08-23 by MartyG-RealSense