realsenseai/librealsense #10763: No Frames Received in RealSense Viewer for RGB Camera and L500 Depth Sensor/Infrared

Vkhark, 2022-08-10:
| | |
|---|---|
|**librealsense**|2.49.0 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense L515|
|**Serial Number**|f1381306|
|**Firmware Version**|01.05.08.01|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|L500|
|**Asic Serial Number**|0003cba4cfcb|
|**Firmware Update Id**|0003cba4cfcb|

Please provide a description of the problem

Hello, 
We have bought the Intel Realsense L515 Camera and wanted to record some 3D-Models with it. Until now I've only been trying to make the camera work in the Viewer. 

I downloaded the standalone Viewer. It recognizes the camera, but when I turn on any Streams, I receive the "NO Frames Received" Message. 
[image]

The motion module works and I can also see changes when i move the Camera. 
I get two Warnings when I run the Viewer: 
[image]

I have tried to run the Viewer with the camera on my colleagues Laptops and it worked fine there without any problems. 

Do you have any Idea why I am having this issue?

Thanks in advance, 
Vlad

MartyG-RealSense, 2022-08-10:
Hi @Vkhark  Do the depth and RGB streams work correctly if you shut down and reboot the PC and then launch the RealSense Viewer but do not enable the Motion Module (just depth and RGB)

Vkhark, 2022-08-10:
Hi @MartyG-RealSense, 

No, I am getting the same error even if I don't turn on the motion module. The problem also persists if i start the Viewer as admin. 
[image]

I've also tried to turn the USB-C Input, as it was mentioned in other issues. 
Br, 
Vlad

MartyG-RealSense, 2022-08-10:
As your L515 works correctly on your colleagues laptops, that would seem to rule out a problem with the camera hardware, the USB cable or the camera firmware driver (which is stored inside the camera).

Please next use the instructions at https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789 to check that permission to access the camera and permission for applications to access the camera are both set to On in the Windows camera permission settings.

If the camera privacy settings are set to On, please then try completely uninstalling and reinstalling the two Depth and RGB camera drivers in the Device Manager interface of Windows using instructions at the link below.

https://support.intelrealsense.com/hc/en-us/community/posts/4419989666323-Installing-D400-UWP-driver-disables-Infrared-streams-on-D435?page=1#community_comment_4431239847443

Did the privacy settings or driver uninstall / reinstall resolve your problem?

Vkhark, 2022-08-10:
Camera Access is activated and I already tried reinstalling drivers. When uninstalling drivers I do not get the option to tick a checkbox. There is no option "Delete the driver software for this device" after pressing "uninstall device"

I can also access the RGB & Depth Camera over the Windows Camera App. It just does not work inside the Viewer.

MartyG-RealSense, 2022-08-10:
Thank you.  Looking at your kindly provided images, I see that the camera is detected correctly in the Viewer, and hardware timestamps being unavailable should not prevent RGB and depth streams from being displayed.

Are these streams able to be displayed if you run the**rs-capture.exe** example program?  This program and other pre-made executable examples and tools programs are located in an SDK folder called 'Tools'.  To find the Tools folder quickly, right-click on the RealSense Viewer launch icon on the desktop and select **Open File Location** from its menu.

Vkhark, 2022-08-10:
When i start the rs-capture.exe, a window with a black screen opens and closes again. 

I called it from terminal and received following error:
[image]

Vkhark, 2022-08-10:
Just to be sure, I just restarted my computer again and started rs-capture again. Still same error. 

I have no other applications running.

MartyG-RealSense, 2022-08-10:
Is your model of laptop the same as the one that your colleagues use?  If the camera works correctly on their machines, perhaps there is a characteristic of your laptop's hardware or the version of Windows on it that causes a conflict with the camera.

For example, in the **Kameras** section of the Device Manager, does it make a difference if you right-click on 'Integrated Camera' and select 'Disable device' to temporarily disable the driver of the laptop's built-in webcam (you can re-enable it at any time by right-clicking on 'Integrated Camera' and selecting 'Enable device' from the menu).

Vkhark, 2022-08-11:
Hi again @MartyG-RealSense ,

I have done some Updates on my Windows PC and I started getting a new Error: 
When I start the Viewer an error is shown: 
[image]

This Error appears also when i start the Depth Sensor or the RGB Camera Stream in the Viewer: 
[image]

I have also disabled the integrated camera.... but that did not change anything. 

BR,
Vlad

Vkhark, 2022-08-11:
Soo. I've installed an Ubuntu VM on Virtualbox and forwarded the USB 3.0. 

[image]

At least I can try the camera and SDK in Ubuntu VM now. 

If you have any other suggestions on how to get it to work on Windows, I'd be happy to try them, as I'd prefer to work there instead of the VM.

MartyG-RealSense, 2022-08-11:
Is the Virtualbox installed on the same Windows PC and the camera inserted in the same USB port?  If so then the camera being able to function in the VM suggests that there is not a problem with that computer's USB port.

Vkhark, 2022-08-12:
The VM is installed on the same PC and I am using the same USB port. Only the viewer and SDK on my native Windows on my Laptop are not able to access the Frames, even though the camera is recognized by the Viewer.

MartyG-RealSense, 2022-08-12:
If you are comfortable with Linux style commands and compiling then there is the option to try building the RealSense SDK from source code on Windows with CMake instead of using the usual installer file to see whether that makes a difference.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md

The section of the guide titled **Enabling metadata on Windows** can be skipped because it is easy to do it in the RealSense Viewer instead once the SDK has been installed.

MartyG-RealSense, 2022-08-20:
Hi @Vkhark  Do you require further assistance with this case, please?  Thanks!

Vkhark, 2022-08-23:
Hi @MartyG-RealSense,

It currently works fine in the VM. So i think I'll just stay there. I might check the installation with CMake in the future. though.
I'm trying to create a handheld-sensor where i can walk around and record objects with it. 
I am trying to repeat this: https://dev.intelrealsense.com/docs/rs-motion in Python. 

I am trying to rotate the axis according to the movement of the camera. But the resulting rotation-matrix, that I get after calculating Theta just makes the Axis spin uncontrollably. I calculated the matrix as follows:
[image]

I wondered if you have an example in Python on how to do it?  I am not too experienced with Python or using Depth Sensors, so any help is appreciated. 

Thanks,
Vlad

MartyG-RealSense, 2022-08-23:
At https://github.com/IntelRealSense/librealsense/issues/4391 where a RealSense Python user attempted to create an equivalent of rs-motion that could extract roll, pitch and yaw, they experienced difficulties such as highly fluctuating results.  

They were advised by a RealSense team member that "in an inertial system affected by G-Force only you can extract the actual Pitch and Roll angles.  You also track the left-right rotation, but the initial Heading angle is arbitrary defined".

closed 2022-09-04 by MartyG-RealSense