realsenseai/librealsense #10764: No frame received!

AMITKUMARMALLICK18, 2022-08-10:
| | |
|---|---|
|**librealsense**|2.50.0 RELEASE|
|**OS**|Linux|
|**Name**|Intel RealSense D455|
|**Serial Number**|037322251658|
|**Firmware Version**|05.13.00.50|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|035723061890|
|**Firmware Update Id**|035723061890|

Please provide a description of the problem

AMITKUMARMALLICK18, 2022-08-10:
It works fine for the first time opening realsense viewer. But next onwards: "No Frame Received." Again, unplugging the cable and plugging works fine and exacts the same repeats. Please look into it.

MartyG-RealSense, 2022-08-11:
Hi @AMITKUMARMALLICK18   An example of a case where this issue has been previously experienced is https://github.com/IntelRealSense/librealsense/issues/1889

Some RealSense users in that case resolved the problem by updating their camera firmware driver, though you are already using the latest firmware version and so cannot update to a newer version yet.

Conceivably, a firmware **downgrade** to the previous firmware version **5.12.15.50** could have the same problem-resolving effect.

https://dev.intelrealsense.com/docs/firmware-releases#d400-series-firmware-downloads

closed 2022-08-29 by MartyG-RealSense