realsenseai/librealsense #10765: What exactly does wait_for_frames do?

alexander-soare, 2022-08-11:
I refer to the [pyrealsense2](https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.pipeline.html?highlight=wait_for_frames#pyrealsense2.pipeline.wait_for_frames) version.

I'm aware of some other questions and answers like https://github.com/IntelRealSense/librealsense/issues/2422#issuecomment-423254709. People tend to use the language "wait_for_frames blocks until a new set of frames becomes available". But I'm not sure what "available" means.

The linked docs say **(A)** "The method blocks the calling thread, and fetches the latest unread frames set" but also says **(B)** "Device frames, which were produced while the function wasn’t called, are dropped." To me these statements are at odds because the first says that I will get a backdated frame, whereas the second says I will wait till the next one become available (because the backdated one was dropped). Which one is it?

Here is an example.

Let's just consider a single stream (say color for instance) at ~30 FPS but to keep things simple let's call it 33.333 FPS so that a frame is captured every 30 ms:

- So at t = 0 ms a frame is captured. Let's call this frame f0.
- At t = 30 ms another frame is captured. Let's call it f30.
- At t = 45 ms I call `wait_for_frames`. Let's call this action c45.
- At t = 50 ms I call `wait_for_frames`. Let's call this action c50.
- f60.
- f90.

So here's what I expect to happen under the wording in A.
- For c45 `wait_for_frames` will return the latest frame that I have not yet received. So I will get f30 instantly.
- For c50  technically (A) says will get f0! That's because I never read that frame. I refuse to believe that's the case though. So if I'm a bit looser with my interpretation of A I can guess it means I need to wait 10 ms before I get f60.

And here's what I expect to happen under the wording in B.
- For c45 I won't get f30. That's because it has already been dropped. Instead, I'll wait 15 ms before getting f60.
- For c50 maybe it means I 10 ms before getting f60 again? If I incorporate some of (A) and assume that f60 can't be read twice, it means I wait 40 ms before getting f90.

So you can see there two main interpretations I get from (A) and (B) and even within those some things aren't crystal clear. Can anyone help clarify it for me?

MartyG-RealSense, 2022-08-11:
Hi @alexander-soare  A simpler way to describe wait_for_frames() would be like this:

*  wait_for_frames() blocks until a complete new frame is available.  This should prevent incomplete frames from being delivered.  By comparison, with poll_for frames the frames arrive immediately without blocking and so there is more scope for complications, such as the CPU maxing out at 100% usage unless the CPU is put to sleep for a time.

 *  The **frame queue capacity** of the pipeline is optimized for performance, though there is a risk of frames being dropped.  By default the pipeline can contain 16 frames of each frame type.  You can increase the total number of frames that can circulate in the pipeline before it becomes full by increasing the value of the frame queue capacity.  The more frames that are held, the more of the computer's memory that will be confused.

*  As new frames enter the pipeline, old frames get pushed out.

Unless you are experiencing a problem with the processing of frames in your application and you need to debug that problem, it is usually not necessary to consider how wait_for_frames and the frame queue function at a deeper level than the points listed above.

alexander-soare, 2022-08-11:
@MartyG-RealSense thanks for jumping in. So just to address your last point: the reason I need to consider this so carefully is I want I very good characterisation of delay and frame rate as I'm tuning a controller for doing dynamic robot manipulation.

>  This should prevent incomplete frames from being delivered

Sorry, what is in "incomplete frame"?

> wait_for_frames() blocks until a complete new frame is available.

Does this mean that if a new complete frame becomes available, then I call `wait_for_frames()` 1 ms later, it means I will wait 29 ms for the next complete frame to become available? (as an aside, consider this in terms of dynamic robot manipulation. I'd rather get the frame that's 1 ms stale then wait 29 ms)

MartyG-RealSense, 2022-08-11:
An incomplete frame can be considered to be a "bad" frame and something to be avoided.  More information about this is at https://github.com/IntelRealSense/librealsense/issues/1240

wait_for_frames is an automated process, so it will wait (block) until a complete (good) frame is available and then should wait until the next complete frame is ready, and so on.

If waiting for a complete frame for 30 ms does not suit your particular application then using poll_for_frsmes() (no waiting for frames) may work better for you.

MartyG-RealSense, 2022-08-11:
An incomplete frame can be considered to be a "bad" frame and something to be avoided.  More information about this is at https://github.com/IntelRealSense/librealsense/issues/1240

wait_for_frames is an automated process, so it will wait (block) until a complete (good) frame is available and then should wait until the next complete frame is ready, and so on.

If waiting for a complete frame for 30 ms does not suit your particular application then using poll_for_frames() (no waiting for frames) may work better for you.

alexander-soare, 2022-08-11:
> wait_for_frames is an automated process, so it will wait (block) until a complete (good) frame is available and then should wait until the next complete frame is ready, and so on.

Sorry but this still doesn't explain what "available" means. If a frame is produced at t=0ms and I call `wait_for_frames` at t=1ms does that not mean the frame at t=0ms is the "available" one?

> and then should wait until the next complete frame is ready, and so on.

To be clear, I need to call `wait_for_frames` again if this is to happen, right? What you're saying is that subsequent calls to `wait_for_frames` won't return the same set of frames again right?

> using poll_for_frames() (no waiting for frames) may work better for you.

What does it mean to not wait for frames? I can't have a frame if it doesn't exist. I must wait. Does poll for frames just get the last complete frame? If I call it twice in a row, does it return the same frame twice?

MartyG-RealSense, 2022-08-11:
My understanding is that with poll_for_frames, new frames arrive instantly without a wait period and are not subject to checks to confirm if the frame is complete before it becomes available. 

If there is a frame drop then the SDK may repeat a frame by returning to the 'last known good frame' and continue onwards from that point as described at https://github.com/IntelRealSense/librealsense/issues/1686

https://github.com/IntelRealSense/librealsense/issues/7837 is also a reference that is worth reading.

I am unfamiliar with the mathematics of wait_for_frames() on a millisecond level though so unfortunately do not have advice to offer on that specific aspect of the instruction.  I do apologize.

alexander-soare, 2022-08-11:
Thanks for your time @MartyG-RealSense. Those links had some useful leads.

MartyG-RealSense, 2022-08-17:
Hi @alexander-soare  Do you require further assistance with this case, please?  Thanks!

alexander-soare, 2022-08-17:
@MartyG-RealSense it's not my priority right now but I suppose the question hasn't really been answered by anyone. Up to you if you want to close it.

MartyG-RealSense, 2022-08-17:
Okay, I will close it and invite other RealSense users who read this case and have questions or advice to leave comments.  Thanks again!

closed 2022-08-17 by MartyG-RealSense